dsmrreader/dsmr-reader #880: Question: Are unprocessed telegrams stored in the database?

Jeltel, 2020-02-15:
I'm using xirixiz docker integration to run dsmrreader in docker on a ubuntu system (with Home Assistant). Since updating to 3.3 the telegrams aren't processed anymore. I've tried to run some of the commands from https://github.com/dennissiemensma/dsmr-reader/issues/867 but I'm running into issues performing these actions and therefore getting into the details of the problem. 
I'm not asking you for help to perform these actions on the docker image. I already asked xirixiz in https://github.com/xirixiz/dsmr-reader-docker/issues/96

I do however would like to know if the unprocessed telegrams are stored in the database.

So when I go to another way of running dsmrreader (outside of docker, or another image provider), I can just import the database and the unprocessed telegrams would be processed.

dennissiemensma, 2020-02-16:
If the Status page is still accessible and the `Unprocessed telegrams` counter keeps increasing, they are indeed stored in the database. If the counter does not increase, then there might be another problem.

Jeltel, 2020-02-16:
Can you give me a hint about getting the telegrams from the database and selecting the oldest one. I think one telegrams blocks the whole queue.

(i'll figure out with xirixiz how to get to the databse)

dennissiemensma, 2020-02-16:
```
sudo su - postgres
psql dsmrreader

select * from dsmr_datalogger_dsmrreading where processed = false order by timestamp desc limit 1;
```

dennissiemensma, 2020-02-16:
Er staat een typo in. Het is `asc`, ipv `desc`.

closed 2020-02-19 by dennissiemensma