dsmrreader/dsmr-reader #881: PVOutput uitlezen

DarkShadowNight, 2020-02-16:
Hallo,

Prachtig programma heb je gemaakt.

Ik had gehoopt dat ik dit echter iets beter combineren met m'n data van PVOutput.
Ik weet het... je kan data versturen naar PVOutput.
Dat m'n opbrengst van m'n zonnepanelen publiek zijn boeit me niet, m'n verbruik ligt veel gevoeliger.
Moeilijk is het niet om af te leiden wanneer er niemand thuis / op vakantie is, een potentieel inbraakrisico.
Jaarlijks doneren om dit te verbergen... laat maar zo dan.

Zou je de data van PVOutput niet kunnen inladen, zodat alles toch één geheel vormt en privé blijft?
Zo hoef je niet verschillende omvormers in rekening te brengen, eventueel een soort mogelijkheid om deze waarden via script te kunnen importeren.

**PS:** Ik heb een meter van Fluvius, maar ik moet de Nederlandse variant gebruiken of het werkt niet.
M'n gasmeter komt ook niet toe in het telegram dus vermoedelijk is daar ook nog een probleempje aan meterzijde.

dennissiemensma, 2020-02-16:
Bedankt voor je vriendelijke woorden en je verzoek. Helaas kan ik je niet helpen, omdat het buiten de scope van dit project valt. Voor de uitgebreide uitleg, zie https://github.com/dennissiemensma/dsmr-reader/issues/594#issuecomment-549546725.

Als je het verbergen van de data toch belangrijk vindt, is mijn advies om simpelweg te doneren aan PVOutput. Je doneert niet alleen voor het verbergen, maar je ondersteunt de ontwikkelaar van dat platform, bijvoorbeeld met de kosten voor de hosting ervan. Het is uiteraard aan jezelf om te bepalen of je het dat wel of niet waard vindt.

dennissiemensma, 2020-02-16:
Wat betreft je gasmeter, als je meer informatie nodig hebt kun je altijd een apart issue daarvoor aanmaken.

closed 2020-02-16 by dennissiemensma