dsmrreader/dsmr-reader #882: Django version foutmelding

Kruijd, 2020-02-16:
Na een upgrade was de webserver niet meer beschikbaar.
Ik heb daarom een volledige herinstallatie geprobeerd meer loop vast bij de uitvoering van 
sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt 

De melding is: 
**ERROR: Could not find a version that satisfies the requirement django==3.0.3 (from -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt (line 3)) (from versions: 1.1.3, ...  2.2.10)
ERROR: No matching distribution found for django==3.0.3 (from -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt (line 3))**

Alle packages zijn op de laatste versie (Sudo apt-get update en upgrade)
kan je me een oplossingsrichting hiervoor geven?
Bij voorbaat dank!

dennissiemensma, 2020-02-16:
Bedankt voor je melding. Wat is de output hiervan?

```
/home/dsmr/.virtualenvs/dsmrreader/bin/python3 --version
```

```
/home/dsmr/.virtualenvs/dsmrreader/bin/pip3 --version
```

Kruijd, 2020-02-16:
/home/dsmr/.virtualenvs/dsmrreader/bin/python3 --version
Python 3.5.3
pip 20.0.2 from /home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/pip (python 3.5)

jeroenpeters1986, 2020-02-17:
Hoi @Kruijd,

naar welke versie van DSMR-Reader wilde je updaten? Naar de nieuwste?
Als dat laatste het geval is dan zul je je Python versie moeten upgraden, ik zie in de documentatie op https://dsmr-reader.readthedocs.io/nl/v3/requirements.html namelijk staan dat je minimaal Python 3.6 nodig hebt.

Op die pagina staat ook 
> Ondersteuning voor Python 3.5 is vervallen sinds DSMR-reader v3.0

Ik heb eerder beschreven hoe ik zelf de update heb gedaan in https://github.com/dennissiemensma/dsmr-reader/issues/624#issuecomment-485377652
Hopelijk heb je daar iets aan!

Kruijd, 2020-02-17:
Dankjewel @jeroenpeters1986 , dit klinkt als de verklaring van het probleem, ik ga hiermee aan de slag.
Bedankt voor de hulp!

closed 2020-02-17 by Kruijd