dsmrreader/dsmr-reader #883: Show electricity usage as stacked bar chart

DiedB, 2020-02-16:
I saw that that gas usage in the archive section was switched to a bar chart, which is a nice improvement. The bar chart suits the type of data much better.

I would like to see the electricity chart in the same way, but then as a stacked bar chart (to be able to differentiate between tariffs). The current chart is pretty much unreadable when comparing consumption data between different days/months/years.

[image]

dennissiemensma, 2020-02-17:
Thanks for the suggestion. I'll have a look for the possibilities.

dennissiemensma, 2020-02-17:
@DiedB see the screenshot below:

[image]

Would that suffice?

DiedB, 2020-02-17:
Great, absolutely!

dennissiemensma, 2020-02-17:
Added to v3.4 release.

closed 2020-02-17 by dennissiemensma

dennissiemensma, 2020-02-20:
Uitgebracht in [v3.4](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.4.0)

Helloitsme, 2020-02-22:
1 klein puntje: bij mij gaat de uitlijning van de energie teruggeleverd grafiek niet goed (onder archive)
de staaf van de eerste dag overlapt de Y as

[image]

dennissiemensma, 2020-02-22:
Bedankt voor je melding! Ik zag het zelf ook al en neem het mee voor de volgende release. 

Als je het handmatig wil fixen kun je dit doen:
```
vi /var/www/dsmrreader/static/dsmr_frontend/js/dsmr-reader/archive/electricity-returned.js
```

Rond regel 42 staat iets als:
```
        xAxis: [
            {
                type : 'category',
                boundaryGap: false,
                data : xhr_data.x
            }
        ],
```

Pas deze aan:
```
boundaryGap: false,
# wordt
boundaryGap: true,
```
Je wijziging wordt overigens automatisch overschreven bij de volgende DSMR-reader update die je doet.