dsmrreader/dsmr-reader #884: Datalogger probleem

VBP8501, 2020-02-17:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v3.3`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Hallo,

Ik had vorige week een probleem met de sd kaart van mijn raspberry pi. Daarop een nieuwe SD kaart genomen, raspbian terug geïnstalleerd samen met dsmr-reader, SBFspot en mqtt mosquito broker. Tijdens de installatie heb ik de postgres databank hersteld met een backup zoals beschreven in de documentatie. 
Voorheen geen probleem om P1 poort uit te lezen. Het cu commando geeft ook de telegrammen weer. Maar de datalogger werkt wel niet. Supervisor lanceert de datalogger en geeft geen error. 

Hieronder meer informatie

Geen errors bij 
`./deploy.sh`

DEBUG aangezet maar log blijft leeg
```
tail -100 /var/log/supervisor/dsmr_datalogger.log
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
```

DSMR zit op laatste versie
```
cat dsmrreader/__init__.py
from django.utils.version import get_version

VERSION = (3, 3, 0, 'final', 0)
```

Geen problemen met git
```
git status
On branch v3
Your branch is up to date with 'origin/v3'.
```

Python scripts zonder syntax errors
```
./manage.py --version
3.0.3
```

Hier wel een error:
```
python dsmrreader/settings.py
Traceback (most recent call last):
  File "dsmrreader/settings.py", line 3, in <module>
    from dsmrreader.config.production import *
ModuleNotFoundError: No module named 'dsmrreader'
```

Veel errors hier
```
sudo ps faux | grep datalogger | grep -v grep
dsmr      5316 99.4  5.3  64820 50528 ?        R    08:33  11:23  \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger
pi@Logger:~ $ sudo strace -fv -p 5316

strace: Process 5316 attached
getpid()                                = 5316
stat64("/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", {st_dev=makedev(0xb3, 0x2), st_ino=651251, st_mode=S_IFREG|0644, st_nlink=1, st_uid=1001, st_gid=1001, st_blksize=4096, st_blocks=32, st_size=12350, st_atime=1581890215 /* 2020-02-16T22:56:55.123027856+0100 */, st_atime_nsec=123027856, st_mtime=1581890215 /* 2020-02-16T22:56:55.123027856+0100 */, st_mtime_nsec=123027856, st_ctime=1581890215 /* 2020-02-16T22:56:55.123027856+0100 */, st_ctime_nsec=123027856}) = 0
write(2, "[2020-02-17 08:45:25,488] ERROR "..., 224) = 224
write(2, "ERROR:commands:\nTraceback (most "..., 204) = 204
getpid()                                = 5316
write(2, "[2020-02-17 08:45:25,495] DEBUG "..., 102) = 102
write(2, "DEBUG:commands:dsmr_datalogger.m"..., 82) = 82
_newselect(0, NULL, NULL, NULL, {tv_sec=0, tv_usec=0}) = 0 (Timeout)
getpid()                                = 5316
```

Datalogger met sudo aanroepen werkt wel:
```
pi@Logger:~ $ sudo /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger
[2020-02-17 09:02:05,866] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Starting infinite command loop...
INFO:dsmrreader:Received telegram (base64 encoded): b'L0ZMVTVcMjUzNzY5NDg0X0ENCg0KMC0wOjk2LjEuNCg1MDIxMykNCjAtMDo5Ni4xLjEoMzE1MzQxNDczMzMxMzAzMDMwMzIzMzM1MzQzOCkNCjAtMDoxLjAuMCgyMDAyMTcwOTAyNDNXKQ0KMS0wOjEuOC4xKDAwMTA0My40Mzkqa1doKQ0KMS0wOjEuOC4yKDAwMTMzNy42NjMqa1doKQ0KMS0wOjIuOC4xKDAwMDExOS4xODkqa1doKQ0KMS0wOjIuOC4yKDAwMDAxOS40Mzcqa1doKQ0KMC0wOjk2LjE0LjAoMDAwMSkNCjEtMDoxLjcuMCgwMC4xNTkqa1cpDQoxLTA6Mi43LjAoMDAuMDAwKmtXKQ0KMS0wOjMyLjcuMCgyMzIuNSpWKQ0KMS0wOjUyLjcuMCgwMDAuMCpWKQ0KMS0wOjcyLjcuMCgyMzMuMCpWKQ0KMS0wOjMxLjcuMCgwMDIqQSkNCjEtMDo1MS43LjAoMDAxKkEpDQoxLTA6NzEuNy4wKDAwMypBKQ0KMC0wOjk2LjMuMTAoMSkNCjAtMDoxNy4wLjAoOTk5Ljkqa1cpDQoxLTA6MzEuNC4wKDk5OSpBKQ0KMC0wOjk2LjEzLjAoKQ0KIUUyQjcNCg=='
[2020-02-17 09:02:07,147] DEBUG    MQTT: Queued message for sensors/dsmr/json
DEBUG:commands:MQTT: Queued message for sensors/dsmr/json
[2020-02-17 09:02:07,265] DEBUG    MQTT: Queued message for sensors/dsmr/day-consumption
DEBUG:commands:MQTT: Queued message for sensors/dsmr/day-consumption
[2020-02-17 09:02:07,293] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 0.5s
DEBUG:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 0.5s
```

Kan het zijn dat er ergens een permissie probleem is? Hoe spoor ik deze op en hoe verhelp ik ze?

dennissiemensma, 2020-02-17:
Bedankt voor je melding. 

- `python dsmrreader/settings.py` kun je niet zomaar uitvoeren, dus die fout klopt.

- Ik zie in je `strace` dump wel een fout: `...write(2, "[2020-02-17 08:45:25,488] ERROR "..., 224) = 224...`. Kun je eens proberen om `-s` mee te geven om de lengte van de output strings te verhogen? Wellicht zie je dan de volledige foutmelding: `sudo strace -fv -s 512 -p <PID>`

- Heb je bij de (her)installatie wel de `dsmr` user toegevoegd aan de `dialout` groep? `sudo usermod -a -G dialout dsmr`

VBP8501, 2020-02-17:
> * Heb je bij de (her)installatie wel de `dsmr` user toegevoegd aan de `dialout` groep? `sudo usermod -a -G dialout dsmr`

... Nu werkt het...

Sorry ik schaam mij.. Zelfs met zo'n geweldige documentatie en super user support ben ik nog 1 regel vergeten uitvoeren (en nadien waarschijnlijk 5 keer erover gekeken). Ik wou dat ik kon zeggen dat ik de regel had uitgevoerd en het door hokus pokus toch niet had gepakt maar .bash_history spreekt mij tegen..

Heel hartelijk bedankt voor je tijd! (en voor het geweldige programma!)

closed 2020-02-17 by VBP8501

dennissiemensma, 2020-02-17:
Fijn dat dat het heeft opgelost. Je hoeft je nergens voor te schamen. Zolang het installeren handwerk is, blijft het foutgevoelig ;-)