dsmrreader/dsmr-reader #885: Herindeling dashboard

dennissiemensma, 2020-02-18:
Het dashboard is een mooie pagina met een hoop informatie, maar ik wil hem proberen op te splitsen. Op het dashboard komt dan alleen het verbruik van de huidige dag + maand (is nu nog ingeklapt) te staan.

Op een aparte pagina komen alle 'live' grafieken. Ook omdat er steeds meer en meer grafieken bij komen.

dennissiemensma, 2020-02-18:
Opzet klaar. Het zal even wennen zijn, maar ik denk dat het uiteindelijk wel een stuk logischer is zo. 

De teller rechtsbovenin het dashboard staat zowel op de oude als nieuwe pagina. En het dashboard heeft een knop om direct naar de grafieken te gaan.

closed 2020-02-18 by dennissiemensma

dennissiemensma, 2020-02-20:
Uitgebracht in [v3.4](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.4.0)

Mixpower, 2020-02-21:
> Uitgebracht in [v3.4](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.4.0)

Is er een mogelijkheid om dit weer terug te zetten?
Ik vond het oude overzicht namelijk veel handiger omdat alles op één plek meteen duidelijk te zien is, indien mogelijk graag een mogelijkheid om zelf de indeling te kiezen.

dennissiemensma, 2020-02-22:
Bedankt voor je terugkoppeling. Ik begrijp dat ieder zijn voorkeur heeft, alleen is het ook goed om te weten dat er op termijn meer grafieken toegevoegd worden aan de nieuwe pagina. En voor het dashboard komt er op termijn ook meer informatie, zoals trends of vergelijkingen met vorige dagen/maanden. Om het overzichtelijk te houden is het gesplitst. Mijn advies is om het vooral even aan te kijken.

Mixpower, 2020-02-22:
Snap ik, alleen het kosten overzicht samen met de live grafieken vond ik persoonlijk heel fijn.
Zoals ik al zei misschien is het een optie om het dashboard aanpasbaar te maken zodat je zelf door middel van widgets het naar eigen voorkeur in te delen.