dsmrreader/dsmr-reader #886: Upgrade naar v3.3 geeft fout

williamheuts, 2020-02-18:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v3.3`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Ik ontvang de volgende fout na een upgrade naar versie 3.3
---
Running migrations:
  Applying dsmr_backend.0013_allow_higher_backend_sleep...Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 86, in _execute
    return self.cursor.execute(sql, params)
psycopg2.errors.UniqueViolation: duplicate key value violates unique constraint "django_migrations_pkey"
DETAIL:  Key (id)=(79) already exists.
---

 --- Reloading running apps...

 * Reloading process: dsmr_webinterface (Gunicorn)   [OK]
 * Reloading process: dsmr_backend                   [OK]
 * Reloading process: dsmr_datalogger                [OK]
 * Reloading process: dsmr_mqtt                      [??] PID file does not exist
---

Gevolg is dat de webinterface niet werkt. Admin page werkt wel naar behoren.

Server Error
Sorry, something unexpected happened.
Exception:

ProgrammingError: column dsmr_frontend_frontendsettings.gas_graph_style does not exist
LINE 1: ...ontend_frontendsettings"."dashboard_graph_width", "dsmr_fron...

Ik begrijp niet goed wat hier is gebeurt. Gaarne enige hulp om de juiste fout te identificeren en op te lossen.

dennissiemensma, 2020-02-18:
Bedankt voor je melding, wil je dit eens proberen:

```
sudo su - postgres
psql dsmrreader

BEGIN;
SELECT setval(pg_get_serial_sequence('"django_migrations"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "django_migrations";
COMMIT;
```

Daarna twee keer `CTR+D` (logout) doen en:
```
sudo su - dsmr
./deploy.sh
```

williamheuts, 2020-02-19:
Dat deed het hem. Bedankt!

closed 2020-02-19 by dennissiemensma