dsmrreader/dsmr-reader #888: MQTT waardes veel te laag

sanderpleijers, 2020-02-19:
**DSMR Reader via Docker**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v3.3`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `Synology Docker`
* DSMR-protocol slimme meter): `v2`

MQTT waardes kloppen niet/ zijn niet gevuld:

day-consumption:
[image]

Terwijl DSMR zelf aangeeft in de GUI:
[image]

Configuratie MQTT Dagverbruik gesplitst topic (standaard waardes!):

```
[mapping]
# DATA = JSON FIELD
electricity1 = dsmr/day-consumption/electricity1
electricity2 = dsmr/day-consumption/electricity2
electricity1_returned = dsmr/day-consumption/electricity1_returned
electricity2_returned = dsmr/day-consumption/electricity2_returned
electricity_merged = dsmr/day-consumption/electricity_merged
electricity_returned_merged = dsmr/day-consumption/electricity_returned_merged
electricity1_cost = dsmr/day-consumption/electricity1_cost
electricity2_cost = dsmr/day-consumption/electricity2_cost
electricity_cost_merged = dsmr/day-consumption/electricity_cost_merged

# Gas (if any)
gas = dsmr/day-consumption/gas
gas_cost = dsmr/day-consumption/gas_cost
total_cost = dsmr/day-consumption/total_cost

# Your energy supplier prices (if set)
energy_supplier_price_electricity_delivered_1 = dsmr/day-consumption/energy_supplier_price_electricity_delivered_1
energy_supplier_price_electricity_delivered_2 = dsmr/day-consumption/energy_supplier_price_electricity_delivered_2
energy_supplier_price_electricity_returned_1 = dsmr/day-consumption/energy_supplier_price_electricity_returned_1
energy_supplier_price_electricity_returned_2 = dsmr/day-consumption/energy_supplier_price_electricity_returned_2
energy_supplier_price_gas = dsmr/day-consumption/energy_supplier_price_gas
```

Ingevulde tarieven (klopt dit zo wel?):
[image]

Enig idee waarom waardes zoals gas, gas_cost, total_cost niet ingevuld zijn (óf 0.000)?

dennissiemensma, 2020-02-19:
Bedankt voor je melding. Ik heb het zelf even geprobeerd, maar kan het helaas niet reproduceren. 

Er kunnen meerdere oorzaken zijn, maar het belangrijkste is denk ik om eerst te checken wat de broker ruw ontvangt. Ben je in de gelegenheid om dit op de commandline te checken met `mosquitto_sub`?

Voor een broker op localhost kun je dit doen, je ziet dan met `'#'` alle topics. En anders kun je de hostname of IP invullen ipv `localhost`.
```
mosquitto_sub -v -h localhost -p 1883 -t '#'
```

sanderpleijers, 2020-02-26:
Probleem lijkt opgelost door dsmr topic te verwijderen op de broker en te updaten naar 3.4.

closed 2020-02-26 by sanderpleijers

dennissiemensma, 2020-02-26:
Bedankt voor je update, mooi dat dat het oplost!