dsmrreader/dsmr-reader #890: elektriciteit teruggaven zichtbaar ondanks niet fysiek aanwezig.

BlackDex, 2020-02-19:
* DSMR-reader versie: `v3.3`
* Hardware: `Raspberry Pi 2 Model B Rev 1.1`
* DSMR-protocol slimme meter: `v2/3`

Er worden elektriciteit teruggaven gemeten, maar die heb ik niet.
Is er een manier om dit uit te schakelen?
Hierdoor zie ik tegenwoordig een groen statistiek box welke nooit iets aan geeft.

Tevens is dit ook zichtbaar bij de archieven.

dennissiemensma, 2020-02-19:
Je kunt dit geforceerd verbergen in de Backend-configuratie:

- `Disable electricity returned capability`

[image]

closed 2020-02-19 by dennissiemensma

BlackDex, 2020-02-20:
Mijn excuses voor deze vraag.
Ik heb die optie volledig over het hoofd gezien.

dennissiemensma, 2020-02-20:
Geen probleem joh! Met het oerwoud aan opties en instellingen moet ik soms zelf ook even zoeken waar wat zit. :]