dsmrreader/dsmr-reader #892: DSMR stopt met ontvangen van data

coen17st, 2020-02-20:
Hi,

Ik ben al een paar weken bezig met een probleem. Op een random moment stopt DSMR-Reader met het verkrijgen of verwerken van data. Zie ook status pagina afbeelding in deze link https://ibb.co/kqSyDw2

Ik heb DSMR-Reader draaien op:
OS host: **Windows Server 2019 64bit** -> **Virtualbox** -> OS guest: **ubuntu-18.04.3-live-server-amd64** 
Ook hetzelfde probleem in Debian OS

Op het moment dat er geen data meer verwerkt/ontvangen wordt check ik d.m.v. `sudo supervisorctl status` en is alles dan op dat moment gewoon RUNNING

- DSMR_BACKEND = RUNNING
- DSMR_DATALOGGER = RUNNING
- DSMR_MQTT = RUNNING
- DSMR_WEBINTERFACE = RUNNING

Vervolgens doe ik `sudo supervisorctl restart dsmr_datalogger` en alles doet het hierna weer normaal.

De `LOGGING['loggers']['commands']['level'] = 'DEBUG'` heb ik aangezet en ben door diversen log files heen gegaan maar ik weet niet zo goed waar ik naar moet kijken en kom voor zo ver ik weet geen errors tegen waar ik iets mee kan.

Ik heb MQTT in verbinding staan met Homeassistant, daarin een MQTT Broker Addon geinstalleerd,

Hopelijk kan iemand me hiermee helpen....

dennissiemensma, 2020-02-20:
Bedankt voor je melding. Het is belangrijk om te vermelden welke versie van DSMR-reader je gebruikt.

- Sinds vanavond is er ook weer een nieuwe versie (v3.4) uit, die zou je sowieso ook nog kunnen proberen.

- Verder lijken je issues wat op #848, hoewel de host daar MacOS is en niet Windows. Je zou daar nog even doorheen kunnen lezen. Wellicht is het een issue door Virtualbox, maar daar kwamen we er ook niet exact achter.

- Qua logs zou je kunnen kijken of er toch nog iets in de datalogger-logs staat. Ook zonder DEBUG-level zouden fouten gelogd moeten worden.
```
sudo tail -n 50 /var/log/supervisor/dsmr_datalogger*
```

coen17st, 2020-02-21:
Dank voor je super snelle reactie!. Ik heb DSMR-Reader gisteravond meteen geupdate en is nu al bijna 24uur online zonder problemen, zolang is het niet niet eerder gelukt en lijkt het dus opgeslost te zijn :)

dennissiemensma, 2020-02-22:
Fijn om te horen, bedankt voor je terugkoppeling. 

Vergeet overigens niet om `LOGGING['loggers']['commands']['level'] = 'DEBUG'` weer uit te zetten als je n.a.v. die hierboven aangezet had.

closed 2020-02-22 by dennissiemensma