dsmrreader/dsmr-reader #893: Backup van oudere versie importeren

andrew040, 2020-02-20:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3,3
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 4
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: v5

Ik had een Pi met dsmrreader v2.5rc1 draaien, maar deze liep steeds vaker vast en was op een gegeven moment wilde dsmrreader enkel nog het dashboard tonen. De status pagina opvragen leverde een timeout op in nginx. Ik heb daarom een nieuwe Pi vers voorzien van v3.3, maar de database van v2.5rc1 is niet compatible daarmee. In v3.3 zijn wat functies (en dus kolommen) toegevoegd waardoor de import mislukt.

Ik heb nu dus een v3.3 database en een v2.5rc1 backup. Hoe kan ik deze twee alsnog verenigen?

dennissiemensma, 2020-02-20:
Bedankt voor je vraag. Kun je aangeven wat er exact misgaat en welke foutmeldingen je ziet?

Je geeft aan dat de import mislukt, maar die kan bijna nooit mislukken (mits de backup zelf goed is), de applicatie crasht hooguit. Kan het zijn dat je bedoelt dat DSMR-reader zelf niet meer goed draait? Of mislukt de database-import daadwerkelijk?

De applicatie houdt zelf in de database bij welke 'versie' het is en nieuwere installaties weten daardoor altijd wat er moet gebeuren om de database bij te werken (mits je geen oudere versie van DSMR-reader v1.5 gebruikt). Bij het uitvoeren van de deploy.sh, na de database-import, zou dit automatisch moeten gebeuren. 
Zo niet, dan kun je het handmatig doorvoeren. Dit zou alle migraties moeten uitvoeren:
```
sudo su - dsmr
./manage.py migrate
```

Daarna kun je kijken of alle migraties zijn uitgevoerd en een `[X]` ervoor hebben staan.

```
./manage.py showmigrations
```

dennissiemensma, 2020-02-20:
Het is trouwens belangrijk dat je de import doet op een lege database en niet op een database die al aangemaakt is bij de installatie.

andrew040, 2020-02-20:
Allereerst bedankt voor je snelle reactie.

De import mislukte in eerste instantie omdat ik in de backup probeerde te importeren in een reeds aangemaakte - doch niet met meetgegevens gevulde database. In dat geval komen er vrijwel alleen maar foutmeldingen voorbij en wordt er niet daadwerkelijk data in de database weggeschreven.

Nu ik er nog eens over nadenk is deze na installatie inderdaad niet écht leeg. Vervolgens de database gedropt en een nieuwe aangemaakt (d.w.z. zonder enige tabel), waarna de import wel slaagde. Ik was er niet van op de hoogte dat er een migrate script was, maar nu ik dit gedraaid heb lijkt het erop alsof de applicatie weer draait en de backupdata is zichtbaar!

dennissiemensma, 2020-02-22:
Top, fijn om te horen!

closed 2020-02-22 by dennissiemensma