dsmrreader/dsmr-reader #894: Onregelmatig uitlezen P1 en USB input overrun

bikeymouse, 2020-02-20:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.4
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 2b en 3b
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Docker
* DSMR-protocol slimme meter *(v4 of v5)*: v5.

Ik heb DSMR-reader in een Docker container draaien op een RPI 3b waar ook andere containers op draaien. Sinds kort heb ik echter regelmatig (iedere < 24 uur) dat mijn hele Raspberry niet meer te bereiken is. Alleen herstarten helpt dan.

Bij het bekijken van de syslog kwam ik daarin heel veel van de volgende berichten tegen:
kernel: [361634.417358] ttyUSB ttyUSB0: 5 input overrun(s)

Dat bracht mij er toe de DSMR-backend wat beter te bekijken, en ik zag daar nu dat de Telegrammen heel onregelmatig binnenkomen.

Ik heb de datalogger "proces sleep" ingesteld op 60 seconden en de back-end process sleep op 30 seconden maar zie vaak in de front-end dat het 5 minuten duurt sinds de laatste Telegram is ontvangen of verwerkt. 

Als ik bij de DSMR-uitlezing in de backend kijk zie ik inderdaad soms telegrammen met 5 minuten tussenpauze, maar gek genoeg ook dat er in 1 minuut 5 uitlezingen zijn gedaan, zie deze [screenshot](https://ibb.co/8XBmp8x).

Dit gebeurde initieel op een RPI3 met DSM3.3 maar om te testen heb ik de P1 kabel op de RPI2b aangesloten en daar de Docker container opgestart (en daarmee ook een updgrade naar DSMR 3.4) en daar doet zich hetzelfde probleem voor.

Het lijkt dus niet aan de Raspberry of OS-omgeving te liggen, dus blijft over DSMR of de P1 kabel die dan spontaan niet meer goed werkt?

Enig idee waardoor de "input overrun" errors van komen? En zou dit kunnen samenhangen met het andere probleem wat hier gemeld is m.b.t. seriële poort?

Verder weet ik niet of het er relatie is, maar het lijkt ook dat de netwerkverbinding van de Raspberry Pi er (mogelijk door de problemen met de seriële poort?) uitklapt, want die kan ik dan dus niet niet meer bereiken via het netwerk (en hij is head-less helaas). 
Soms kom ik er dan pas na een paar uur achter, maar dan blijkt hij niet echt te zijn gecrasht  gezeten, want DSM heeft dan nog wel data verzameld gedurende die periode. Heel vreemd.

dennissiemensma, 2020-02-22:
Bedankt voor je melding. 

- Dit soort dingen zijn lastig te debuggen, maar als ik het zo lees dan is het goed om te beginnen met DEBUG-logging aan te zetten: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging
- Herstart de datalogger 
- Doe daarna `sudo tail -f /var/log/supervisor/dsmr_datalogger*`

Ik ben benieuwd naar de output van de 'sleep' daar. Bij mij staat er iets als:
```
[2020-02-22 11:31:27,947] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 0.5s
DEBUG:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 0.5s
```

Kun je checken of dat bij jou op 60 staat? Gezien je aangeeft je toch meerdere telegrammen per minuut ontvangt. 
Het kan zijn dat de naam van de logfile wat willekeurig is, bij mij is het bijvoorbeeld: `/var/log/supervisor/dsmr_datalogger-stderr---supervisor-fBrLO4.log`

bikeymouse, 2020-02-22:
Hi, dank voor je reactie.
Ik gebruik de Docker versie van xirixiz en heb daar in de draaiende container eerst in settings.py
de logger un-comment (de "#" weggehaald en level op 'DEBUG').

Daarna de container opnieuw gestart. Probleem is alleen dat ik in /desmr/var/log alleen een nginx directory heb staan en verder niets.

Ik heb wel een /dsmr/logs directory met een django.log en dsmrreader.log.
Die laatste bevat ongeveer 1 x per uur de volgende melding:

```
[2020-02-22 14:12:45,201] WARNING  @ services | Rejected telegram (Invalid telegram CRC. The calculated checksum '12646' (3166) does not match the telegram checksum '18516' (4854)) (base64 encoded): b'L0lTSzVcMk01NTBULTEwMTINCg0KMS0zOjAuMi44KDUwKQ0KMC0wOjEuMC4wKDIwMDIyMjE0MDY0NVcpDQowLTA6OTYuMS4xKDQ1MzAzMDM0MzQzMDMwMzczMjM1MzEzOTM1MzczNjMxMzgpDQoxLTA6MS44LjEoMDA1ODM3LjIyNiprV2gpDQoxLTA6MS44LjIoMDAxOTIwLjc0MyprV2gpDQoxLTA6Mi44LjEoMDAwMDg3LjgxMyprV2gpDQoxLTA6Mi44LjIoMDAwMjgxLjkyOCprV2gpDQowLTA6OTYuMTQuMCgwMDAxKQ0KMS0wOjEuNy4wKDAxMC4wMDAqa1cpDQowLTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzMzOTMwMzAzMTM4MzQzNDM5MzQzOTM1MzEzOCkNCjAtMToyNC4yLjEoMjAwMjIyMTQxMDAyVykoMDE0MzMuODM4Km0zKQ0KITQ4NTQNCg=='
[2020-02-22 14:12:45,210] ERROR    @ services | Invalid telegram CRC. The calculated checksum '12646' (3166) does not match the telegram checksum '18516' (4854)
Traceback (most recent call last):
  File "/dsmr/dsmr_datalogger/services.py", line 104, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/dsmr/dsmr_parser/parsers.py", line 42, in parse
    self.validate_checksum(telegram_data)
  File "/dsmr/dsmr_parser/parsers.py", line 82, in validate_checksum
    raise InvalidChecksumError(
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '12646' (3166) does not match the telegram checksum '18516' (4854)
```

Op zich niet schokkend 1 x per uur een CRC-fout, alhoewel de foutafhandeling wel erg grof lijkt(?).

Als ik de algemene log van de container bekijk (via Portainer) dan zie ik alleen de reguliere info van de start-up van de container, helaas geen debug-info:

```
[ INFO ] Checking for HTTP AUTHENTICATION configuration...

[ INFO ] ENABLE_HTTP_AUTH is disabled, nothing to see here.

[ INFO ] Starting supervisord...

2020-02-22 14:17:15,654 INFO Included extra file "/etc/supervisor.d/supervisord.ini" during parsing

2020-02-22 14:17:15,656 INFO Set uid to user 0 succeeded

2020-02-22 14:17:16,030 INFO RPC interface 'supervisor' initialized

2020-02-22 14:17:16,032 CRIT Server 'unix_http_server' running without any HTTP authentication checking

2020-02-22 14:17:16,036 INFO supervisord started with pid 18

2020-02-22 14:17:17,050 INFO spawned: 'dsmr_backend' with pid 20

2020-02-22 14:17:17,066 INFO spawned: 'dsmr_datalogger' with pid 21

2020-02-22 14:17:17,086 INFO spawned: 'dsmr_webinterface' with pid 22

2020-02-22 14:17:17,101 INFO spawned: 'nginx' with pid 23

2020-02-22 14:17:18,107 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)

2020-02-22 14:17:18,108 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)

2020-02-22 14:17:18,110 INFO success: nginx entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)

2020-02-22 14:17:20,537 INFO success: dsmr_webinterface entered RUNNING state, process has stayed up for > than 3 seconds (startsecs)
```
Vraag is dus eigenlijk hoe ik uit de docker-container versie extra logging krijg.

dennissiemensma, 2020-02-22:
Ik denk dat je die vraag het beste bij xirixiz kunt leggen. Mijn Docker-kennis is nog vrij beperkt.

bikeymouse, 2020-02-23:
Hi,
Ik ben inmiddels wat verder, want heb met environment variabele de logging aangekregen.

Ik zie inderdaad netjes een bevestiging dat de datalogger-sleep op 60 seconden staat:

```
[2020-02-23 16:08:43,385] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 60.0s

DEBUG:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 60.0s
```
(opvallend: 2 meldingen...?)

En de backend op 30 seconden:
`[2020-02-23 16:09:01,725] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 30.0s`

Vervolgens ook heel netjes iedere minuut melding van ontvangst van een telegram:

```
[2020-02-23 16:01:40,282] DEBUG    Received telegram:
DEBUG:commands:Received telegram:
[2020-02-23 16:02:40,689] DEBUG    Received telegram:
DEBUG:commands:Received telegram:
[2020-02-23 16:03:41,129] DEBUG    Received telegram:
DEBUG:commands:Received telegram:
[2020-02-23 16:04:41,717] DEBUG    Received telegram:
DEBUG:commands:Received telegram:
[2020-02-23 16:05:42,120] DEBUG    Received telegram:
DEBUG:commands:Received telegram:
[2020-02-23 16:06:42,478] DEBUG    Received telegram:
DEBUG:commands:Received telegram:
[2020-02-23 16:07:42,846] DEBUG    Received telegram:
DEBUG:commands:Received telegram:
[2020-02-23 16:08:43,184] DEBUG    Received telegram:
DEBUG:commands:Received telegram:
```

Echter wel vreemd is dat ieder telegram 2 x wordt gelogd, 1 x met timestamp en 1 x zonder.
Zou er een dubbel datalogger proces draaiien?

Verder als ik dan vervolgens in de database kijk in deze periode 16:01 - 16:08 uur zou ik verwachten ook netjes iedere minuut een telegram gelogd te zien, echter blijken ze samengevoegd op andere tijdstippen:

[image]

bikeymouse, 2020-02-23:
Ok nog wat info, ik heb even een telegram uit de log opgezocht in de backend. Dan zie je dat deze inderdaad met een andere timestamp wordt opgeslagen:
Telegram uit de log met timestamp 16:57:02:
```
2020-02-23 16:57:02,402] DEBUG    Received telegram:
/ISK5\2M550T-1012
1-3:0.2.8(50)
0-0:1.0.0(200223165359W)
0-0:96.1.1(4530303434303037323531393537363138)
0:1.8.1(005851.454*kWh)
0:1.8.2(001920.743*kWh)
0:2.8.1(000088.837*kWh)
0:2.8.2(000281.928*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.170*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00006)
0-0:96.7.9(00004)
1-0:99.97.0(2)(0-0:96.7.19)(190614125106S)(0000003280*s)(190909000512S)(0000003281*s)
1-0:32.32.0(00006)
1-0:52.32.0(00017)
1-0:72.32.0(00012)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(227.1*V)
1-0:52.7.0(223.4*V)
1-0:72.7.0(223.1*V)
1-0:31.7.0(000*A)
1-0:51.7.0(001*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.000*kW)
1-0:41.7.0(00.131*kW)
1-0:61.7.0(00.124*kW)
1-0:22.7.0(00.082*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303339303031383434393439353138)
0-1:24.2.1(200223165005W)(01442.930*m3)
!7645
```
Info in de back-end heeft timestamp 16:53:
[image]

dennissiemensma, 2020-02-23:
Bedankt voor de update. Wat betreft de logging, het kan zijn dat er twee loggers door elkaar heen lopen of dat dezelfde logger dubbel schrijft. Wat betreft de tijdstippen, wellicht dat de slimme meter niet helemaal synchroon loopt qua klok. 

Wat heb je overigens ingesteld in de admin voor het groeperen van metingen? Zie `› Dsmr_consumption › Consumption configuration`.

bikeymouse, 2020-02-23:
Ik heb even gekeken, ik had voor het groeperen van de metingen ingesteld dat dit per minuut moest zijn. Ik heb dat nu aangepast naar "per meting". 
Dat lijkt echter (ook na herstart) geen verschil te maken, want ik zie nog steeds onregelmatige metingen in de back-end met meerdere metingen per minuut en dan soms weer 4-5 minuten niets.
En ook al loopt de slimme meter niet helemaal synchroon qua klok, ik zie de metingen in de log wel netjes met 1 per minuut binnenkomen. Dan is het toch gek dat dit niet zo wordt opgeslagen?
(en dan heb ik natuurlijk nog steeds de "USB overrun" meldingen in syslog).

dennissiemensma, 2020-02-24:
Ik denk dat je wellicht even in de log van de dsmr_backend moet kijken wat die exact doet. Als je DEBUG-logging nog aan hebt staan, zou er uitgebreid moeten staan wat die doet. 

Wat vooral interessant is is wanneer die metingen verwerkt wordt: `Compact: Processed reading: ...`
En dan met name de tijdstippen/intervallen ertussen.

De reden dat ik vroeg naar je groepering, is dat groeperen per minuut ervoor zorgt dat de applicatie wacht tot zowel de 'minuut' van de onverwerkte metingen voorbij is, als dat er nieuwe metingen voor de volgende minuut zijn. De melding is dan iets als: `Compact: Waiting for newer readings before grouping data...`. Wellicht dat dat nog iets vreemds deed met de tijd die uit sync is, maar dat lijkt dus niets uit te maken.

dennissiemensma, 2020-02-24:
Ik kan opzich wel een branch voor je maken met wat meer debug-logging, maar ik weet niet hoe je die makkelijk kan wisselen als je Docker draait. Ik vermoed dat de release namelijk hard in de image gebakken is. Wat voor productie prima is, maar debuggen wat lastig maakt.

bikeymouse, 2020-02-24:
Hi, dank voor de nieuwe tips om het verder uit te zoeken.
Als ik in de log filter op "compact", dan krijg ik het volgende overzicht:

```
[2020-02-24 20:40:41,267] DEBUG    Compact: Processed reading: 637381 @ 2020-02-24 20:37:22+01:00
[2020-02-24 20:41:43,164] DEBUG    Compact: Processed reading: 637382 @ 2020-02-24 20:37:23+01:00
[2020-02-24 20:42:44,960] DEBUG    Compact: Processed reading: 637383 @ 2020-02-24 20:37:24+01:00
[2020-02-24 20:43:46,816] DEBUG    Compact: Processed reading: 637384 @ 2020-02-24 20:42:22+01:00
[2020-02-24 20:44:48,786] DEBUG    Compact: Processed reading: 637385 @ 2020-02-24 20:42:23+01:00
[2020-02-24 20:45:50,576] DEBUG    Compact: Processed reading: 637386 @ 2020-02-24 20:42:24+01:00
[2020-02-24 20:46:52,390] DEBUG    Compact: Processed reading: 637387 @ 2020-02-24 20:42:25+01:00
[2020-02-24 20:47:54,146] DEBUG    Compact: Processed reading: 637388 @ 2020-02-24 20:42:26+01:0
[2020-02-24 20:48:55,877] DEBUG    Compact: Processed reading: 637389 @ 2020-02-24 20:47:24+01:00
[2020-02-24 20:49:57,574] DEBUG    Compact: Processed reading: 637390 @ 2020-02-24 20:47:25+01:00
[2020-02-24 20:50:59,513] DEBUG    Compact: Processed reading: 637391 @ 2020-02-24 20:47:26+01:00
[2020-02-24 20:51:30,436] DEBUG    Compact: Processed reading: 637392 @ 2020-02-24 20:47:27+01:00
etc.
```
Allemaal dus netjes iedere minuut. Hoewel ik de back-end process sleep op 30 seconden heb gezet (wat ook uit de log zichtbaar blijkt, maar ik weet niet of dat tegenstrijdig is.

bikeymouse, 2020-02-24:
> Ik kan opzich wel een branch voor je maken met wat meer debug-logging, maar ik weet niet hoe je die makkelijk kan wisselen als je Docker draait. Ik vermoed dat de release namelijk hard in de image gebakken is. Wat voor productie prima is, maar debuggen wat lastig maakt.

Dat is niet zo makkelijk wisselen nee, helaas.
Ik heb het idee dat het dus in de USB-serial verbinding ligt, want daar zijn dus fouten met buffer-overruns.

dennissiemensma, 2020-02-24:
Heb je trouwens die datalogger-sleep altijd zo hoog gehad? Zo ja, dan kun je eens proberen om te kijken wat er gebeurt als je die juist heel laag zet, op `0` of `0,5` of `1`. Wie weet gaat dat juist die overruns tegen omdat de buffer dan sneller wordt 'leeg' gelezen.

bikeymouse, 2020-02-24:
Ja die heb ik altijd zo hoog gehad. Gedachte was: ik hoef maar 1 x per minuut mijn meter uit te lezen, dan mag het verwerken ook wel 1 minuut duren.
Ik nog even getest met het naar 1 seconde zetten van de datalogger-sleep, maar dat maakt voor de USB buffer overruns niet uit. 
Vervolgens ook de datalogger op 1 seconde gezet. Gevolg: nu kreeg ik helemaal veel bufferoverruns, zo'n beetje iedere seconde...
Begin toch het idee te krijgen dat de kabel misschien niet meer goed is, want het lijkt er op dat de leesactie van de port op zich dus de buffer-overruns geneert.

dennissiemensma, 2020-02-24:
Wat je sowieso nog kunt doen is even gedurende 5 a 10 minuten handmatig de kabel uitlezen met `cu`. En dan kijken of je nog nieuwe buffer-overruns krijgt.

Ik vermoed dat het een software-matig probleem is en geen hardware. Maar dat is natuurlijk een aanname. Helaas heb ik hier geen DSMR-v5 meter om het te kunnen reproduceren.

bikeymouse, 2020-02-24:
Ik denk dat je gelijk hebt. Ik cu even 5 minuten aangezet en had 0 fouten terwijl er meer dan § telegram per seconde voorbijkwam.
Dank voor het meedenken, trouwens!

dennissiemensma, 2020-02-25:
Wil je eens proberen of het nog uitmaakt of je de datalogger op `0` of `0,5` seconde sleep zet? 

Ik lees namelijk wat over dat de overruns voort kunnen komen uit dat software de buffer te weinig/traag uitleest. Het verlagen van de sleep kan het verhelpen, het is de poging waard. https://www.linuxquestions.org/questions/linux-kernel-70/tty-usb-input-overruns-4175512707/#post5211813

bikeymouse, 2020-02-25:
Ha Dennis,

Dank voor de tip. Ik heb het even een half uur geprobeerd en inderdaad ik heb dan geen overruns meer. 

Alleen genereert dit dan wel 60+ telegrammen per minuut en in de database en een load van 40% op de Raspberry Pi. Dat is wel heel erg overdreven als je alleen gewoon wat inzicht in je verbruik wilt hebben, toch?

Ik gebruik DSMR nu een jaar en heb het hiervoor altijd ook met 1 meting per minuut gebruikt en dat ging toch altijd goed. En nu dus overruns (die er mogelijk vroeger ook waren), maar vooral heel onregelmatige uitlezing (wat vroeger niet was). Is er iets op dit gebied anders geworden?

dennissiemensma, 2020-02-26:
Er zijn wat verschillen de afgelopen versies:

- < v3.3 = verbinding opnieuw openen na elk telegram
- v3.3 = herschreven datalogger die verbinding open houdt, maar met nog een bug bij wegvallen verbinding
- v3.4 = herschreven datalogger die verbinding open houdt, met die bug opgelost

Maar blijkbaar zijn de uitwerkingen heel verschillend per gebruiker. Sommigen hebben een lagere load, andere een hogere load, jij hebt overflows. Dus het is voor mij gissen wat goed werkt.

Een load van 40% is opzich niet raar voor iets wat constant de poort uitleest, maar het is wat onnodig. Wellicht dat ik weet terug ga naar een variant van v3.2, als die voor de meesten goed werkte.

bikeymouse, 2020-02-26:
Ok, ik denk dat ik een stap verder ben waar het zit.

Ik ben gewoon gestart met een lege database en vervolgens met oudere docker-imagse om te kijken of het ergens in de software is ontstaan.

Gestart met DSMR V2.15.1 en dat ging prima. Netjes weer strak iedere minuut een meting (met een backend sleep op 30 seconden) en geen overruns meer..
Aha, dus geen hardware issue.

Vervolgens een paar keer ge-upgrade, en uiteindelijk bij de overgang van DSMR V2.21 naar V3.21 -> V3.3 begint het probleem op te spelen!

Even naar de change-log gekeken, en het kan haast geen toeval zijn, maar daar is een change ["datalogger improvements"](https://github.com/dennissiemensma/dsmr-reader/issues/847) 
Misschien dat dit je op weg helpt?

bikeymouse, 2020-02-26:
Ha, ha, onze berichten kruisten net. En met dezelfde conclusie, dus dat is fijn!
Ik heb nu dus weer V3.21 draaien, met 0,2% load en 1 x per minuut even een piekje naar 15-20%, veel beter dus.

Ik snap ziezo niet waarom je je meter vaker dan een minuut zou willen uitlezen, of je moet het voor loadbalancing gebruiken ofzo om snel grote gebruikers (laadpaal, warmtepomp) te kunnen afschakelen als je netbelasting te hoog wordt. Voor gewone reportage is 1x per minuut toch prima?

dennissiemensma, 2020-02-26:
Ik begrijp je standpunt helemaal en ik ben het er mee eens. 

Echter ga ik niet voor alle gebruikers beslissen hoevaak ze hun slimme meter (mogen) uitlezen, dus die keuze laat ik bij hen. Wat ik wel kan doen is de datalogger weer (deels) maken zoals die was. Dat lijkt me een betere oplossing voor nu.

dennissiemensma, 2020-02-26:
Doorgevoerd voor de volgende release.

closed 2020-02-26 by dennissiemensma

bikeymouse, 2020-02-26:
> Ik begrijp je standpunt helemaal en ik ben het er mee eens.
>
> Echter ga ik niet voor alle gebruikers beslissen hoevaak ze hun slimme meter (mogen) uitlezen, dus die keuze laat ik bij hen. Wat ik wel [...]

Tuurlijk als de gebruiker het flexibel kan instellen heb je best of both worlds.

Goed dat je het zo snel bijgewerkt hebt en dank voor de snelle reacties, top!

dennissiemensma, 2020-02-29:
Uitgebracht in [v3.5](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.5.0)