dsmrreader/dsmr-reader #897: tarief calculatie werkt niet meer sinds 3.3 en 3.4

rodeho, 2020-02-21:
Native DSMR 3.4 op een Ubuntu VM met een losse Pi welke via API updates plaatst in de backend.

De energie calculaties staan op zwart (met een - erin) dat is sinds kort. Het energie tarief is correct en ook toegepast voor deze datum. Ik heb nog getracht de energiecalc een "start trigger" te geven door de eind datum een paar dagen te verschuiven.

rodeho, 2020-02-21:
[image]

rodeho, 2020-02-21:
[image]
16e nog in orde 

[image]

17e defect

dennissiemensma, 2020-02-22:
Bedankt voor je melding. 

- Zou je eens willen kijken of je de datums van je contract(en) goed staan?

Op je beide screenshots staat `Energy price` als leeg en dat betekent dat er geen prijs voor die datum is gevonden. Dat veld wordt namelijk live opgehaald bij het bekijken van het Archief.

De prijzen in `Cost` worden echter eenmalig berekend per dag. 

- Als je met terugwerkende kracht de prijzen goed instelt, kun je daarna dit uitvoeren om het te fixen:
```
sudo su - dsmr
./manage.py dsmr_stats_recalculate_prices
```

rodeho, 2020-02-22:
[image]

Dennis. Ik had de prijzen gecontroleerd, maar alleen de eerste regel. Er blijkt overlap in te zitten na het aanpassen van de tarieven door de gewijzigde overheidsbelastingen op 1 Jan. 

Ik zal zo op de PC even e.a aanpassen. En het script runnen. 

Misschien goed om een check in te bouwen?

rodeho, 2020-02-22:
[image]

Het is opgelost!

Wat voor mij nog iets is wat ik al eerder aan wilde geven. Het “today notitie knopje” dat zou voor mij een user selectable optie mogen worden. En die knop van de live grafieken valt er nu ook vrij lelijk overheen. 

Zou mooier zijn om dat allemaal te kunnen verbergen zodat je de data helder tot je kunt nemen zonder afleiding. Wat volgens mij in lijn is met het verplaatsen van de grafieken “overzicht houden”

rodeho, 2020-02-22:
Voor ik het vergeet; enorm veel dank voor al je ontwikkeling!

dennissiemensma, 2020-02-22:
Bedankt voor je input. Ik zal kijken of het mogelijk is om er een check op te maken.

dennissiemensma, 2020-02-22:
Wat betreft die knoppen. Ik moet toegeven dat ik het doorgaans alleen in de browser test op een normaal scherm en niet mobiel. Het schaalt wel automatisch mee, maar dat geeft dus niet altijd het goede resultaat. Ik zal er een keertje doorheen lopen om te kijken wat er anders kan.

dennissiemensma, 2020-02-24:
@rodeho ik heb nu voor de volgende release ook de layout aangepast, zodat het (meeste) wat strakker en rustiger uitlijnt op telefoonschermen.

* Desktop screenshots: https://github.com/dennissiemensma/dsmr-reader/blob/wip/v3.5.0/docs/_static/screenshots/frontend/
* Mobiel:
----
[image]
----
[image]
----
[image]
----
[image]
----
[image]

rodeho, 2020-02-25:
Goedemorgen @dennissiemensma het ziet er goed uit, een heel stuk rustiger! Volgens mij heb je tevens de uitdaging opgelost waarbij in het archive de tabel breder was dan de grafieken.

Ik ga geen beta draaien, maar ik wacht vol verrassing op de release van 3.5.

dennissiemensma, 2020-02-26:
Ik heb gekeken naar de validatie voor overlappende contracten. Helaas krijg ik het niet sluitend wegens een tekortkoming in Django admin validatie.

closed 2020-02-26 by dennissiemensma

rodeho, 2020-03-01:
Dennis 

Op de iPhone gaat het zwart nog buiten de kaders. Ik denk door het vrij forse woord “teruggeleverd”

Zou je daar nog eens naar willen kijken? Want links van de zwarte bolletjes is nog ruimte. 

[image]

rodeho, 2020-03-01:
@dennissiemensma ff een tag omdat hij al gesloten is.

dennissiemensma, 2020-03-01:
Bedankt voor je terugkoppeling. Ik denk echter dat het vrij lastig wordt om dat op te lossen.

Er zijn namelijk zoveel formaten en resoluties, dat het onmogelijk is om alle varianten te proberen en te testen. Als ik het simuleer krijg ik bijvoorbeeld:

[image]

En ik kan het woord wel iets inkorten, maar dan ziet het er weer vaag uit op Desktop. Wellicht dat ik die termen in de toekomst ook zelf-instelbaar maak. Dan kan iedereen zijn of haar eigen termen gebruiken en evt inkorten.