dsmrreader/dsmr-reader #898: Staafgrafieken archief

donnie-dominguez, 2020-02-21:
Ik ben blij met de staafgrafieken in het ' archief'. Erg overzichtelijk. Als je met de muis over een staaf van de grafiek ' hovert' dan is hoogtarief/laagtarief te zien. Wellicht is het mogelijk om in datzelfde pop-upje ook het totaalverbruik (laagtarief+hoogtarief) te zetten? Uiteraard is dat ook makkelijk zelf uit te rekenen, maar dit zou het nog overzichtelijker maken denk ik.

dennissiemensma, 2020-02-22:
Ik zal kijken of dat mogelijk is met de library die ik gebruik voor de grafieken. Lijkt me goede toevoeging als dat kan.

dennissiemensma, 2020-02-24:
Ik heb hiernaar gekeken, maar ik kreeg het helaas niet zo snel voor elkaar.

donnie-dominguez, 2020-02-25:
Dat is jammer. Bedankt voor het kijken.

closed 2020-02-26 by dennissiemensma