dsmrreader/dsmr-reader #899: Datum blijft "hangen"

peternijssen, 2020-02-22:
Hallo Dennis,

Excuus voor de onduidelijke titel. Bij mij worden de grafieken niet weergegeven. Ik heb issue #817 reeds bekeken en toegepast, maar helaas maakt dat geen verschil. Echter merkte ik nu ook dat bij mij de datum vast staat.

Op de dashboard pagina staat: THURSDAY JANUARY 2ND, 2020  
Bij de PVOutput pagina staat: Next export: Jan. 2, 2020, 1:40 p.m.

Verder wil bij mij de status pagina niet laden (time-out) en bij archief kan ik enkel 2 januari en ouder selecteren.

Ik zie in de database wel records van de afgelopen dagen / maanden worden toegevoegd.

Hopelijk is het een beetje duidelijk. Enig idee wat hier het probleem is?

dennissiemensma, 2020-02-22:
Ik denk dat je het beste debug-logging kunt inschakelen: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

En daarna kijken in `/var/log/supervisor/dsmr_backend.log`. Wellicht zie je daar dan een melding staan met meer informatie.

peternijssen, 2020-02-22:
Reeds gedaan. Helaas in geen enkele log ook maar iets van een fout. :(
Ik heb vandaag ook geupdate naar de allerlaatste versie, maar helaas geen verschil.

Andere ideeen waar ik naar kan kijken?

dennissiemensma, 2020-02-22:
Je zult in dit geval niet op zoek moeten naar een fout, maar meer naar een 'constatering' waardoor die hangt. In debug-logging geeft de applicatie aan wat die doet is en waarom. Grote kans dat er dus iets in staat.

Wat geeft de debug log voor de backend als output?

peternijssen, 2020-02-23:
Ik zie dat de datum inmiddels is verschoven naar vrijdag 4 januari. Ik denk door van alles wat ik heb geprobeerd, hij weer alles aan het in halen is. Gaat alleen even duren zo te merken.

Mocht het toch niet lukken, dan laat ik het weten. Bedankt voor je hulp in ieder geval!

dennissiemensma, 2020-02-23:
Wat voor hardware gebruik je? Het verwerken van de metingen zou doorgaans niet zo lang moeten duren, tenzij je op een Pi 1 of 2 ofzo draait.

Je kunt ook even in de database zelf kijken of die uberhaupt voortgang maakt met verwerken:
```
sudo su - postgres
psql dsmrreader

select timestamp from dsmr_datalogger_dsmrreading where processed = false order by timestamp asc limit 10;
```

Als je dat een paar keer doet, zou ik verwachten dat die niet op dezelfde blijft hangen.

peternijssen, 2020-02-24:
Ik zit op een Pi 3B+ geloof ik. Of iets in die trant. Inmiddels ben ik door naar 6 jan, 4 dagen verder.
In de aangegeven tabel staan nog zo'n 2.5 miljoen records op niet processed. Inmiddels lijkt hij ook te blijven hangen, want er gebeurt nog niets. Of ben ik niet geduldig.

Onderstaand de deug log.

```
[2020-02-24 20:10:07,000] WARNING   [!] PVOutput: No data found (yet)
[2020-02-24 20:24:31,839] WARNING   [!] PVOutput: No data found (yet)
[2020-02-24 20:38:49,700] WARNING   [!] PVOutput: No data found (yet)
[2020-02-24 20:54:28,166] WARNING   [!] PVOutput: No data found (yet)
[2020-02-24 21:08:32,953] WARNING   [!] PVOutput: No data found (yet)
[2020-02-24 21:24:00,383] WARNING   [!] PVOutput: No data found (yet)
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
[2020-02-24 21:29:43,079] DEBUG    dsmr_backend.management.commands.dsmr_backend: Starting infinite command loop...
[2020-02-24 21:29:43,087] DEBUG    SP: 1 backend service(s) ready to run
[2020-02-24 21:29:43,088] DEBUG    SP: Running "Generate consumption data" (dsmr_consumption.services.run)
```

2 dagen terug zag ik nog:
```
[2020-02-22 15:42:01,566] DEBUG    Compact: Processed reading: 20462823 @ 2020-01-02 16:14:08+01:00
[2020-02-22 15:42:01,606] DEBUG    Compact: Processed reading: 20462824 @ 2020-01-02 16:14:10+01:00
[2020-02-22 15:42:01,645] DEBUG    Compact: Processed reading: 20462825 @ 2020-01-02 16:14:11+01:00
```

dennissiemensma, 2020-02-24:
Ik vermoed dat de database van je installatie wat indexes mist. Zonder indexes en miljoenen records is die operatie namelijk heel traag. Dat ik die link niet eerder heb gelegd.

Wil je eens kijken wat de output is op:
```
sudo su - postgres
psql dsmrreader

\dS dsmr_datalogger_dsmrreading
```

En dan vooral het onderste blok van de output, dat zou er zo ongeveer uit moeten zien:
```
Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_9d090a40" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1_uniq" btree ("timestamp")
```

peternijssen, 2020-02-24:
Daar had ik ook niet aan gedacht 🙈

Ik heb nog een MySQL DB, maar ik kan de vertaalslag wel maken. Ik kom er binnen enkele dagen op terug als ik weer thuis ben.

dennissiemensma, 2020-02-24:
Prima. Die indexes zouden trouwens automatisch aangemaakt moeten zijn, maar het afgelopen jaar zijn er meerdere gebruikers bij wie die soms niet aangemaakt zijn.

```
USE dsmrreader;
SHOW INDEXES FROM dsmr_datalogger_dsmrreading;
```

peternijssen, 2020-02-28:
Alle 3 de indexes zijn aanwezig. Dus dat lijkt gewoon in orde te zijn. Inmiddels zitten we op Zondag 26 januari. Dus we zijn inmiddels 20 dagen opgeschoten in 4 dagen tijd, dus hij lijkt steeds sneller te gaan.

dennissiemensma, 2020-02-29:
Dat is fijn! 

Echter zou dat nog steeds niet zo lang moeten duren en dat het steeds sneller gaat, geeft mij alleen maar meer vermoedens dat er nog ergens een index mist. Zou ook deze tabellen willen checken?

## dsmr_consumption_electricityconsumption
```
Indexes:
    "dsmr_consumption_electricityconsumption_pkey" PRIMARY KEY, btree (id)
    "dsmr_consumption_electricityconsumption_read_at_key" UNIQUE CONSTRAINT, btree (read_at)
    "dsmr_consumption_electrici_phase_power_current_l1_3d9fd70d" btree (phase_power_current_l1)
    "dsmr_consumption_electrici_phase_voltage_l1_72e3c8f8" btree (phase_voltage_l1)
    "dsmr_consumption_electrici_phase_voltage_l2_917726c7" btree (phase_voltage_l2)
    "dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37" btree (phase_voltage_l3)
    "dsmr_consumption_electricityc_currently_delivered_5bf9be2d_uniq" btree (currently_delivered)
    "dsmr_consumption_electricityco_currently_returned_5f91fed7_uniq" btree (currently_returned)
    "dsmr_consumption_electricityconsumption_2241f8ee" btree (phase_currently_delivered_l2)
    "dsmr_consumption_electricityconsumption_5714c500" btree (phase_currently_delivered_l1)
    "dsmr_consumption_electricityconsumption_e90842a6" btree (phase_currently_delivered_l3)
```

## dsmr_consumption_gasconsumption
```
Indexes:
    "dsmr_consumption_gasconsumption_pkey" PRIMARY KEY, btree (id)
    "dsmr_consumption_gasconsumption_read_at_key" UNIQUE CONSTRAINT, btree (read_at)
```

## dsmr_stats_daystatistics
```
Indexes:
    "dsmr_stats_daystatistics_pkey" PRIMARY KEY, btree (id)
    "dsmr_stats_daystatistics_day_key" UNIQUE CONSTRAINT, btree (day)
    "dsmr_stats_daystatistics_total_cost_c898d987" btree (total_cost)
```

## dsmr_stats_hourstatistics
```
Indexes:
    "dsmr_stats_hourstatistics_pkey" PRIMARY KEY, btree (id)
    "dsmr_stats_hourstatistics_hour_start_key" UNIQUE CONSTRAINT, btree (hour_start)
```

peternijssen, 2020-02-29:
## dsmr_consumption_electricityconsumption
```
PRIMARY
read_at
dsmr_consumption_electricit_currently_delivered_0f28e53a
dsmr_consumption_electricit_currently_returned_b7a910c8
dsmr_consumption_electricit_phase_currently_delivered_l_9998ce55
dsmr_consumption_electricit_phase_currently_delivered_l_0315221e
dsmr_consumption_electricit_phase_currently_delivered_l_21a515a7
dsmr_consumption_electricit_phase_voltage_l1_72e3c8f8
dsmr_consumption_electricit_phase_voltage_l2_917726c7
dsmr_consumption_electricit_phase_voltage_l3_1a9d7c37

```
Behalve dat enkele namen anders zijn, mist deze: `dsmr_consumption_electrici_phase_power_current_l1_3d9fd70d`

## dsmr_consumption_gasconsumption
```
PRIMARY
read_at
```
Lijkt dus compleet te zijn

## dsmr_stats_daystatistics
```
PRIMARY
day
```

lijkt dus `dsmr_stats_daystatistics_total_cost_c898d987` te missen

## dsmr_stats_hourstatistics
```
PRIMARY
hour_start
```

Lijkt dus compleet te zijn

peternijssen, 2020-02-29:
Ook nog vreemd (denk ik);
Bij "Archive" en "Compare" is de meest recente datum die ik kan selecteren, 1 januari 2020. Ondanks dat ik tot nu toe opgebouwd heb tot 31 januari.

dennissiemensma, 2020-03-01:
Het is goed om te weten dat het verwerken van de metingen en de datums/dagtotalen in het Archief twee losse iteraties zijn. 

Als je debug-logging nog aan hebt staan, kun je wel zien waarom die ergens hangt bij het genereren van de dagtotalen. Je krijgt dan vermoedelijk een van deze log-meldingen te zien: https://github.com/dennissiemensma/dsmr-reader/blob/v3.5.0/dsmr_stats/services.py#L76

----

Wat betreft de automatisch gekozen indexes. Ik had zelf niet verwacht dat er zo'n groot verschil zou zitten tussen mysql en postgresql. 
- Hoeveel onverwerkte metingen heb je momenteel nog? Dit kun je als het goed is zien op de Status-pagina.

peternijssen, 2020-03-01:
Ik heb zojuist een gezamenlijke index op "processed" en "timestamp" gelegd (waarbij de eerste processed is). Query draait nu in minder dan 0.01 seconde. Dat lijkt dus voor mij de oplossing te zijn.

Die dagelijkse stats ga ik naar kijken. Zojuist debugging weer aan gezet.

dennissiemensma, 2020-03-01:
Ik denk eerlijk gezegd dat als die query nu snel gaat, de achterstand vrij snel weggewerkt wordt. En dan volgt het Archief vrij snel vanzelf.

dennissiemensma, 2020-03-01:
Je kunt even af en toe een count doen op het aantal records in `dsmr_datalogger_dsmrreading` waarbij processed fale/0 is. Die zou nu snel moeten dalen.

peternijssen, 2020-03-01:
Yep, het gaat nu razendsnel. Inmiddels zag ik deze voorbij komen;
```
[2020-03-01 20:49:20,421] DEBUG    SP: Running "Generate day and hour statistics" (dsmr_stats.services.run)
[2020-03-01 20:49:22,759] DEBUG    Stats: Missing consumption data for: 2020-01-02
[2020-03-01 20:49:22,971] DEBUG    SP: Rescheduled "Generate day and hour statistics" to 2020-03-01 21:49:22.760399+01:00 (ETA 0:59:59.788593)
```

Laten we even afwachten tot hij bij is.

dennissiemensma, 2020-03-02:
@peternijssen heeft dit het uiteindelijk opgelost?

peternijssen, 2020-03-03:
Gisterenavond rond 22u was hij helemaal bij. Dat ging dus enorm snel. De stats pagina doet het ook weer door de DB optimalisatie.

De day / hour wordt nog niet berekend. Ik blijf dit in de logs krijgen:
```
[2020-03-03 07:02:15,040] DEBUG    SP: Running "Generate day and hour statistics" (dsmr_stats.services.run)
[2020-03-03 07:02:15,324] DEBUG    Stats: Missing consumption data for: 2020-01-02
[2020-03-03 07:02:15,337] DEBUG    SP: Rescheduled "Generate day and hour statistics" to 2020-03-03 08:02:15.324701+01:00 (ETA 0:59:59.987031)
```

dennissiemensma, 2020-03-03:
Top! In dat geval sluit ik deze, voor je initiele issue, en kan je het beste #909 volgen. 

Dat gaat namelijk over waar je nu tegenaan loopt en heeft vermoedelijk dezelfde oorzaak.

closed 2020-03-03 by dennissiemensma

peternijssen, 2020-03-03:
Bedankt! Mocht je nog een index toevoegen, dan hoor ik dat graag, dan verwijder ik de mijne. Anders laat ik hem gewoon staan omdat alles daardoor veel soepeler draait.

dennissiemensma, 2020-03-03:
Ik zou hem voor nu lekker laten staan. Officieel ondersteunt het project al een tijdje geen MySQL meer, maar zolang ik er beperkt extra werk aan heb vind ik het prima om af en toe mee te debuggen.