dsmrreader/dsmr-reader #900: Verkeerde buiten temperatuur

Erik050572, 2020-02-22:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.4
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 3
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: both
* DSMR-protocol slimme meter *(v4 of v5)*: v5

Goedemiddag Dennis,

Ik weet niet of de bug in het programma zit of bij het "KNMI":
De grafiek voor de buitentemperatuur geeft sinds de upgrade naar V3.4 de hele tijd dezelfde waarde weer
Is dit op te lossen?

Mvg,
Erik

coen17st, 2020-02-22:
Dit is bij mij ook het geval, heb een weerstation uit Arhem geselecteerd, in DSMR-Reader wordt een constante temperatuur van 5 graden celcius aangegeven en wijkt af van de werkelijke temperatuur.

cranberryNL, 2020-02-22:
HIer idem (v3.4). Sinds de update naar 3.4 blijft de temperatuur hangen op 5.6 gr C. (Weerstation De Bilt)

dennissiemensma, 2020-02-22:
Bedankt voor jullie melding. Ik zie hetzelfde bij mijzelf, maar kan alleen constateren dat dat de temperatuur is die Buienradar via haar API teruggeeft.

Arnhem:
[image]

Bilt:
[image]

https://data.buienradar.nl/2.0/feed/json

dennissiemensma, 2020-02-22:
Wellicht morgen even kijken of het zichzelf dan oplost.

kgroot, 2020-02-23:
Ik heb het probleem nog steeds. De Bilt is 5.6 graden constant. Ook de geschiedenis (van zaterdag) geeft 5.6 graden.

kommer, 2020-02-23:
Hoi,

Ik zit nog op v3.3 en ik heb ook temperaturen die blijven hangen. Misschien
werkt de API van buienradar niet meer goed?

Groeten,
Kommer

dennissiemensma, 2020-02-23:
Ik zie in de API nog steeds dezelfde waardes. Je kunt overwegen om even een berichtje te sturen naar Buienradar zelf.

kommer, 2020-02-24:
Hoi Dennis en anderen,

Ik heb contact gehad met buienradar en ze hebben het volgens mij opgelost.
In mijn DSMR reader zie ik weer andere waarden!

cranberryNL, 2020-02-24:
Is inderdaad opgelost, o.b.v. mijn grafiek hier ergens tussen 14:00 en 15:00.

dennissiemensma, 2020-02-24:
Dank @kommer !

closed 2020-02-24 by dennissiemensma