dsmrreader/dsmr-reader #903: Grafieken volgorde

vincedh, 2020-02-23:
Hoi Dennis,

Is het misschien een idee om de volgorde van de live grafieken aan te (kunnen) passen?
Ik vind 'RECENT GASVERBRUIK' veel interessanter dan 'RECENTE DISTRIBUTIE ELEKTRICITEITSFASEN' of 'RECENTE VOLTAGES FASEN'

Hoor graag je reactie, bedankt!

dennissiemensma, 2020-02-23:
Bedankt voor je input. Ik zal het overwegen, maar op de korte termijn laat ik het voor wat het is.

vincedh, 2020-02-24:
Helemaal goed, dank voor je reactie.

Phyxion, 2020-03-23:
Vind ik ook een goed idee, misschien als ze niet aan te passen zijn dan gas boven het voltage zetten, zoveel kan je daar namelijk niet mee.

dennissiemensma, 2020-08-24:
Voor de toekomstige v4.3 release heb ik iets toegevoegd wat de volgorde van grafieken instelbaar maakt in de admin. Het is niet de meest mooie oplossing, maar het werkt en is wat mij betreft afdoende voor nu.

closed 2020-08-24 by dennissiemensma

dennissiemensma, 2020-08-28:
Uitgebracht in [v4.3](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.3.0).