dsmrreader/dsmr-reader #904: Kleuren van grafieken

vincedh, 2020-02-24:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.4
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 3
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: native
* DSMR-protocol slimme meter *(v4 of v5)*: v5

Hoi Dennis,

Ik heb 2 verschillende kleuren voor hoog en laagtarief (zie afb 1).
Maar als ik bij afgelopen vrijdag kijk (een dag met verschillende tarieven), geeft hij in de laagperiode de rode kleur en bij hoog tarief de groene kleur (precies andersom dus), zie afb 2.

Op de pagina 'Trends' gaat het wel goed, zie afb 3.

Vriendelijke groet,
[image]
[image]
[image]

dennissiemensma, 2020-02-24:
Bedankt voor je melding. Dit lijkt er al heel lang in te zitten, maar het is tot nu toe nog niemand opgevallen. Ik fix het in de volgende release.

closed 2020-02-24 by dennissiemensma

dennissiemensma, 2020-02-29:
Uitgebracht in [v3.5](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.5.0)