dsmrreader/dsmr-reader #905: dsmr-reader version to home-assistant

RubenDijk, 2020-02-24:
Hallo,

Zou het mogelijk zijn om versie info mee te sturen over mqtt naar --> home-assistant
Zo dat het zichtbaar is in home-assistant of er een update beschikbaar is.

Sensors:
Installed version
Latest version

dennissiemensma, 2020-02-24:
Bedankt voor je verzoek. Ik zal het tzt overwegen, maar ik ga hier op de korte termijn sowieso niet mee bezig.

RubenDijk, 2020-02-25:
Hey Dennis,

Bedankt voor je reactie.

dennissiemensma, 2020-05-28:
Ik heb hier opnieuw naar gekeken. De achterliggende gedachte van het verzoek begrijp ik goed, maar ik denk dat het wat te veel is om dit helemaal te gaan ontsluiten via MQTT. 
Wat mij betreft is het wat logischer om de scheidingslijn te leggen bij alleen het delen van 'gemeten' gegevens via MQTT. Technische gegevens over DSMR-reader zelf horen daar dan niet bij.

Een andere belangrijke reden is dat het doorgaans niet erg is als je een tijdje achterloopt op DSMR-reader versie. De basisfunctionaliteit van dit project zal niet veranderen. Ook worden er zelden nog ernstige bugs gevonden, want die zijn er nu na een aantal jaren vrijwel allemaal uit. Sterker nog, ik denk dat je met updaten meer risico loopt om tegen een nieuwe bug aan te lopen, dan wanneer je op 1 versie blijft voor een tijdje. ;-)

closed 2020-05-28 by dennissiemensma