dsmrreader/dsmr-reader #908: Domoticz naar DSMR

niekniek89, 2020-02-25:
Goedenavond,

ik heb een vraagje:

ik heb een collega van mij gevonden die alles voor mij heeft omgebouwd (van Domoticz naar DSMR). alle waardes uit Domoticz lijken in DSMR te kloppen (3 jaar aan data), en ziet er allemaal super uit. (info gehaald uit https://github.com/dennissiemensma/dsmr-reader/issues/520)

echter zie ik de waardes m.b.t. het terug leveren (zonnepanelen) niet terug in DSMR. moet deze optie nog ergens aangezet worden? nu is het zo dat DSMR nog in een test omgeving (docker) draait, en nog niet is voorzien van een P1 kabel. worden deze waardes aangevuld zodra de p1 kabel is aangesloten, en hij ziet dat ik terug lever?

bij voorbaat dank voor jullie hulp.

Niek van den Broek

dennissiemensma, 2020-02-25:
Bedankt voor je vraag. DSMR-reader toont pas teruglevering zodra er daadwerkelijk teruglevering is. Dus als je zonnepanelen hebt, en de meterstand voor terugleverin loopt op, wordt het getoond.

Als je nu toch nog aan het testen bent, kun je het beste even de P1 kabel voor 1 of 2 minuutjes via DSMR-reader laten doen. Daarna weer terugzetten. Als het goed is zie je dan (na een minuut) teruglevering en heb je een idee van hoe het werkt.

dennissiemensma, 2020-02-25:
Wellicht kun je trouwens na de switch alsnog beiden (Domoticz) gebruiken wanneer je over bent, mits Domoticz die waardes via MQTT kan hanteren (zelf geen ervaring mee). Dan kun je ze een tijdje naast elkaar draaien.

closed 2020-02-25 by dennissiemensma

niekniek89, 2020-02-25:
@dennissiemensma , bedankt voor je snelle reactie!

Dus als ik het goed begrijp, wordt het “terugleveren” pas weergegeven zodra hij dit daadwerkelijk “ziet” Gebeuren n.a.v. de p1 kabel.

Hoe zit het met de historische data die er al in staat? Laat hij daarvan ook pas de waardes “terugleveren” zien zodra de p1 kabel zit aangesloten?

Alle historische data is perfect overgezet naar DSMR, behalve het “terugleveren” is te zien (historisch).

Wederom bedankt voor je support!

dennissiemensma, 2020-02-25:
Als je de data alleen in het Archief hebt geimporteerd dan klopt dat. Afhankelijk van in welke tabel je de data hebt geimporteert, checkt DSMR-reader de 'features' die het moet tonen (gas, teruglevering):

https://github.com/dennissiemensma/dsmr-reader/blob/v3.4.0/dsmr_backend/services/backend.py#L34

niekniek89, 2020-02-25:
Alle data (terugleveren, gebruik en gas) zijn inderdaad in het archief geïmporteerd. Verbruik (per dag, maand en jaar) worden voor stroom en gas al getoond, echter het terugleveren nog niet.

Deze worden dus alsnog getoond (historisch) als de p1 meter is aangesloten, en de juiste tabellen zijn gebruikt t.b.v de import (als ik het goed begrijp)?

Bedankt voor je hulp.

dennissiemensma, 2020-02-25:
Ja. Je kunt voor de zekerheid even checken of deze optie uitstaat. Dit forceert het verbergen (zou standaard uit moeten staan):

https://github.com/dennissiemensma/dsmr-reader/issues/890#issuecomment-588387512

niekniek89, 2020-02-25:
Super bedankt voor je hulp! We gaan ermee aan de slag! Mochten we nog tegen kleine dingetjes aanlopen, zal ik deze hier nog stellen (indien mogelijk)?

Baseje, 2020-02-26:
Ik ben de collega welke het script heeft gemaakt op basis van #520 ,
Heb de API calls gedaan middels Powershell (Lukt mij wat makkelijker dan PYTHON haha)

Ik las in #520 dat je de volgende commando na het importen moest gebeuren;
sudo su - dsmr ./manage.py dsmr_backend_delete_aggregated_data --statistics --readings --consumption

Krijg alleen onderstaande melding;
bash-5.0# ./manage.py dsmr_backend_delete_aggregated_data --statistics --readings --consumption
Unknown command: 'dsmr_backend_delete_aggregated_data'
Type 'manage.py help' for usage.

Enig idee wat ik hier aan kan doen?

dennissiemensma, 2020-02-26:
Dat command was destijds nodig omdat diegene alle metingen had geimporteerd. Hebben jullie de metingen geimporteerd of alleen dagtotalen?

Baseje, 2020-02-26:
Hebben alle gegevens gepushed met de API (Dag totalen) en dan dus van 00.00 en van 23.59.
Daarna kwamen ze automatisch in de dagtotalen toch?

Missen dus alleen de teruglevering, terwijl we deze wel hebben gepushed!

dennissiemensma, 2020-02-26:
Ja zie https://github.com/dennissiemensma/dsmr-reader/issues/908#issuecomment-591020760

Je kunt ofwel de kabel aansluiten, ofwel handmatig een record toevoegen aan dsmr_consumption_electricityconsumption.

Baseje, 2020-02-28:
Helemaal gelukt! Na het plaatsen van de kabel en het moment dat er echt teruggeleverd werd werd ook de historie zichtbaar. Lijkt net alsof deze trigger nodig is om het moment van terugleveren te tonen!

Volledige beschreven stappen;
[Domoticz to DSMRReader.zip](https://github.com/dennissiemensma/dsmr-reader/files/4266583/Domoticz.to.DSMRReader.zip)

1. Export database binnen Domoticz
2. Open de database middels 'DB browser for SQLite' (https://sqlitebrowser.org/)
3. Exporteer de tabellen 'Meter_Calendar' en 'Multimeter_Calendar' met als seperator ';'.
4. Zet de API binnen DSMR Reader aan en kopieer de key.
5. Open het powershell script middels de Powershell ISE.
6. Pas het powershell de volgende zaken aan: 'LOCATIE CSV' , 'APIKEY' en 'URL-DSMR' met de gegevens zoals eerder geëxporteerd.
7. Start vervolgens het Powershell script, waarna je flink een aantal API calls ziet gebeuren.

Het powershell script importeert de CSV's en zorgt ervoor dat exact de juiste conventie gepusht word per API call. Dit gebeurd voor iedere meting om 00.00 en om 23.59 (Dit zodat de dagen gevuld worden).

Vervolgens kan het wel even duren voordat gegevens zichtbaar worden in DSMR Reader Archief.

dennissiemensma, 2020-02-29:
Fijn om te horen, bedankt voor de terugkoppeling!

basroovers, 2020-03-31:
@Baseje bedankt voor het delen van het script. Het werkte bij mij als een trein.
Ik moest wel de DeviceRowID aanpassen naar de correcte id voor mijn Domoticz configuratie. In mijn geval was het id 3.

remb0, 2020-04-10:
wat ik ook probeer met mijn beperkte kennis krijg ik het niet voor elkaar.   het script loopt foutloos, maar binnen dsmr veranderd er niets.
in de bijlage mijn input bestanden en het ingevulde powershell.    kan iemand kijken wat er fout is of aangeven waar ik moet kijken binnen dsmr?
[domoticz_to_dsmr.zip](https://github.com/dennissiemensma/dsmr-reader/files/4459965/domoticz_to_dsmr.zip)

Ik zou het fantastisch vinden om heel mijn historie in deze tool te hebben :)

dennissiemensma, 2020-04-10:
Ik denk dat je je het beste kan wenden tot het Tweakers forum. Op Github lezen doorgaans weinig mensen mee.

Baseje, 2020-04-14:
> wat ik ook probeer met mijn beperkte kennis krijg ik het niet voor elkaar. het script loopt foutloos, maar binnen dsmr veranderd er niets.
> in de bijlage mijn input bestanden en het ingevulde [...]

Is het al gelukt?

remb0, 2020-04-14:
nee helaas..

HellStorm666, 2020-10-13:
Bij mij werkte ie eerst niet.
Maar in het schript staat bij ##Filter (regel 12) $_.DeviceRowID -eq "29"
Als ik mijn MultiMeter_Calendar.csv open, staat het overgrote deel op device id 1.
Ik heb dus de 29 vervangen door een 1. En dan doet ie t wel.

HellStorm666, 2020-10-15:
Vreemde is, Elektra komt er dus in, Gas niet.
Meende eerst dat het script dit fout deed en dus niet mee stuurt, maar in de echo @Data staat:
`echo @Data
-electricity_delivered_1:
10656.456
-electricity_delivered_2:
9982.104
-electricity_returned_2:
3678.891
-electricity_currently_returned:
0
-electricity_currently_delivered:
0
-timestamp:
2020-10-13T00:00
-extra_device_timestamp:
2020-10-13T00:00
-extra_device_delivered:
6189.285
-electricity_returned_1:
1580.173`

Wat dus laat zien dat gas wel degelijk mee gaat.
En toch laat DSMR het niet zien...

remb0, 2021-02-20:
ik heb het nogmaals geprobeerd en in de postgress db zie ik de records van voor 2019 :)
ook in admin > dsmr-uitlezing (alleen lezen) zie ik de gegevens terug.
alleen in frontend > archief > kan ik niet verder terug dan 2019 (wanneer ik begonnen ben met dsmr).  de uitgelezen data (via het script op https://github.com/dsmrreader/dsmr-reader/issues/908)   lijkt dus wel in dsmr te staan maar wordt nog niet in het archief gedeelte geplaatst.   wat kan ik hieraan doen?
bij geplande processen heeft alles gedraaid (retention data rotation, generate consumption data en generate day and hour statistics.)

wat vergeet ik?

dehart, 2021-12-30:
Bedankt voor het Script Baseje! 👌
Zojuist mijn Domoticz data gemigreerd naar DSMR, erg blij mee. 😁

Ik liep tegen 1 probleem aan die ik hier wil melden voor een eventuele volgende. (naast dat ik ook DeviceRowID 1 had)

Ik zag alleen het Gas maar niet de Electra.
Toen ik de database tabel `dsmr_consumption_electricityconsumption` ging inspecteren kwam ik er achter dat alle timestamps er 1 minuut erbij hadden gekregen. (in `dsmr_consumption_gasconsumption` was dat niet)

Dit had ik opgelost met de volgende query:
`update dsmr_consumption_electricityconsumption set read_at = read_at - interval '1 minute' where read_at < '2021-08-05 00:00:00.000000 +01:00';`
(mijn geimporteerde data was van voor 05-08-2021 vandaar deze where clause)
daarna de dag statistieken leeg gemaakt:
`truncate dsmr_stats_daystatistic;`
en het `Generate day and hour statistics` proces gestart.

PS de data die van de API terug komt geeft wel de juiste timestamp. ik heb geen idee waar die extra minuut vandaan komt.

ChefkeGremmen, 2023-01-05:
@Baseje @dehart ; bedankt voor het uitzoeken heren, gisterenavond is het me eindelijk gelukt om mijn migratie af te ronden en jullie kennis en kunde kwamen heel goed van pas! @dennissiemensma ; is het een overweging om dit mee te nemen in de documentatie?

dennissiemensma, 2023-01-05:
Op den duur verplaats ik nieuwe documentatie hier naar Github, ik verwacht dat het dan ook veel makkelijker wordt om te linken naar oude issues/threads.