dsmrreader/dsmr-reader #910: DSMR - seconde interval (rechtsboven)

niekniek89, 2020-02-29:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

DSMR Reader --->Docker
V3.4
Intel NUC
Docker

Goedemiddag,

ik heb een vraag over de interval (lijkt het op), die rechtsboven in beeld komt.
bij mij begint deze vanaf 31 sec te lopen, tot hij de 41 bereikt heeft.
daarna worden de gegevens ververst (zie bijlage).

klopt dit dat hij dit vanaf de 30e seconden doet? (oplopen naar de 40 seconden). Daarna begint hij weer bij de 30e seconden.
[image]

ik hoor graag of dit klopt.
alvast bedankt.

dennissiemensma, 2020-03-01:
Bedankt voor je melding. Ik begrijp dat het verwarrend is.

De tijd die getoond/gebruikt wordt komt uit je slimme meter. Soms wijkt die af en bij de ene meter wat meer dan bij de andere. Bij mij thuis blijft die ook altijd tussen de 20 en 30 seconden.

closed 2020-03-01 by dennissiemensma

niekniek89, 2020-03-02:
@dennissiemensma , bedankt voor je snelle reactie en support!
gelukkig maar, er is dus niks spannends aan de hand, en kan dit gewoon negeren.