dsmrreader/dsmr-reader #911: Inzoomen grafieken

vincedh, 2020-03-01:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.5
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 3
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: native
* DSMR-protocol slimme meter *(v4 of v5)*: v5

Hoi Dennis,

Sinds de upgrade naar 3.5 mis ik de functie om 'in te kunnen zoomen' bij de live grafieken van recent gasverbruik & recente temperaturen. Bij de andere grafieken zit de functie er nog wel in.

dennissiemensma, 2020-03-01:
Bedankt voor je melding. Ik had ze er bewust uitgehaald omdat de temperatuur maar maximaal 24 meetpunten per 24 uur heeft. En voor gas idem, maar ik besef me nu dat voor DSMR-v5 meters er meer meetpunten (kunnen) zijn.

Ik zal ze in de volgende release weer toevoegen.

dennissiemensma, 2020-03-02:
Aangepast voor volgende release.

closed 2020-03-02 by dennissiemensma

dennissiemensma, 2020-03-06:
Uitgebracht in [v3.6](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.6.0)