dsmrreader/dsmr-reader #912: Layout klopt niet

goegol, 2020-03-01:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.5.0.
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 4
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: v5

Layout onderaan zit nu stuk witte achtergrond

[image]

Tevens heb ik nu hoop loze ruimte bovenin.
Valt dat nog te optimaliseren? door bijv. meer gebruik te maken van colummen?

[image]

dennissiemensma, 2020-03-01:
Bedankt voor je feedback. Ik zal voor de volgende release kijken of ik het nog iets beter kan maken.

dennissiemensma, 2020-03-02:
Aantal wijzigingen doorgevoerd voor volgende release.

closed 2020-03-02 by dennissiemensma

dennissiemensma, 2020-03-06:
Uitgebracht in [v3.6](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.6.0)

Mixpower, 2020-03-07:
Ik vind het toch aardig storend dat het huidig verbruik nu niet meer word weergegeven rechts bovenin, al deze updates zijn fijn maar er worden nu naar mijn idee wel heel veel onnodige aanpassingen aan de layout gedaan.

dennissiemensma, 2020-03-07:
@Mixpower bedankt voor je stem en ik begrijp je gevoel daarbij. Vermoedelijk zijn dit de laatste wijzigingen die nodig waren.  Een hoop aanpassingen waren nodig voor #858, #885, #878 en #901. 
Toen ik daar mee bezig ging, heb ik een hoop andere kleine foutjes en inconsistenties opgelost, waardoor het geheel wellicht als een grote wijziging voelt. Vanuit mijzelf ben ik geen vormgever of designer, dus het blijft altijd zoeken naar wat goed werkt en een redelijk compromis is.

Als alternatief staat het je natuurlijk vrij om een oudere versie te draaien met een layout die je meer aanstaat. Qua features denk ik namelijk dat DSMR-reader sowieso al een tijdje geleden 'klaar' is en dus inhoudelijk niet veel meer verandert. Het doet waar het voor gemaakt is. Er zullen altijd wensen blijven komen, maar ik beperk dat waarschijnlijk tot het exporteren van data op nieuwe manieren (zoals MQTT of InfluxDB) en wellicht nog een extra manier voor inzage in vergelijkingen of periodeselecties.

Mixpower, 2020-03-07:
@dennissiemensma Snap ik volkomen hoor het blijft altijd lastig om de juiste balans te vinden, maar het in één oogopslag het huidig verbruik kunnen zien vond ik wel heel fijn en naar mijn idee ook niet iets dat hoeft te verdwijnen. De rest van de aanpassingen kan ik nog wel inkomen, maar misschien is een soort van modulaire layout zodat je doodmiddel van widgets je eigen dashboard kan inrichten een idee voor zover dit te implementeren is natuurlijk.