dsmrreader/dsmr-reader #913: update

style2k, 2020-03-01:
raspbery pi 3 dsrm versie 2.6

[image]

gister werkte het nog prima nu lijkt de boel weer is vast te lopen . het is kwart over 7 maar hier klopt de tijdlijn niet ook zie je dat die eerst 9 55 aangeeft en daarna weer 9,54 

[image]

ook hier heeft die ineens van de 1 op de andere dag grote achterstand op onverwerkte gegevens . dit is al de zoveelste keer dat er dingen niet kloppen een pi is zo onbetrouwbaar . maar goed hoe kan dit gewoon steeds vastlopen of achter de feiten aan .
want er is geen beweging te krijgen in onverwerkte berichten . vorige keer ging het langzaam maar nu gaat het dus totaal niet .

ik zit op versie 2.6 en denk dat ik toch ga updaten maar hoe voorkom ik steeds deze problemen ??

dennissiemensma, 2020-03-01:
Zoals je zelf al aangeeft denk ik dat je het beste eerst kunt updaten en dan kijken of het iets oplost. 

Je zult daarvoor eerst naar v2.15 moeten updaten en daarna hoger. Dat kan gewoon met:
```
sudo su - dsmr
./deploy.sh
```

Zit je inmiddels al op v2.15?

style2k, 2020-03-01:
ik zit al op 3.5

dennissiemensma, 2020-03-01:
Je hoeft absoluut niet terug. In je vorige bericht stond nog dat je op 2.6 zat en dat het niet helemaal lukte, vandaar de suggestie.

- Als je al op 3.5 zit, ben ik benieuwd of het aantal onverwerkte telegrammen nu wel (langzaam) terugloopt?
- En heb je een DSMR v4-meter of een v5-meter?

style2k, 2020-03-01:
ik kan naar het dashboard maar als ik naar een andere pagina wil status of live statestieken blijft die laden dus kom nie verder .
en hoe zie ik of ik een v4 of v5 meter heb ?

dennissiemensma, 2020-03-01:
- Het staat waarschijnlijk op je meter of het v4 of v5 is. Maar je kunt het ook zien aan de interval van telegramen. Weet je of je elke seconde een meting krijgt of elke 10 seconden?

- Verder kun je proberen om alle processen even te herstarten. Wellicht zijn die niet goed opgekomen:

```
sudo supervisorctl restart all
```

style2k, 2020-03-01:
staat nix op de meter . en ik kom nergens op dashboard en als ik wat anders kies blijft die zoeken . kom nergens anders op ook zie ik op het dashboard dat ik geen gegevens heb onderaan waar staat wat de data is tot vandaag . allemaal leeg

dennissiemensma, 2020-03-01:
Dat kan kloppen als er nog onverwerkte telegrammen zijn. Heb je inmiddels wel alle processen herstart voor de zekerheid?

dennissiemensma, 2020-03-01:
Wat krijg je te zien als je dit doet:

```
sudo su - postgres
psql dsmrreader

select count(id) from dsmr_datalogger_dsmrreading where processed = false;
```

style2k, 2020-03-01:
dsmrreader=# select count(id) from dsmr_datalogger_dsmrreading where processed = false;
 count 
-------
  4605
(1 row)

dsmrreader=#

dennissiemensma, 2020-03-01:
Als als je die query een paar keer uitvoert (pijltje omhoog en dan enter), zie je dat het aantal afneemt? Of neemt het alleen maar toe?

style2k, 2020-03-01:
ja omlaag dan weer iets omhoog en dan weer laag

dennissiemensma, 2020-03-01:
Wat krijg je hier voor output te zien:

```
sudo su - postgres
psql dsmrreader

select id, timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 5;
````

style2k, 2020-03-01:
4584
(1 row)

dsmrreader=# select id, timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 5;
   id    |       timestamp        
---------+------------------------
 3585657 | 2020-03-01 20:43:55+01
 3585656 | 2020-03-01 20:43:48+01
 3585655 | 2020-03-01 20:43:41+01
 3585654 | 2020-03-01 20:43:34+01
 3585653 | 2020-03-01 20:43:27+01
(5 rows)

dsmrreader=#

dennissiemensma, 2020-03-01:
WIl je eens kijken of er goede indexes op de tabel staan?

```
sudo su - postgres
psql dsmrreader

\dS dsmr_datalogger_dsmrreading
```

Het gaat om het blok onderaan. Dat moet iets zijn als:

```
Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_9d090a40" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1_uniq" btree ("timestamp")
```

style2k, 2020-03-01:
Table "public.dsmr_datalogger_dsmrreading"
             Column              |           Type           | Collation | Nullable |                         Default                         
---------------------------------+--------------------------+-----------+----------+---------------------------------------------------------
 id                              | integer                  |           | not null | nextval('dsmr_datalogger_dsmrreading_id_seq'::regclass)
 timestamp                       | timestamp with time zone |           | not null | 
 electricity_delivered_1         | numeric(9,3)             |           | not null | 
 electricity_returned_1          | numeric(9,3)             |           | not null | 
 electricity_delivered_2         | numeric(9,3)             |           | not null | 
 electricity_returned_2          | numeric(9,3)             |           | not null | 
 electricity_currently_delivered | numeric(9,3)             |           | not null | 
 electricity_currently_returned  | numeric(9,3)             |           | not null | 
 extra_device_timestamp          | timestamp with time zone |           |          | 
 extra_device_delivered          | numeric(9,3)             |           |          | 
 processed                       | boolean                  |           | not null | 
 phase_currently_delivered_l1    | numeric(9,3)             |           |          | 
 phase_currently_delivered_l2    | numeric(9,3)             |           |          | 
 phase_currently_delivered_l3    | numeric(9,3)             |           |          | 
 phase_currently_returned_l1     | numeric(9,3)             |           |          | 
 phase_currently_returned_l2     | numeric(9,3)             |           |          | 
 phase_currently_returned_l3     | numeric(9,3)             |           |          | 
 phase_voltage_l1                | numeric(4,1)             |           |          | 
 phase_voltage_l2                | numeric(4,1)             |           |          | 
 phase_voltage_l3                | numeric(4,1)             |           |          | 
 phase_power_current_l1          | integer                  |           |          | 
 phase_power_current_l2          | integer                  |           |          | 
 phase_power_current_l3          | integer                  |           |          | 

~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
(END)

dennissiemensma, 2020-03-01:
Als je naar beneden scrollt, wat staat er in het blokje 'Indexes'? Ik hoef alleen dat te weten.

style2k, 2020-03-01:
ja verder naar beneden kom ik niet

dennissiemensma, 2020-03-01:
Vermoedelijk komt het doordat er een achterstand is met het verwerken van metingen, dus dat probeer ik voor je op te lossen.

dennissiemensma, 2020-03-01:
Zie je wel meer output als je het op deze manier probeert?

```
sudo su - postgres

psql -d dsmrreader -c '\dS dsmr_datalogger_dsmrreading'
```

style2k, 2020-03-01:
nee ook niet stopt ook na curent

dennissiemensma, 2020-03-01:
Oke, nog een laatste variant:

```
sudo su - postgres
psql dsmrreader

SELECT indexname, indexdef FROM pg_indexes WHERE tablename = 'dsmr_datalogger_dsmrreading';
```

style2k, 2020-03-01:
dsmrreader=# SELECT indexname, indexdef FROM pg_indexes WHERE tablename = 'dsmr_datalogger_dsmrreading';
 indexname | indexdef 
-----------+----------
(0 rows)

dsmrreader=#

dennissiemensma, 2020-03-01:
Top. Dan is het duidelijk wat de oorzaak is en daar is wat aan te doen. 

Wil je beginnen met kijken of alle migraties uitgevoerd zijn, dit kan met:

```
sudo su - dsmr
./manage.py showmigrations --list
```

Je hoeft er alleen even doorheen te scrollen en te kijken of je regels ziet met:
```
[ ] naam ...
ipv
[X] naam ...
```

Als er overal een `[X]` voor staat dan is het goed.

style2k, 2020-03-01:
ja 
 ] 0013_process_sleep
[ ] 0014_day_total_cost_index
 [ ] 0005_weather_refactor_scheduling
 [ ] 0006_schedule_weather_update
de rest staat een x voor

dennissiemensma, 2020-03-01:
In dat geval mag je nog eerst dit doen:

```
sudo su - dsmr
./manage.py migrate
```

Daarna zou je geen migraties meer moeten zien zonder `[X]`.

style2k, 2020-03-01:
als ik hetzelfde doe zie ik nog steeds die vakjes zonder x

dennissiemensma, 2020-03-01:
Krijg je wel output te zien als je die `./manage.py migrate` uitvoert?

style2k, 2020-03-01:
mrreader) dsmr@raspberrypi:~/dsmr-reader $ ./manage.py migrate
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  Applying dsmr_mqtt.0013_process_sleep...Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 328, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 369, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 83, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/commands/migrate.py", line 233, in handle
    fake_initial=fake_initial,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 117, in migrate
    state = self._migrate_all_forwards(state, plan, full_plan, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 147, in _migrate_all_forwards
    state = self.apply_migration(state, migration, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 245, in apply_migration
    state = migration.apply(state, schema_editor)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/migration.py", line 124, in apply
    operation.database_forwards(self.app_label, schema_editor, old_state, project_state)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/operations/special.py", line 190, in database_forwards
    self.code(from_state.apps, schema_editor)
  File "/home/dsmr/dsmr-reader/dsmr_mqtt/migrations/0013_process_sleep.py", line 20, in migrate_forward
    process_sleep=decimal.Decimal(settings.DSMRREADER_MQTT_SLEEP)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/conf/__init__.py", line 77, in __getattr__
    val = getattr(self._wrapped, name)
AttributeError: 'Settings' object has no attribute 'DSMRREADER_MQTT_SLEEP'

dennissiemensma, 2020-03-01:
Ik denk dat het goed is om aan te geven als je zulke fouten ziet, want er is dus nog meer aan de hand. Het command crasht namelijk hier.

Open even je `dsmrreader/settings.py` (`/home/dsmr/dsmr-reader/dsmrreader/settings.py`) en verwijder de regel met `DSMRREADER_MQTT_SLEEP = ...`

Probeer het daarna nogmaals met migrate.

style2k, 2020-03-01:
DSMRREADER_DATALOGGER_SLEEP = 5
deze regel bedoel je ? die hebben we er volgens mij toen de 1e keer ingezet toen ik die problemen had

dennissiemensma, 2020-03-01:
Er staan meerdere in vermoed ik. De foutmelding hierboven gaat over `DSMRREADER_MQTT_SLEEP`.

Ze moeten er allemaal uit. 

Je kunt die sleep(s) nu instellen via de admin-interface. 
- `/admin/dsmr_datalogger/dataloggersettings/1/change/`
- `/admin/dsmr_mqtt/mqttbrokersettings/` (onderaan)

dennissiemensma, 2020-03-01:
Ik vermoed dat de applicatie het nu ook niet lekker doet door die config. Dat is in latere versies veranderd namelijk. Besef wel dat je opeens zo'n 15 versies verder bent, met alle wijzigingen van dien.

Als je de settings.py hebt aangepast, kun je dat migrate-command weer proberen.

style2k, 2020-03-01:
nee alleen die ik noemde staat er in echt nix anders met sleep

dennissiemensma, 2020-03-01:
Prima. Werkt de `migratie` nu wel zonder fouten?

style2k, 2020-03-01:
kan het setting bestand niet wegschrijven

dennissiemensma, 2020-03-01:
Ben je wel `dsmr` gebruiker?

```
sudo su - dsmr
vi dsmrreader/settings.py
```

dennissiemensma, 2020-03-01:
En anders even kijken wat de permissies zijn met:

```
ls -l /home/dsmr/dsmr-reader/dsmrreader/settings.py
```

style2k, 2020-03-01:
-rw-r--r-- 1 dsmr dsmr 1502 Oct  8 22:31 /home/dsmr/dsmr-reader/dsmrreader/settings.py

wil de file gewoon kunne nediten vanuit de txt editor van de raspian desktop ik ben gewoon admin dus snap nie dat ik geen rechten heb in de dsmr folder heb

dennissiemensma, 2020-03-01:
Ik kan je daar verder niet bij helpen. Mijn advies is om een tekst-editor op de commandline aan te leren. `vi` is wellicht wat lastig, maar `nano` is al stukken makkelijker:

```
sudo su - dsmr
nano dsmrreader/settings.py
```

Je kunt na de wijzigingen opslaan met CTRL + X dacht ik (staat onderaan).

style2k, 2020-03-01:
ok gelukt overal staat nu een x voor

dennissiemensma, 2020-03-01:
Wil je nu proberen alles te herstarten om te kijken of de interface het daarmee weer doet?

```
sudo supervisorctl restart all
```

style2k, 2020-03-01:
ja de interface deed het voor zover wel hoor dashbord maar als ik verder ga nix nu kom ik wel op bijv archief maar hij blijft laden
[image]

wel is op het dashbord nog steeds geen info over verbruik te zien 

[image]

en weet dus niet of het komt doordat die nog data aan verwerken is dat het allemaal zo traag gaat

dennissiemensma, 2020-03-01:
Zorg er in ieder geval voor dat de datalogger weer op 5 of 10 seconden sleep staat. Dat regel je nu via de webinterface:

[image]

Daarna:
```
sudo supervisorctl restart dsmr_datalogger
```

dennissiemensma, 2020-03-01:
Ik stel voor om dan morgen verder te gaan. Kijk wel even of de datalogger ondertussen nieuwe data in je database opslaat:

```
sudo su - postgres
psql dsmrreader

select id, timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 5;
```

Die tijdstippen moeten dan recent zijn.

style2k, 2020-03-01:
ok instellingen veranderd en op 10 gezet gaat nu wat vlotter lijkt . alleen kan ik van maart nog geen gegevens zien in archief van vandaag.en niet op het dashbord terwijl die met verwerken al bij 12.50 vanmiddag zit ofzo

style2k, 2020-03-02:
het is weer bijgewerkt nu . het laden van de site is ook wat vlotter maar nog steeds een stuk trager dan voorheen . maar de data is weer bij dus

dennissiemensma, 2020-03-02:
Dat is fijn om te horen. Op de eerste dag van de maand is er nooit een Archief beschikbaar voor die maand. Ik zal kijken of ik daar een melding of iets voor kan maken.

dennissiemensma, 2020-03-02:
Wat betreft de performance, kun je eens kijken wat de load op je Pi is? Bijvoorbeeld met `top` of `htop`?

style2k, 2020-03-02:
ja cpu gaat redelijk op en neer van 9 procent to 99 dus, wat me ook opvalt is dat in dashboard het reente gebruik van vandaag hoger is als het totaal gebruik van vandaag en gisteren

dennissiemensma, 2020-03-02:
- Kun je ook zien welk proces zoveel CPU inneemt?
- Ik begrijp niet helemaal wat je bedoelt qua verbruik.

style2k, 2020-03-02:
postgres gebruikt meeste . komt ook meeste voor . staat verder ook nix op de pi is standaard pi met dsrm verder nix . ja dingetje voor remote desktop maar stond er al lang op ook toen het wel sneller ging. verder lege setup.

[image]

o nee zie het al is gebruik deze maand tot en met gisteren vandaag zit daar nog niet bij dus  dat zal wel kloppen :)

dennissiemensma, 2020-03-02:
Ik denk dat die hoge CPU komt omdat er indexes missen. Je kunt proberen die nu alsnog toe te voegen:

```
sudo su - postgres
psql dsmrreader

ALTER TABLE public.dsmr_datalogger_dsmrreading ADD CONSTRAINT dsmr_datalogger_dsmrreading_pk PRIMARY KEY (id);

CREATE INDEX dsmr_datalogger_dsmrreading_processed_add8a852 ON public.dsmr_datalogger_dsmrreading USING btree (processed);

CREATE INDEX dsmr_datalogger_dsmrreading_timestamp_1f7020d1 ON public.dsmr_datalogger_dsmrreading USING btree (timestamp);
```

dennissiemensma, 2020-03-02:
Het toevoegen van een index kan overigens even duren, afhankelijk van hoeveel metingen er in staan.

style2k, 2020-03-02:
htop lijkt rustiger nu. alleen swp wat dat ook is staat op 100 procent tenminste die balk is zeg maar rood 100 procent. maar de website laad nu een stuk sneller idd

dennissiemensma, 2020-03-02:
Je kunt nog even de Pi herstarten, zodat de swap weer helemaal leeg is en alles vers is. 

De swap wordt gebruikt wanneer er te weinig geheugen is en staat op de disk. Dit is een noodgreep van het OS. Mogelijk kwam dat door de missende index.

style2k, 2020-03-02:
ja ff rebooten nog ga er van uit dat dat ook verholpen is dan

dennissiemensma, 2020-03-02:
Mooi. Mocht je nog tegen iets aanlopen dan hoor ik het graag.

closed 2020-03-02 by dennissiemensma