dsmrreader/dsmr-reader #915: Linux

Ferryb, 2020-03-02:
Hee Dennis compimenten voor je werkzaamheden.

Ik wil overstappen van mijn pi naar een NUC zodat mijn data minder gevaar loopt.
Draait de DSMR-reader ook op Ubuntu ?

dennissiemensma, 2020-03-02:
Hallo Ferry. Je kunt DSMR-reader zeker op Ubuntu draaien. 

Ik ontwikkel het namelijk ook op Ubuntu en verwacht dat het op meerdere distro's goed draait. Keuze genoeg dus!

Ferryb, 2020-03-03:
OK dank je. Dan ga ik even verder met puzzelen want krijg het niet voor elkaar :(

dennissiemensma, 2020-03-03:
Kun je aangeven waar je tegenaan loopt? Krijg je ergens een foutmelding?

Ferryb, 2020-03-03:
Ja klopt krijg een foutmelding.
Het gata goed tot ik bij onderstaande aangkomen ben.

ferry@NUC:~$ sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 insta                                                                                                                                                             ll -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt -r /h                                                                                                                                                             ome/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/postgresql.txt
WARNING: The directory '/home/ferry/.cache/pip' or its parent directory is not o                                                                                                                                                             wned or is not writable by the current user. The cache has been disabled. Check                                                                                                                                                              the permissions and owner of that directory. If executing pip with sudo, you may                                                                                                                                                              want sudo's -H flag.
Collecting coreapi==2.3.3
  Downloading coreapi-2.3.3-py2.py3-none-any.whl (25 kB)
Collecting crcmod==1.7
  Downloading crcmod-1.7.tar.gz (89 kB)
     |████████████████████████████████| 89 kB 11.2 MB/s
ERROR: Could not find a version that satisfies the requirement django==3.0.3 (fr                                                                                                                                                             om -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt (line                                                                                                                                                              3)) (from versions: 1.1.3, 1.1.4, 1.2, 1.2.1, 1.2.2, 1.2.3, 1.2.4, 1.2.5, 1.2.6                                                                                                                                                             , 1.2.7, 1.3, 1.3.1, 1.3.2, 1.3.3, 1.3.4, 1.3.5, 1.3.6, 1.3.7, 1.4, 1.4.1, 1.4.2                                                                                                                                                             , 1.4.3, 1.4.4, 1.4.5, 1.4.6, 1.4.7, 1.4.8, 1.4.9, 1.4.10, 1.4.11, 1.4.12, 1.4.1                                                                                                                                                             3, 1.4.14, 1.4.15, 1.4.16, 1.4.17, 1.4.18, 1.4.19, 1.4.20, 1.4.21, 1.4.22, 1.5,                                                                                                                                                              1.5.1, 1.5.2, 1.5.3, 1.5.4, 1.5.5, 1.5.6, 1.5.7, 1.5.8, 1.5.9, 1.5.10, 1.5.11, 1                                                                                                                                                             .5.12, 1.6, 1.6.1, 1.6.2, 1.6.3, 1.6.4, 1.6.5, 1.6.6, 1.6.7, 1.6.8, 1.6.9, 1.6.1                                                                                                                                                             0, 1.6.11, 1.7, 1.7.1, 1.7.2, 1.7.3, 1.7.4, 1.7.5, 1.7.6, 1.7.7, 1.7.8, 1.7.9, 1                                                                                                                                                             .7.10, 1.7.11, 1.8a1, 1.8b1, 1.8b2, 1.8rc1, 1.8, 1.8.1, 1.8.2, 1.8.3, 1.8.4, 1.8                                                                                                                                                             .5, 1.8.6, 1.8.7, 1.8.8, 1.8.9, 1.8.10, 1.8.11, 1.8.12, 1.8.13, 1.8.14, 1.8.15,                                                                                                                                                              1.8.16, 1.8.17, 1.8.18, 1.8.19, 1.9a1, 1.9b1, 1.9rc1, 1.9rc2, 1.9, 1.9.1, 1.9.2,                                                                                                                                                              1.9.3, 1.9.4, 1.9.5, 1.9.6, 1.9.7, 1.9.8, 1.9.9, 1.9.10, 1.9.11, 1.9.12, 1.9.13                                                                                                                                                             , 1.10a1, 1.10b1, 1.10rc1, 1.10, 1.10.1, 1.10.2, 1.10.3, 1.10.4, 1.10.5, 1.10.6,                                                                                                                                                              1.10.7, 1.10.8, 1.11a1, 1.11b1, 1.11rc1, 1.11, 1.11.1, 1.11.2, 1.11.3, 1.11.4,                                                                                                                                                              1.11.5, 1.11.6, 1.11.7, 1.11.8, 1.11.9, 1.11.10, 1.11.11, 1.11.12, 1.11.13, 1.11                                                                                                                                                             .14, 1.11.15, 1.11.16, 1.11.17, 1.11.18, 1.11.20, 1.11.21, 1.11.22, 1.11.23, 1.1                                                             [... cut ...]                                                                                                                                         2.2.2, 2.2.3, 2.2.4, 2.2.5, 2.2.6, 2.2.7, 2.2.8, 2.2.9, 2.2.10)
ERROR: No matching distribution found for django==3.0.3 (from -r /home/dsmr/dsmr                                                                                                                                                             -reader/dsmrreader/provisioning/requirements/base.txt (line 3))

dennissiemensma, 2020-03-03:
Wat krijg je hier te zien:

```
sudo su - dsmr
python3 --version
```

Ferryb, 2020-03-03:
ferry@NUC:~$ sudo su - dsmr
(dsmrreader) dsmr@NUC:~/dsmr-reader$ python3 --version
Python 3.5.2

dennissiemensma, 2020-03-03:
DSMR-reader v3.0+ ondersteunt alleen Python 3.6 of hoger. Je zult die (of hoger) eerst moeten installeren.

Je kunt waarschijnlijk gewoon stap 3 en 5 uitvoeren van deze pagina:
https://dsmr-reader.readthedocs.io/en/v3/faq/v3_upgrade.html#python-version-upgrade-part-1-2

dennissiemensma, 2020-03-03:
Als alternatief kun je overigens ook de hele homedir van DSMR-reader weggooien, Python 3.6+ installeren en dan de installatie helemaal opnieuw doen. Net wat je voorkeur heeft.

dennissiemensma, 2020-03-03:
(er vanuitgaande dat dit een verse installatie was zonder bestaande data/settings of iets wat je wilt behouden)

Ferryb, 2020-03-03:
Dank je het is me uiteindelijk gelukt :)

dennissiemensma, 2020-03-03:
Fijn om te horen, dank voor de terugkoppeling.

closed 2020-03-03 by dennissiemensma