dsmrreader/dsmr-reader #916: extra API velden voor /api/v2/consumption/today

braham2019, 2020-03-03:
Is het mogelijk om aan de API /api/v2/consumption/today twee velden toe te voegen ?
- een veld dat electricity1 en electricity2 opgeteld weergeeft
- een veld dat electricity1_returned en electricity2_returned opgeteld weergeeft 

Ik gebruik de API om de info op mijn DAKboard scherm te tonen van verbruik en teruggave, maar daar staan nu vier waarden (omwille van tarief 1 en 2); ik zou er liever twee hebben, als het zou kunnen. DAKboard laat niet toe dat ik die zelf optel alvorens ik ze weergeef :-(

Bedankt voor de nuttige toepassing!

dennissiemensma, 2020-03-03:
Bedankt voor je verzoek. Ik zal t.z.t. kijken of ik dat kan toevoegen.

dennissiemensma, 2020-03-03:
Ik heb even voor je gekeken en de data was al praktisch aanwezig. Echter verborg ik deze hardcoded, dus ik zal hem toevoegen en ook de docs bijwerken.

dennissiemensma, 2020-03-03:
Gaat mee in de volgende release.

closed 2020-03-03 by dennissiemensma

dennissiemensma, 2020-03-06:
Uitgebracht in [v3.6](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.6.0)

braham2019, 2020-03-06:
Prachtig, dankjewel !