dsmrreader/dsmr-reader #917: Upgrade error and Fresh installation error

jossie67, 2020-03-04:
Laaste versie 
Raspberry3 / Ubuntu
DSMR Protocol v4

Vandaag mijn dsmr-reader geupgrade, na de upgrade een error

"Server Error
Sorry, something unexpected happened.
Exception:
ProgrammingError: relation "dsmr_consumption_electricityconsumption" does not exist
LINE 1: SELECT (1) AS "a" FROM "dsmr_consumption_electricityconsumpt..."

Ik dacht ligt aan mij SD formateren, versie installatie , database restoren

Weer de zelfde foutmelding.....

Ik nu even een verse installatie op Ubuntu zonder restore...

En weer de zelfde foutmedling ..
Zit er een fout in de repository?

dennissiemensma, 2020-03-04:
Bedankt voor je melding. Wat krijg je hier te zien?

```
sudo su - dsmr
python3 --version
```

En kun je daarna dit eens proberen:
```
sudo su - dsmr
./manage.py migrate
```

jossie67, 2020-03-06:
Dennis,
Thanks for you replay, I checked and recheck, I found the issue.
Op de com-port /dev/ttyUSB0 was de " ir-schreib-lesekopf" aangesloten ivm de P1
Door de /dev/ttyUSB0  te veranderen in /dev/ttyUSB1 werkte alles weer.
Weer een leerpuntje

closed 2020-03-06 by jossie67

dennissiemensma, 2020-03-06:
Top, bedankt voor je terugkoppeling.