dsmrreader/dsmr-reader #919: Parsing telegram 3-fasige Fluvius meter faalt

hollie, 2020-03-05:
**Wat gebruik je?**
* DSMR-reader versie: v3.5
* Hardware: RaspberryPi 3B+
* Native of Docker: Docker
* DSMR-protocol slimme meter: v5

**Situatie**
Als ik de parsing van de telegrammen op `Nederlands DSMR versie 4/5` zet wordt de data geparsed maar zijn de tijden piek/dal omgewisseld zoals eerder gemeld in vorige tickets.

Als ik de parsing instel op `België / Fluvius` worden de telegrammen niet geparsed.

Het is me niet duidelijk hoe ik kan achterhalen wat er precies verkeerd loopt.

PS: Bedankt om deze software te ontwikkelen en te delen!

**Voorbeeld telegram**
```
/FLU5\253769484_A

0-0:96.1.4(50213)
0-0:96.1.1(3153414733313030303635363830)
0-0:1.0.0(200305222945W)
1-0:1.8.1(000172.987*kWh)
1-0:1.8.2(000160.643*kWh)
1-0:2.8.1(000023.457*kWh)
1-0:2.8.2(000004.819*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.638*kW)
1-0:2.7.0(00.000*kW)
1-0:32.7.0(230.3*V)
1-0:52.7.0(230.5*V)
1-0:72.7.0(229.3*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(001*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(37464C4F32313230303031343332)
0-1:24.4.0(0)
0-1:24.2.3(632525252525W)(00000.000)
!186C
``` 

Ik wil dit graag verder helpen debuggen maar het is me niet helemaal duidelijk waar te beginnen. Tips zijn welkom!

dennissiemensma, 2020-03-06:
Bedankt voor je melding. Ik zal straks even kijken naar je telegram.

Als de Nederlandse instellingen voor je werken, raad ik je aan om die gewoon te gebruiken. Sinds DSMR-reader `v3.4.0` kun je zelf de tarieven naar wens hernoemen en dus ook omdraaien (of wijzigen naar "dag- en nachttarief").

dennissiemensma, 2020-03-06:
Klopt het trouwens dat je geen gasmeter hebt? Of is de gasstand inderdaad `0`?

dennissiemensma, 2020-03-06:
De reden dat de Fluvius instelling niet werkt is doordat de gasmeter een ongeldige datum/tijd doorgeeft `632525252525W`. Ik heb dat vaker gezien, maar geen directe oplossing voor.

Doordat de Nederlandse instelling **niet** je gasmeter uitleest, gaat dit 'goed'.

dennissiemensma, 2020-03-06:
Ik heb hier een fix voor gemaakt voor de volgende release.

closed 2020-03-06 by dennissiemensma

hollie, 2020-03-06:
Hey Dennis, 

bedankt om dit te bekijken en een fix te voorzien.

Ik heb wel een gasmeter maar er is een probleem mee, alleen weet Fluvius nog niet wat het probleem precies is... Er komen wmbus pakketten uit de gasmeter (die heb ik zelf ontvangen, maar ik ken de key natuurlijk niet om ze te decoderen). Ik denk dat er iets misgelopen is met de koppeling tijdens installatie. Moet dus nog verder bekeken worden...

In ieder geval: het heeft inderdaad geen zin om het hele pakket invalid te verklaren alleen omdat het timefield van de gasmeter niet correct is. 

Alvast een fijn weekend!
 Lieven.

dennissiemensma, 2020-03-19:
Uitgebracht in [v3.7](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.7.0)

hollie, 2020-03-20:
Dag @dennissiemensma bedankt, de parsing van de packets loopt nu zoals verwacht. Koffie/biertje verstuurd via paypal, enjoy!