dsmrreader/dsmr-reader #921: Feature request: notificaties zonder login zien

neverstoppedaskingwhy, 2020-03-07:
Hi Dennis,

In oudere 2.x releases was de inhoud van de notificaties op het dashboard nog zichtbaar. Nu is er een admin login voor nodig. Graag zou ik weer in één blik op het dashboard zonder login zien, of het zich bij de notificatie bijv. om een buienradar API fout gaat, of dat er een nieuwe versie is. 

Misschien is dit ook al ergens instelbaar?

Groetjes,
Mark

dennissiemensma, 2020-03-09:
Bedankt voor je verzoek. Ik heb ze ooit achter de login gezet omdat ik niet zeker wist of er wel of geen gevoelige info in stond. 
Echter, als iemand DSMR-reader bewust open zet voor het Internet, is het een eigen keuze, dus ik zal de login-vereiste weer weghalen.

dennissiemensma, 2020-03-09:
Aangepast. Voor het markeren als gelezen moet je straks nog wel inloggen. Dat was eerder ook zo.

closed 2020-03-09 by dennissiemensma

dennissiemensma, 2020-03-19:
Uitgebracht in [v3.7](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.7.0)