dsmrreader/dsmr-reader #922: DSMR - Fase 1 teruglevering

niekniek89, 2020-03-10:
Goedemorgen,

Ten 1e wil ik melden dat ik sinds kort DSMR reader naar volle tevredenheid heb draaien!!
Mijn vraag is als volgt (weet ook niet of dit mogelijk is):

ik heb momenteel zonnepanelen, waarvan ik de omvormer niet kan uitlezen.
echter zit mijn omvormer op fase1 aangesloten, waardoor ik in DSMR reader zie dat ik over 1 fase terug lever.
is er toevallig een optie dat ik specifiek van fase 1 kan bijhouden per dag, hoeveel dat er op fase 1 is binnengekomen aan terug geleverde kwh?

ik zou dat alleen van fase 1 willen weten (ik kan het nu live zien), maar helaas niet terug zien in history.

bij voorbaat dank

dennissiemensma, 2020-03-10:
Bedankt voor je vraag. Voor zover ik weet houdt de slimme meter dit niet voor je bij per fase.

Wat je nog zou kunnen doen is een meter tussen je omvormer en de fase-aansluiting zetten, maar daar heb ik zelf geen ervaring mee.

dennissiemensma, 2020-03-10:
Ik denk dat je het beste je vraagstuk op het Tweakers-forum kunt leggen. Doorgaans hebben anderen al het (zelfde) probleem opgelost of een workaround. Wellicht is dit topic wat: 

https://gathering.tweakers.net/forum/list_messages/1489893/0

closed 2020-03-10 by dennissiemensma

niekniek89, 2020-03-13:
Bedankt voor je snelle reactie!
En excuses voor mijn late reactie haha.
oke, bedankt voor je tip.