dsmrreader/dsmr-reader #923: Backups compressie level configureerbaar maken

jorkzijlstra, 2020-03-12:
Ik gebruik de dropbox backup waarbij ik een normaal 2Gb limit heb (betaal er niet voor) en inmiddels zijn mijn backups zo groot geworden dat ik over deze grens heen ga.

Nu zag ik dat de compressie level aangepast kan worden in:
https://github.com/dennissiemensma/dsmr-reader/blob/v3/dsmr_backup/services/backup.py#L213

Ik heb deze lokaal even naar 9 gezet en de backup wordt hierdoor 30% kleiner. Uiteraard kost dit wel meer cpu gebruik.

[image]

Het zou dus handig zijn al dit een setting / environmental variable (voor docker) kan worden zodat iedereen de trade off kan maken tussen cpu gebruik and backup grootte.

dennissiemensma, 2020-03-12:
Toegevoegd voor volgende release.

closed 2020-03-12 by dennissiemensma

jorkzijlstra, 2020-03-12:
Heel erg bedankt voor het snelle werk

dennissiemensma, 2020-03-19:
Uitgebracht in [v3.7](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.7.0)