dsmrreader/dsmr-reader #924: verse installatie

Robert126, 2020-03-13:
Hi Dennis,

Ik heb een backup .csv file. Is het mogelijk om deze in mijn nieuwe installatie in te laden?

Alvast dank voor je hulp!

Hartelijke groet,
Robert

dennissiemensma, 2020-03-13:
Er zit nog geen ingebouwde ondersteuning voor in, maar ik kan hem wel evt voor je omzetten naar een SQL-file die je kunt inlezen. Je mag de CSV dan mailen naar github @ dennissiemensma . nl

Of je moet zelf handig zijn met find-replace naar SQL. Het zijn in principe alleen INSERT-queries voor 1 tabel.

dennissiemensma, 2020-03-13:
Zoals besproken zal ik nog iets maken waarmee de dagtotalen met terugwerkende kracht berekend kunnen worden.

dennissiemensma, 2020-03-13:
Verder voor mezelf de notitie dat het direct inlezen van CSV's ook makkelijk gaat in Postgresql (in het verleden deed ik dus een find-replace om per regel questies te maken.

```
copy dsmr_stats_hourstatistics (hour_start, electricity1, electricity2, electricity1_returned, electricity2_returned, gas) FROM '/var/lib/postgresql/uren.csv' WITH (FORMAT csv);
```

Robert126, 2020-03-13:
Hi Dennis,

De import heb ik gedaan. Met de volgende terugkoppeling COPY 5391 verder geen fout melding.

Ik zie alleen onder het kopje Statistics 

METER POSITIONS
25 seconds from now	Consumed	
Laagtarief kWh	2,489.306 	
Hoogtarief kWh	2,307.666	
Electricity (combined) kWh	4,796.972	
Gas m3	1,117.145
METER STATISTICS
Number of power failures in any phase	8
Number of long power failures in any phase	3
Number of voltage sags/dips in phase L1	6
Number of voltage sags/dips in phase L2 (polyphase meters only)	-
Number of voltage sags/dips in phase L3 (polyphase meters only)	-
Number of voltage swells in phase L1	1
Number of voltage swells in phase L2 (polyphase meters only)	-
Number of voltage swells in phase L3 (polyphase meters only)	-
TELEGRAM READINGS
Total reading count	   Show count <http://192.168.178.30:8080/statistics#>
ALL TIME HIGHS & LOWS

Highest consumption (L1+)	 3,016 Watt 
Friday March 13th 2020 at 7:15 p.m.
Lowest consumption (L1+)	 104 Watt 
Friday March 13th 2020 at 5:56 p.m.

Veder zie ik de historie niet terug.
Dank tot dus ver!!
Fijne avond en goed weekend!
Robert

dennissiemensma, 2020-03-14:
In DSMR-reader 3.7 kun je dit straks zelf oplossen door het volgende uit te voeren:

```
sudo su - dsmr

./manage.py dsmr_stats_reconstruct_missing_day_statistics
./manage.py dsmr_stats_recalculate_prices
```

Ik laat het nog weten wanneer die release beschikbaar is.

closed 2020-03-14 by dennissiemensma

Robert126, 2020-03-14:
Hi Dennis,

Hartelijke dank voor je prettige ondersteuning!

Fijn weekend!

Robert

dennissiemensma, 2020-03-19:
Uitgebracht in [v3.7](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.7.0)

Robert126, 2020-03-24:
Hi Dennis,

Als eerste super dat je zo snel al een nieuwe versie hebt kunnen opleveren!

Sorry voor mijn late reactie. Even druk geweest met mantel zorg vandaar.

Ik heb het volgende gedaan

Al eerste alles verwijderd:
https://dsmr-reader.readthedocs.io/nl/v3/faq/uninstall.html#faq-uninstalling-the-application <https://dsmr-reader.readthedocs.io/nl/v3/faq/uninstall.html#faq-uninstalling-the-application>

Daarna weer een schone installatie willen uitvoeren:
https://dsmr-reader.readthedocs.io/nl/v3/installation/quick.html <https://dsmr-reader.readthedocs.io/nl/v3/installation/quick.html>
Stuite allen op het volgende 
sudo sudo -u postgres createuser -DSR dsmrreader
createuser: creation of new role failed: ERROR:  role "dsmrreader" already exists

Enig advies hoe ik de huidige gebruiker kan verwijderen, zodat ik een schone nieuwe installatie kan uitvoeren.

Hartelikke groet,
Robert

dennissiemensma, 2020-03-24:
Het is zonde dat je dit hebt gedaan, want het upgraden is slechts 1 command uitvoeren en daar hoeft je applicatie niet voor down of verwijderd te worden:

https://dsmr-reader.readthedocs.io/nl/v3/faq/update.html

Je hoeft geen nieuwe gebruiker aan te maken als die al bestaat. Hetzelfde geldt voor de database.

Robert126, 2020-03-24:
Hi Dennis,

Er was kennelijk iets mis gegaan met het wachtwoord.
Heb je een tip hoe ik de user volledig kan verwijderen?

Hartelijke groet,
Robert

dennissiemensma, 2020-03-24:
Werkt dit niet om het wachtwoord te resetten?

```
sudo sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"
```

dennissiemensma, 2020-03-24:
En anders kun je hem verwijderen met
```
sudo sudo -u postgres dropuser dsmrreader
```

Robert126, 2020-03-25:
Hi Dennis,

Dank voor je input!

Het systeem draaid weer :-)

Ik ben al even aan het zoeken maar hoe kan ik nu de “oude” backup dsmrreader-data-export—hour_2019-03-19.csv terug zetten in het systeem.

En volgens mij zijn de getalen die er nu instaan te groot zie screen shot.

Ik hoor graag van je!

Hartelijk dank voor je hulp!

Robert

dennissiemensma, 2020-03-27:
Ik ben even kwijt wat ik je toegestuurd had. Was dat een losse mail met queries? Eventuele uitgaande mails (en inkomende bijlages) heb ik verwijderd omdat het niet mijn gegevens zijn, dus als ik je iets gemaild had qua gegevens of instructies heb ik dat wellicht ook niet meer.