dsmrreader/dsmr-reader #925: Bad Gateway 502

TheMClaeys, 2020-03-14:
So, this morning I was trying to log in on dsmr as I was working on my Home Assistant and wanted to set it up to forward info over mqtt. I was welcomed with a 502 Bad Gateway from Nginx and I don't really understand why. Can someone assist in troubleshooting. It's the second time it happened, the first time I messed up and lost all my data, I want to prevent that this time (I had backup but I couldn't import it anymore, I got an error when I followed the instructions on the site then).

My Hardware is a Raspberry Pi 3B+

So, a 502 Bad Gateway error indicates an issue with the page behind it, and the info seems to indicate this also:

> sudo supervisorctl
> dsmr_backend                     BACKOFF   Exited too quickly (process log may have details)
> dsmr_datalogger                  BACKOFF   Exited too quickly (process log may have details)
> dsmr_webinterface                FATAL     Exited too quickly (process log may have details)
> supervisor>

When checking the status of Nginx and PostgreSQL I see that nginx is active (running) and Postgresql is active (exited). I couldn't find why it exited.

I tested first with pg_isready if there was response from postgresql.
> pg_isready -h 172.0.0.1
> 172.0.0.1:5432 - no response

Then I checked the logs of Postgresql, they don't really indicate an issue there also (I found incomplete startup packet in the log, but searching I couldn't really find info on it being a breaking error).

> PostgreSQL logs
> 2020-03-08 16:17:21.163 GMT [526] LOG:  listening on IPv6 address "::1", port 5432
> 2020-03-08 16:17:21.163 GMT [526] LOG:  listening on IPv4 address "127.0.0.1", port 5432
> 2020-03-08 16:17:21.169 GMT [526] LOG:  listening on Unix socket "/var/run/postgresql/.s.PGSQL.5432"
> 2020-03-08 16:17:21.294 GMT [527] LOG:  database system was interrupted; last known up at 2020-03-08 16:35:16 GMT
> 2020-03-08 16:17:21.492 GMT [527] LOG:  database system was not properly shut down; automatic recovery in progress
> 2020-03-08 16:17:21.507 GMT [527] LOG:  redo starts at 1/34873978
> 2020-03-08 16:17:21.515 GMT [527] LOG:  invalid record length at 1/348764D8: wanted 24, got 0
> 2020-03-08 16:17:21.515 GMT [527] LOG:  redo done at 1/348764B0
> 2020-03-08 16:17:21.516 GMT [527] LOG:  last completed transaction was at log time 2020-03-08 16:36:16.692984+00
> 2020-03-08 16:17:21.609 GMT [526] LOG:  database system is ready to accept connections
> 2020-03-08 16:17:22.207 GMT [536] [unknown]@[unknown] LOG:  incomplete startup packet
> 2020-03-14 11:41:12.227 GMT [526] LOG:  received fast shutdown request
> 2020-03-14 11:41:12.232 GMT [526] LOG:  aborting any active transactions
> 2020-03-14 11:41:12.247 GMT [526] LOG:  background worker "logical replication launcher" (PID 533) exited with exit code 1
> 2020-03-14 11:41:12.256 GMT [528] LOG:  shutting down
> 2020-03-14 11:41:12.336 GMT [526] LOG:  database system is shut down
> 2020-03-14 11:41:57.263 GMT [517] LOG:  listening on IPv6 address "::1", port 5432
> 2020-03-14 11:41:57.266 GMT [517] LOG:  listening on IPv4 address "127.0.0.1", port 5432
> 2020-03-14 11:41:57.271 GMT [517] LOG:  listening on Unix socket "/var/run/postgresql/.s.PGSQL.5432"
> 2020-03-14 11:41:57.383 GMT [518] LOG:  database system was shut down at 2020-03-14 11:41:12 GMT
> 2020-03-14 11:41:57.435 GMT [517] LOG:  database system is ready to accept connections
> 2020-03-14 11:41:57.991 GMT [525] [unknown]@[unknown] LOG:  incomplete startup packet
> 2020-03-14 12:03:56.029 GMT [517] LOG:  received fast shutdown request
> 2020-03-14 12:03:56.034 GMT [517] LOG:  aborting any active transactions
> 2020-03-14 12:03:56.048 GMT [517] LOG:  background worker "logical replication launcher" (PID 524) exited with exit code 1
> 2020-03-14 12:03:56.059 GMT [519] LOG:  shutting down
> 2020-03-14 12:03:56.125 GMT [517] LOG:  database system is shut down
> 2020-03-14 12:03:56.834 GMT [993] LOG:  listening on IPv6 address "::1", port 5432
> 2020-03-14 12:03:56.834 GMT [993] LOG:  listening on IPv4 address "127.0.0.1", port 5432
> 2020-03-14 12:03:56.839 GMT [993] LOG:  listening on Unix socket "/var/run/postgresql/.s.PGSQL.5432"
> 2020-03-14 12:03:56.904 GMT [994] LOG:  database system was shut down at 2020-03-14 12:03:56 GMT
> 2020-03-14 12:03:56.947 GMT [993] LOG:  database system is ready to accept connections
> 2020-03-14 12:03:57.537 GMT [1001] [unknown]@[unknown] LOG:  incomplete startup packet

There was also a log.1 file, but it only had the following in it:

> 2020-03-07 14:06:23.500 GMT [26908] dsmrreader@dsmrreader LOG:  could not receive data from client: Connection reset by peer

I don't get a lot wiser from the other logs also, but that's mostly since I don't really know what to look for though.

dsmr_backend.log

> The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io$
> [2020-02-11 22:10:50,935] ERROR    dsmr_backend.management.commands.dsmr_backend: [!] Exception raised. Traceback (most recent call last):
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 86, in _execute
>     return self.cursor.execute(sql, params)
> psycopg2.errors.AdminShutdown: terminating connection due to administrator command
> SSL connection has been closed unexpectedly
> 
[... 77 lines cut ...]
> Traceback (most recent call last):
>   File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 127, in telegram_to_reading
>     parsed_telegram = parser.parse(data)
>   File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 43, in parse
>     self.validate_checksum(telegram_data)
>   File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 89, in validate_checksum
>     expected_crc_hex
> dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '64819' (FD33) does not match the telegram checksum$
> [2020-02-15 07:56:08,415] WARNING  @ services | Rejected telegram (Invalid telegram CRC. The calculated checksum '60948' (EE14) does not match$
> [2020-02-15 07:56:08,416] ERROR    @ services | Invalid telegram CRC. The calculated checksum '60948' (EE14) does not match the telegram check$
> Traceback (most recent call last):
>   File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 127, in telegram_to_reading
>     parsed_telegram = parser.parse(data)
>   File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 43, in parse
>     self.validate_checksum(telegram_data)
>   File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 89, in validate_checksum
>     expected_crc_hex
> dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '60948' (EE14) does not match the telegram checksum$
> [2020-02-15 08:06:52,468] WARNING  @ services | Rejected telegram (Invalid telegram CRC. The calculated checksum '14985' (3A89) does not match$
> [2020-02-15 08:06:52,468] ERROR    @ services | Invalid telegram CRC. The calculated checksum '14985' (3A89) does not match the telegram check$
> Traceback (most recent call last):
>   File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 127, in telegram_to_reading
>     parsed_telegram = parser.parse(data)

Any help would be helpfull, I don't really want to lose all the data again like in Januari (that was my fault though). I  checked my backups on Dropbox, it seems like the issue is there since the 8th of March, then was my last backup apparently.

dennissiemensma, 2020-03-14:
It's a bit hard to find the exact cause here. Did you already tried restarting the system? And check whether PostgreSQL starts properly: `sudo systemctl status postgresql`.

When it doesn't, you might want to check whether the SD-card has any corruption. But I'm not sure how to check that properly, as my failing SD-cards so far simply died entirely, so I was not able to start the OS at all.

TheMClaeys, 2020-03-14:
I tried restarting the system yes, however it doesn't load. It starts normally as far as I see, but the page of DSMR itself doesn't load.

Postgresql doesn't load, the error is stated in the first post:

> When checking the status of Nginx and PostgreSQL I see that nginx is active (running) and Postgresql is active (exited). I couldn't find why it exited.
> 
> I tested first with pg_isready if there was response from postgresql.
> 
>     pg_isready -h 172.0.0.1
>     172.0.0.1:5432 - no response

The database is not situated on the SD-Card though, it's on a mounted USB stick. I just checked, it's still mounted (I enabled auto-mount if the system would go down). So I doubt corruption on the SD-card, it's barely running for 3 months.

TheMClaeys, 2020-03-14:
I think the eventual error where it currently strands is the following:

> 
> [2020-03-14 23:12:17 +0100] [7408] [INFO] Starting gunicorn 20.0.4
> [2020-03-14 23:12:17 +0100] [7408] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (7408)
> [2020-03-14 23:12:17 +0100] [7408] [INFO] Using worker: sync
> [2020-03-14 23:12:17 +0100] [7411] [INFO] Booting worker with pid: 7411
> [2020-03-14 23:12:17 +0100] [7411] [ERROR] Exception in worker process
> Traceback (most recent call last):
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
>     worker.init_process()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/workers/base.py", line 119, in init_process
>     self.load_wsgi()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
>     self.wsgi = self.app.wsgi()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
>     self.callable = self.load()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
>     return self.load_wsgiapp()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
>     return util.import_app(self.app_uri)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/util.py", line 358, in import_app
>     mod = importlib.import_module(module)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
>   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
>   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
>   File "/home/dsmr/dsmr-reader/dsmrreader/wsgi.py", line 12, in <module>
>     from django.core.wsgi import get_wsgi_application
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/wsgi.py", line 2, in <module>
>     from django.core.handlers.wsgi import WSGIHandler
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/wsgi.py", line 6, in <module>
>     from django.core.handlers import base
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 7, in <module>
>     from django.db import connections, transaction
> ValueError: source code string cannot contain null bytes
> [2020-03-14 23:12:17 +0100] [7411] [INFO] Worker exiting (pid: 7411)
> [2020-03-14 23:12:17 +0100] [7408] [INFO] Shutting down: Master
> [2020-03-14 23:12:17 +0100] [7408] [INFO] Reason: Worker failed to boot.

I tried a few things found online (sed -i 's/\x0//g' , rt, etc...) but no solution yet.

dennissiemensma, 2020-03-15:
I see. If Postgresql refuses to start, you'll need to to fix that **first**. DSMR-reader and Gunicorn/Django do not effect whether the database starts or not. You will need to be able to connect to the database:

```
sudo su - postgres
psql dsmrreader
```

You might want to check the USB storage for any filesystem errors. Having it mounted doesn't guarantee that it's healthy. Also have a look in `/var/log/syslog` for any errors.

- Do make sure the permissions/owner for the USB-storage are set to `postgres`. Although I would expect specific permission errors in the Postgresql log if these do not suffice.

----

If you still cannot find the cause, try unmounting the USB-storage, change the data directory to the SD-card (for test) and create a new cluster (for test):

- `sudo supervisorctl stop all`
- https://dsmr-reader.readthedocs.io/en/v3/faq/database.html
- `pg_createcluster 9.5 main --start` (or whatever version number you're running)

If this works (Postgresql doesn't exit), then the issue is probably with the USB-storage. You could even try an empty database for test:

- Create a new database:
```
sudo sudo -u postgres createuser -DSR dsmrreader
sudo sudo -u postgres createdb -O dsmrreader dsmrreader
sudo sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"

sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py migrate
sudo supervisorctl start all
```

TheMClaeys, 2020-03-15:
I checked the commands above.
```
sudo su - postgres
psql dsmreader
```
These work.
I can use \l to show the databases:

> dsmrreader-# \l
>                                    List of databases
>     Name    |   Owner    | Encoding |   Collate   |    Ctype    |   Access privileges
> ------------+------------+----------+-------------+-------------+-----------------------
>  dsmrreader | dsmrreader | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 |
>  postgres   | postgres   | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 |
>  template0  | postgres   | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 | =c/postgres          +
>             |            |          |             |             | postgres=CTc/postgres
>  template1  | postgres   | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 | =c/postgres          +
>             |            |          |             |             | postgres=CTc/postgres
> (4 rows)
> 

No issue there apparently.

The syslog show only the following errors:

> Mar 15 00:08:10 DSMR supervisord[502]: 2020-03-15 00:08:10,020 INFO spawned: 'dsmr_backend' with pid 8178
> Mar 15 00:08:10 DSMR supervisord[502]: 2020-03-15 00:08:10,564 INFO exited: dsmr_backend (exit status 1; not expected)
> Mar 15 00:08:14 DSMR supervisord[502]: 2020-03-15 00:08:14,577 INFO spawned: 'dsmr_datalogger' with pid 8179
> Mar 15 00:08:15 DSMR supervisord[502]: 2020-03-15 00:08:15,107 INFO exited: dsmr_datalogger (exit status 1; not expected)
> Mar 15 00:09:50 DSMR supervisord[502]: 2020-03-15 00:09:50,797 INFO spawned: 'dsmr_backend' with pid 8182
> Mar 15 00:09:51 DSMR supervisord[502]: 2020-03-15 00:09:51,294 INFO exited: dsmr_backend (exit status 1; not expected)
> Mar 15 00:09:51 DSMR kernel: [41144.768022] rpi_firmware_get_throttled: 1 callbacks suppressed

File system check of USB I'll try later today, I need to unmount it for that and apparently the target is busy, even after stopping the Postgresql service.

dennissiemensma, 2020-03-15:
I'm a bit confused. So the database server is running but it's reporting as status that it has been exited?

Does this work? 
```
psql -h localhost -U dsmrreader
```
When asked fora password, it should be `dsmrreader`, unless you've changed it manually.

If you can connect, does this work? (no error)

```
sudo su - dsmr
./manage.py migrate
```

TheMClaeys, 2020-03-15:
Apparently it refuses connections:

> psql: could not connect to server: Verbinding is geweigerd
>         Is the server running on host "localhost" (::1) and accepting
>         TCP/IP connections on port 5432?
> could not connect to server: Verbinding is geweigerd
>         Is the server running on host "localhost" (127.0.0.1) and accepting
>         TCP/IP connections on port 5432?
> 

Or do I have to do that under a different user like dsmr? (sorry, know my way a bit in Linux but far from advanced).

dennissiemensma, 2020-03-15:
Does the other command work?

```
sudo su - dsmr
./manage.py migrate
```

TheMClaeys, 2020-03-15:
No, I get an error. Looks like the same error I described above somewhere.

> 
> (dsmrreader) dsmr@DSMR:~/dsmr-reader $ ./manage.py migrate
> Traceback (most recent call last):
>   File "./manage.py", line 8, in <module>
>     from django.core.management import execute_from_command_line
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 14, in <module>
>     from django.core.management.base import (
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 11, in <module>
>     from django.core import checks
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/checks/__init__.py", line 9, in <module>
>     import django.core.checks.database  # NOQA isort:skip
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/checks/database.py", line 1, in <module>
>     from django.db import connections
> ValueError: source code string cannot contain null bytes

dennissiemensma, 2020-03-15:
You should try reinstalling all the requirements to see if that fixes the issue:

```
sudo supervisorctl stop all

sudo su - dsmr 
rm -rf ~/.virtualenvs/dsmrreader/
virtualenv /home/dsmr/.virtualenvs/dsmrreader --no-site-packages --python python3
/home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/postgresql.txt
```

Then try again:
```
logout
sudo su - dsmr
./manage.py migrate
```

TheMClaeys, 2020-03-15:
Ok, that did seem to change some things:

> (dsmrreader) dsmr@DSMR:~/dsmr-reader $ ./manage.py migrate
> Traceback (most recent call last):
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
>     self.connect()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
>     return func(*args, **kwargs)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 197, in connect
>     self.connection = self.get_new_connection(conn_params)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
>     return func(*args, **kwargs)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 185, in get_new_connection
>     connection = Database.connect(**conn_params)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/psycopg2/__init__.py", line 126, in connect
>     conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
> psycopg2.OperationalError: could not connect to server: Connection refused
>         Is the server running on host "localhost" (::1) and accepting
>         TCP/IP connections on port 5432?
> could not connect to server: Connection refused
>         Is the server running on host "localhost" (127.0.0.1) and accepting
>         TCP/IP connections on port 5432?
> 
> 
> The above exception was the direct cause of the following exception:
> 
> Traceback (most recent call last):
>   File "./manage.py", line 10, in <module>
>     execute_from_command_line(sys.argv)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
>     utility.execute()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
>     self.fetch_command(subcommand).run_from_argv(self.argv)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 328, in run_from_argv
>     self.execute(*args, **cmd_options)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 369, in execute
>     output = self.handle(*args, **options)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 83, in wrapped
>     res = handle_func(*args, **kwargs)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/commands/migrate.py", line 86, in handle
>     executor = MigrationExecutor(connection, self.migration_progress_callback)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 18, in __init__
>     self.loader = MigrationLoader(self.connection)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/loader.py", line 49, in __init__
>     self.build_graph()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/loader.py", line 212, in build_graph
>     self.applied_migrations = recorder.applied_migrations()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/recorder.py", line 76, in applied_migrations
>     if self.has_table():
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/recorder.py", line 56, in has_table
>     return self.Migration._meta.db_table in self.connection.introspection.table_names(self.connection.cursor())
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
>     return func(*args, **kwargs)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 260, in cursor
>     return self._cursor()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 236, in _cursor
>     self.ensure_connection()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
>     return func(*args, **kwargs)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
>     self.connect()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
>     raise dj_exc_value.with_traceback(traceback) from exc_value
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
>     self.connect()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
>     return func(*args, **kwargs)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 197, in connect
>     self.connection = self.get_new_connection(conn_params)
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
>     return func(*args, **kwargs)
[... cut ...]
>         Is the server running on host "localhost" (127.0.0.1) and accepting
>         TCP/IP connections on port 5432?
> 

So it looks like a firewall rule maybe still blocking something (I never changed any rule, so that's strange though). The iptables are empty though.

> pi@DSMR:~ $ sudo iptables -L
> Chain INPUT (policy ACCEPT)
> target     prot opt source               destination
> 
> Chain FORWARD (policy ACCEPT)
> target     prot opt source               destination
> 
> Chain OUTPUT (policy ACCEPT)
> target     prot opt source               destination
>

TheMClaeys, 2020-03-15:
Update: nevermind, I quickly also checked if postgresql was running, it was inactive. So I started it again (`sudo systemctl status postgresql`)

The output is now the following:

> Operations to perform:
>   Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
> Running migrations:
>   No migrations to apply.
>

closed 2020-03-16 by TheMClaeys

dennissiemensma, 2020-03-16:
Good to hear. Maybe the starting order of DSMR-reader and the database on boot mismatch, causing the application to crash. Not sure though.

TheMClaeys, 2020-03-16:
Would there be a way to influence the boor order to have it always the correct way?
I don't recall it having a power failure, but it's always possible. I did disconnect it from the network a around that day to change the cable to a cable of better lenght, possibly I touched the power connector then.

dennissiemensma, 2020-03-17:
I'm not sure if you can change the booting order, but it's something to keep in mind if this (randomly) happens again.