dsmrreader/dsmr-reader #926: Suggestie: Live grafieken standaard laten uitzoomen naar laatste uur laten zien in plaats van de afgelopen minuut.

iMartien, 2020-03-16:
Het is denk ik overzichtelijker om de live grafieken standaard op het afgelopen uur in te stellen in plaats van de afgelopen minuut / laatste paar seconden.

Of misschien de mogelijkheid om de standaard zoom in te stellen?

dennissiemensma, 2020-03-16:
Je kunt dit als het goed is wijzigen via `Interfaceconfiguratie` > `Dashboard grafiek breedte`. 

Daar kun je geen tijdspanne invoeren, maar wel het aantal punten dat zichtbaar is. Houd er rekening mee dat de performance in je browser wat minder kan zijn bij een hoog aantal datapunten.

basroovers, 2020-03-31:
Ik zou het zelf fijn vinden om in de live grafiek de mogelijkheid te hebben om met een dropdown de time range in te kunnen stellen. Zie hier een voorbeeld uit Grafana.

[image]

dennissiemensma, 2020-03-31:
Bedankt voor je toevoeging. Toevallig is hier net #947 voor aangemaakt en had ik niet gezien dat het huidige ticket er (nog) was. Voor nu sluit ik deze en gaat het via #947.

closed 2020-03-31 by dennissiemensma

dennissiemensma, 2020-05-03:
In de volgende release kan je zelf de range instellen in uren, van 1 tot 168 (week).