dsmrreader/dsmr-reader #927: V3.7.0 upgrade issue mbt tot backups?

jorkzijlstra, 2020-03-20:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.7.0
* Hardware *(RaspberryPi 3, 4 of iets anders)*: Rock PI 4B
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Docker
* DSMR-protocol slimme meter *(v4 of v5)*: v5

Ik ben net naar v3.7.0 gegaan en had wat issues met het opstarten. De `Generate consumption data` begon achter te lopen op de status pagina. De telegrammen werden wel netjes geregistreerd.

Ik heb DEBUG loglevel even aangezet en zag alleen deze voorbij komen. Het leek erop dat zodra `could not find a "pg_dump" to execute` langs kwam de data achter begon te lopen.

```
[2020-03-20 12:44:23,546] DEBUG    Compact: Processed reading: 18605507 @ 2020-03-20 12:43:58+01:00
[2020-03-20 12:44:23,556] DEBUG    SP: Rescheduled "Generate consumption data" to 2020-03-20 12:44:24.548590+01:00 (ETA 0:00:00.991772)
[2020-03-20 12:44:23,587] INFO      - Creating new partial backup: /dsmr/backups/archive/2020/03/dsmrreader-postgresql-partial-backup-2020-03-20.sql
[2020-03-20 12:44:23,961] DEBUG     - Backup exit code: 0
[2020-03-20 12:44:24,043] INFO      - Created and compressed statistics backup: /dsmr/backups/archive/2020/03/dsmrreader-postgresql-partial-backup-2020-03-20.sql.gz
[2020-03-20 12:44:24,059] INFO      - Creating new full backup: /dsmr/backups/dsmrreader-postgresql-backup-Friday.sql
could not find a "pg_dump" to execute
```

Bij het opstarten van V3.7.0 leek er dus vanalles gedaan te worden met de backups / inclusief uploaden. Na een aantal herstarts lijkt alles nu wel weer stabiel. 
Het commando `pg_dump` staat wel op het path, ik kan het zelf namelijk gewoon uitvoeren.

Ik zal morgen even kijken of de backup gelukt is

dennissiemensma, 2020-03-21:
Bedankt voor je melding. Is de backup inmiddels gelukt?

Er is inhoudelijk niets veranderd aan de backups. Het compressielevel is nu instelbaar, maar die stond eerder op 1 (snelste) en nu nog steeds als default.

Verder zijn de backups een volgend punt wat ik asynchroon kan maken aan het backend-proces, zodat die niet elke run checkt of er een backup gedaan moet worden.

jorkzijlstra, 2020-03-21:
Net gechecked en ja is gelukt. Geen idee wat er dus aan de hand was

closed 2020-03-21 by jorkzijlstra

dennissiemensma, 2020-03-21:
Toch bedankt voor je oplettenheid hoor. Het kan altijd zijn dat er iets speelt waar ik niet vanaf weet.