dsmrreader/dsmr-reader #928: DSMR_datalogger restarts every 5-7 seconds.

Josechap, 2020-03-21:
DSMR_datalogger keeps restarting even 5-7 seconds. Datalogger is not receiving any telegrams
My current setup uses a decryptor for the luxembourg smarty meter and then publishes it into a com-port using socat.

To debug I've deactivated the socat feed and the smarty meter decryptor and even then, the datalogger keeps restarting every couple of seconds.

**Wat gebruik je?**
* DSMR-reader version: 3.7
* Hardware: Raspberry pi 3B+
* Native
* DSMR-protocol slimme meter: Luxembourg Smarty Meter

dennissiemensma, 2020-03-22:
Thanks for reporting this issue. Could you have a look in de log files to see what the exact error is?

https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#logging

Josechap, 2020-03-22:
Thanks for the followup. This is what i'm finding in dsmr_datalogger-stderr---supervisor-NIJ4MT.log:

> ERROR:commands:dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/serial/serialposix.py", line 265, in open
>     self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
> FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyusb0'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once
>     self.run(data=self.data, **options)
>   File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 21, in run
>     dsmr_datalogger.services.datalogger.read_and_process_telegram()
>   File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 69, in read_and_process_telegram
>     read_serial_port(**connection_parameters)
>   File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 51, in read_serial_port
>     timeout=20,  # Max time to wait for data.
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/serial/serialutil.py", line 240, in __init__
>     self.open()
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/serial/serialposix.py", line 268, in open
>     raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
> serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyusb0: [Errno 2] No such file or directory: '/dev/ttyusb0'
> 
> [2020-03-22 17:04:17,135] INFO     dsmr_datalogger.management.commands.dsmr_datalogger: Exiting on next run...
> INFO:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Exiting on next run...
> [2020-03-22 17:04:17,136] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 0.5s
> DEBUG:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 0.5s
> [2020-03-22 17:04:17,136] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Exited
> DEBUG:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Exited

dennissiemensma, 2020-03-22:
It seems that the application cannot find (or access) the port specified: 

```
[Errno 2] No such file or directory: '/dev/ttyusb0'
```

When this happens, you can check for yourself in the OS whether the serial port is available:
```
ls -l /dev/ttyusb0
```

Then test it manually with `cu`: https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html#your-first-reading-optional

When `cu` cannot access it, DSMR-reader cannot either for sure.

If you can't find `/dev/ttyusb0`, check whether it's available as another name:
```
ls -l /dev/tty*
```

If it did switch names, try assigning it by ID in DSMR-reader admin (serial port path). You can find the ID here (it's just an alias/symlink):
```
ls -l /dev/serial/by-id/
```

For example:
```
ls -l /dev/serial/by-id/
total 0
lrwxrwxrwx 1 root root 13 Mar  9 20:43 usb-FTDI_FT232R_USB_UART_A5Z5VGV5-if00-port0 -> ../../ttyUSB0
```

Then `/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A5Z5VGV5-if00-port0` would by my ID to use.

Josechap, 2020-03-22:
Thanks. I've done quite a bit of progress. This is where I'm at. I fixed the error with ttyusb0 (should've been ttyUSB0... case sensitive :)). I had to change the com-port I'll use because I have a luxembourgish smart meter and therefore I need to decrypt the telegrams. I am publishing the decrypted telegram feed to /dev/pts/1, and now the logs have the following error:

PermissionError: [Errno 13] Permission denied: '/dev/pts/1'

I think i just need to provide access for dsmr reader to read /dev/pts/1?

closed 2020-03-22 by Josechap

dennissiemensma, 2020-03-22:
Thanks for sharing the outcome. By default the application uses the `dsmr` user and during installation i'll be added to the `dialout` group, which should be sufficient. 

But in your case, you might need some custom permissions. Be sure to try a reboot too, as sometimes permissions by `socat`/sockets/ports are reset after the process ends.