dsmrreader/dsmr-reader #929: Telegrams stoppen na 1 tot 3 dagen

jghaanstra, 2020-03-22:
Ik ben recent gemigreerd van het uitlezen van de P1 poort via Home Assistant door DSMR Reader. Ik ben zeer tevreden over de mogelijkheden en de layout van DSMR Reader, super gedaan. Ik loop echter tegen het probleem aan dat na 1 tot 3 dagen er geen telegram berichten meer binnenkomen. Dit is op te lossen door de FTDI USB er even uit te trekken en weer in te doen of my Synology volledig te herstarten waarna het verwerken van de telegrams weer wordt hervat. Uiteraard zijn dit geen werkbare workarounds. Dergelijke problemen heb ik nooit gehad bij het uitlezen van de P1 poort via Home Assistant.

Ik gebruik zowel DSMR Reader als Home Assistant in een docker container (van xirixiz) waarbij /dev/ttyUSB0 als device is geregistreerd. Ik zie in de log van de DSMR Reader Docker container niks bijzonders op het moment dat het binnenkomen van de berichten stopt. Ik heb de FTDI USB al vervangen maar daarmee is het probleem niet verholpen. Het aanzetten van debug logging lukt niet omdat `./post-deploy.sh` niet werkt in de Docker container. Heeft er iemand een idee waar dit aan kan liggen en wat ik er aan kan doen om dit op te lossen?

**Wat gebruik je?**
* DSMR-reader versie: v3.7
* Hardware: Synology DS216+II
* Native of Docker : Docker van xirixiz
* DSMR-protocol slimme meter: v4.2

dennissiemensma, 2020-03-22:
Bedankt voor je melding. Het is wat lastig voor je omdat ik zelf nog te weinig met Docker doe en je daarmee niet kan helpen. Wellicht kun je het debug inschakelen en uitlezen van de logs bij xirixiz zelf vragen (misschien staat het ook in de docs of gesloten issues daar).

Wat betreft het vastlopen. Helpt het herstarten van de docker-container op moment van vastlopen ook niet? Dit om te kijken welke schakels nog meer ervoor zorgen bij een herstart dat het weer door loopt.

jghaanstra, 2020-03-22:
Herstarten van de Docker container lost het probleem niet op. Het lijkt daarmee een probleem dat ook hardware of in ieder geval core OS (Linux van Synology) gerelateerd is. Ik begrijp echter niet waarom dit probleem zich met een Home Assistant container niet voordoet. Ergens in de DSMR Reader of de DSMR Reader Docker container gaat er iets niet goed.

Maar als er niet zo een oplossing voor handen is dan sluit ik deze ticket. Ik ben bang dat er niks anders op zit dan terug te gaan naar Home Assistant.

closed 2020-03-22 by jghaanstra

dennissiemensma, 2020-03-22:
Bedankt voor het proberen. Je bent niet de enige met dit probleem bij virtualisatie (zoals bij #848), maar er is tot nu toe geen eenduidige oorzaak gevonden.
En wellicht was ik niet duidelijk genoeg. Ik kan je wel helpen met debuggen, maar niet met docker-specifieke issues. Mocht het je lukken om de logging binnen de container in te zien, deel gerust eventuele foutmeldingen die je ziet. Het kan zomaar iets zijn zoals in #928, al vermoed ik hier iets anders.

jghaanstra, 2020-03-29:
Ik voeg wat extra informatie toe voor mensen die hetzelfde probleem ervaren. Wellicht helpt het om in de toekomst dit op te lossen.

Ik heb een P1 splitter gebruikt en heb beide FTDI kabels afgesloten op mijn Synology NAS. De ene lees ik uit in een DSMR Reader docker container, de andere in een Home Assistant docker container. Na 1 tot 3 dagen stopt het binnenkomen van de telegrams in de DSMR Reader container terwijl in Home Assistent de verwerking gewoon doorgaat. Daarmee is het in ieder geval bevestigd dat het geen hardware probleem is en dat het wordt veroorzaakt door DSMR Reader of de gebruikte docker container (of een combinatie daarvan).

Ik blijft proberen de oorzaak te achterhalen. Mocht ik meer te weten komen zal ik dat hier toevoegen.

dennissiemensma, 2020-03-29:
Bedankt voor de update. Normaal zou ik zeggen probeer ook eens alleen de datalogger op de host te installeren, maar ik weet niet of dat makkelijk kan met je OS.

Is het nog gelukt om de foutlogs binnen de DSMR-container uit te lezen? Mogelijk staat daar ook nog meer informatie in.

jghaanstra, 2020-03-30:
Het is niet eenvoudig om op de OS van Synology de datalogger te installeren aangezien er geen package manager beschikbaar is. De logs binnen de DSMR container die beschikbaar zijn geven geen extra informatie.

Ik heb het opgegeven en ben nu aan het kijken of ik via Home Assistant de API van DSMR Reader kan aanroepen. Daarmee gaat het uitlezen van Home Assistant maar kan ik toch gebruik maken van DSMR Reader.