dsmrreader/dsmr-reader #930: Afrondingsfout bij teruggeleverde elektriciteit

KobeBergmans, 2020-03-22:
Bij het bekijken van de live grafiek zijn er soms waarden van de teruggeleverde stroom niet goed afgerond. Zie afbeelding
[image]

**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.7
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 3 Datalogger en Ubuntu backend en webserver.
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: vDSMR 5 (Belgie)

dennissiemensma, 2020-03-22:
Bedankt voor je melding. Het was me zelf nog niet opgevallen, maar ik zie het nu bij mijn eigen grafiek ook. Ik neem het mee voor de volgende release.

dennissiemensma, 2020-04-01:
Het lijkt heel specifiek op te treden bij vaste getallen, zoals `1023`. Vermoedelijk doordat het een float is gemaakt en dat blijven altijd onbetrouwbare dingen. En bij nader inzien hoeven het ook geen floats te zijn, maar gewoon integers, gezien het over eenheden gaat zonder kommagetal.

dennissiemensma, 2020-04-01:
Aangepast voor volgende release.

closed 2020-04-01 by dennissiemensma