dsmrreader/dsmr-reader #931: In live graphs, make it configurable to show energy returned or delivered in datazoom miniature

JeanMiK, 2020-03-22:
The daily graphs use echarts javascript components to present the data of the current day.

Each graph has an echart datazoom component to allow zooming in on a particular section of the day. This datazoom component has a miniature graph of the whole day as an orientation, but the echarts documentation does not indicate where the data for this curve is taken from. It **looks** like it is simply taking (for an x-axis datazoom) the data of the **first** data series.

In the current dsmr code, this is always the _energy delivered_. It would be great to make this configureable, so that depending on a setting in the front-end configuration, either the energy delivered or the energy returned would appear in the miniature graph.

Rationale: in some cases, a dsmr smart-meter may be used exclusively for a PV installation, so that the energy delivered represents only the uninteresting, almost constant energy consumption of the inverter, and the only relevant data is the "energy returned", i.e. the energy produced by the PV installation. In such a case, it would be better to have this latter data displayed in the datazoom component.

In Luxembourg, this is the case for *every* PV installation, as here the legislative environment / the energy provider rules impose dedicated smart meters for any PV installation.

From what I can see, it would be possible to achieve this by inverting in dsmr_frontend/static/dsmr_frontend/js/dsmr-reader/live/electricity.js the usage of series[0] and series[1].

dennissiemensma, 2020-03-22:
Thank you for the suggestion. Would you mind adding some sort of screenshot to indicate the problem you're trying to solve here? I might have an idea what you're talking about, but I'm not 100% sure.

JeanMiK, 2020-03-22:
Sure.

 

In the screenshot below, you see the live graph for my PV smart-meter. The miniature graph in the datazoom element below the ‘recent electricity consumption’ graph shows an almost flat line with just two spikes around the moments where the inverters kick in or stop (see the second picture for a zoom around that moment).

 

This is to say, the miniature graph in that datazoom field is a representation of the red energy consumption curve.

 

For a PV meter, it would be more relevant, to have a miniature of the green ‘energy delivered’ curve (i.e.something similar to the miniature curve below the “recent phase currents* graph), or a miniature of both, or maybe a miniature of the absolute value of the net energy consumption / production.

 

 

 

This is zooming in on the start of the energy production. The red line is the energy consumption of the inverter, and it is the only thing visible in the miniature:

 

 

 

For a regular house consumption smart-meter like my other smarty, showing the consumption (as done today) is best:

 

 

(this other smart meter does for the moment not return any phase current information, so there is only one graph)

 

 

I understand that the echart javascript package is an external resource that cannot be changed easily, so the simplest would be to just foresee the possibility to invert the order of the two series.

 

Alternatively, it might also be possible to have a single configuration with *three* series, where the first one would be specified with 0 for the parameter “series-line.areaStyle.opacity” in order to have it invisible, and to fill this invisible series with whatever we want to display in the miniature.

 

Regards,

 

Jean-Marie

dennissiemensma, 2020-03-22:
Thanks! But I'm not seeing any images here. Github might strip them from emails. You could try it again by editing your post and upload the images using Guthub in the browser.

JeanMiK, 2020-03-23:
Indeed the mail seems to have cut the pictures, I put them back manually.

dennissiemensma, 2020-04-01:
I've looked into this earlier this week but I was unable to figure a decent solution. Another thing I tried was the latest update for eCharts, but it didn't improve this. For now I'll leave it as is and I'll try again in a while.

dennissiemensma, 2020-06-24:
For now I'll leave this as is. There could be a workaround made in DSMR-reader, but I don't think it's worth the complexity, to be honest.

closed 2020-06-24 by dennissiemensma