dsmrreader/dsmr-reader #932: DSMR Reader start niet meer op

Weppel, 2020-03-23:
Enige tijd na de update naar v3.7 is DSMR Reader vastgelopen en start niet meer op. Ik zie de volgende error in de logfiles /var/log/supervisor/dsmr_*.log:

`Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 328, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 366, in execute
    self.check()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 395, in check
    include_deployment_checks=include_deployment_checks,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 382, in _run_checks
    return checks.run_checks(**kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/checks/registry.py", line 72, in run_checks
    new_errors = check(app_configs=app_configs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/checks/urls.py", line 13, in check_url_config
    return check_resolver(resolver)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/checks/urls.py", line 23, in check_resolver
    return check_method()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/urls/resolvers.py", line 407, in check
    for pattern in self.url_patterns:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/functional.py", line 48, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/urls/resolvers.py", line 588, in url_patterns
    patterns = getattr(self.urlconf_module, "urlpatterns", self.urlconf_module)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/functional.py", line 48, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/urls/resolvers.py", line 581, in urlconf_module
    return import_module(self.urlconf_name)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/dsmr/dsmr-reader/dsmrreader/urls.py", line 8, in <module>
    path('api/v1/', include('dsmr_api.urls.v1')),
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/urls/conf.py", line 34, in include
    urlconf_module = import_module(urlconf_module)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/dsmr/dsmr-reader/dsmr_api/urls/v1.py", line 4, in <module>
    from dsmr_api.views.v1 import DataloggerDsmrReading
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 857, in get_code
  File "<frozen importlib._bootstrap_external>", line 525, in _compile_bytecode
ValueError: bad marshal data (unknown type code)`

dennissiemensma, 2020-03-23:
Bedankt voor je melding. Je kunt dit proberen:

```
sudo su - dsmr
find ~ -type f -iname '*.pyc' -delete

logout
sudo supervisorctl restart all
```

Weppel, 2020-03-24:
Dat heeft het inderdaad opgelost, bedankt! Wel zie ik nu een andere (wellicht niet gerelateerde) error wanneer ik naar /admin/login/?next=/notifications ga:

`Server Error
Sorry, something unexpected happened.
Exception:
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x98 in position 2: invalid start byte
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 143, in _get_response
    response = response.render()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/template/response.py", line 105, in render
    self.content = self.rendered_content

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/template/response.py", line 81, in rendered_content
    template = self.resolve_template(self.template_name)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/template/response.py", line 63, in resolve_template
    return select_template(template, using=self.using)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/template/loader.py", line 42, in select_template
    return engine.get_template(template_name)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/template/backends/django.py", line 34, in get_template
    return Template(self.engine.get_template(template_name), self)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/template/engine.py", line 143, in get_template
    template, origin = self.find_template(template_name)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/template/engine.py", line 125, in find_template
    template = loader.get_template(name, skip=skip)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/template/loaders/cached.py", line 53, in get_template
    template = super().get_template(template_name, skip)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/template/loaders/base.py", line 24, in get_template
    contents = self.get_contents(origin)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/template/loaders/cached.py", line 22, in get_contents
    return origin.loader.get_contents(origin)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/template/loaders/filesystem.py", line 24, in get_contents
    return fp.read()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
`

dennissiemensma, 2020-03-24:
Het lijkt er op dat er iets is het het bestandssysteem. Je loopt nu tegen meerdere issues aan die ik normaal nooit zie.

Wil je eens kijken of er wijzigingen zijn?
```
sudo su - dsmr
git status
```

Wat is de output van `git status`?

Weppel, 2020-03-24:
Output van `git status`: 

```
(dsmrreader) dsmr@dsmrreader:~/dsmr-reader $ git status
On branch v3
Your branch is up to date with 'origin/v3'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        modified:   dsmr_frontend/static/dsmr_frontend/js/dsmr-reader/live/gas.js
        modified:   dsmr_frontend/static/dsmr_frontend/js/dsmr-reader/live/temperature.js

no changes added to commit (use "git add" and/or "git commit -a")
```

Ook na het uitvoeren van de volgende commando's blijft bovenstaande error aanwezig: 
```
(dsmrreader) dsmr@dsmrreader:~/dsmr-reader $ git checkout -- dsmr_frontend/static/dsmr_frontend/js/dsmr-reader/live/gas.js
(dsmrreader) dsmr@dsmrreader:~/dsmr-reader $ git checkout -- dsmr_frontend/static/dsmr_frontend/js/dsmr-reader/live/temperature.js
(dsmrreader) dsmr@dsmrreader:~/dsmr-reader $ git status
On branch v3
Your branch is up to date with 'origin/v3'.

nothing to commit, working tree clean
```

dennissiemensma, 2020-03-24:
Je kunt nog dit proberen:

```
sudo su - dsmr
git reset HEAD --hard

logout
sudo supervisorctl restart all
```

Maar dat doet waarschijnlijk hetzelfde als die checkouts

Weppel, 2020-03-24:
Geeft inderdaad hetzelfde resultaat, nog steeds bovenstaande foutmelding. Nog andere suggesties, of kan ik beter gewoon een verse installatie uitvoeren?

dennissiemensma, 2020-03-24:
Een versie installatie is denk ik het handigste. Je kunt daarbij de database overigens gewoon laten staan.

dennissiemensma, 2020-03-27:
Lost de herinstallatie het op of krijg je dezelfde foutmelding(en)?

Weppel, 2020-03-28:
Sorry voor de late reactie, het is druk geweest. Een herinstallatie lost dit probleem op, bedankt voor de pointers!

closed 2020-03-28 by Weppel