dsmrreader/dsmr-reader #933: DSMR - MQTT mosquitto:  Socket error on client <unknown>, disconnecting

HyperDevil, 2020-03-24:
**Wat gebruik je?**
DSMR en MQTT mosquitto (valid SSL certificate)
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.7
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 3
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: v5

Ik gebruik mosquitto MQTT om de berichten te ontvangen, dit werkt goed op mijn testbench met een volledig open MQTT server (anonymous poort 1883).
Zodra ik de configuratie overzet op een beveiligde MQTT server poort 8883 (met een geldig SSL certificaat) weigert dsmr aan te koppelen, zowel SECURE_RREUIRED als CERT_NONE.
De error op de server is: Socket error on client <unknown>, disconnecting

Ik heb dit eerder gezien ergens anders, wat blijkt is dat de client (in dit geval DSMR) niet het goede certificaat lokaal kan vinden en dus een chain niet kan valideren.

Ik gebruik zelf paho mqtt in mijn eigen python scripts in hierin zet ik altijd:

> tls_certs_path = '/etc/ssl/certs/GlobalSign_Root_CA.pem'

Hoe kan ik dit oplossen in DSMR?

HyperDevil, 2020-03-24:
Vergeten mosquitto reloaden.. 🤦‍♂

closed 2020-03-24 by HyperDevil

dennissiemensma, 2020-03-24:
Begrijp ik het goed dat je dit hebt kunnen oplossen zonder handmatige wijzigingen in de code? Ik heb zelf namelijk geen ervaringen met SSL/TLS en MQTT.

HyperDevil, 2020-03-24:
Ja klopt, het lag aan de MQTT server kant en niet aan DSMR. :)

dennissiemensma, 2020-03-24:
Aah oke, ik was meer benieuwd of je nog in de client wijzigingen moest doen om die root-chain werkend te krijgen in DSMR-reader, maar fijn om te horen dat het uiteindelijk in de MQTT server opgelost kan worden.

resimons, 2025-06-14:
> Ja klopt, het lag aan de MQTT server kant en niet aan DSMR. :)

Ha HyperDevil

> Aah oke, ik was meer benieuwd of je nog in de client wijzigingen moest doen om die root-chain werkend te krijgen in DSMR-reader, maar fijn om te horen dat het uiteindelijk in de MQTT server opgelost kan worden.

Ha Dennis,

toch is dit iets dat in de client opgelost moet worden. Broker.py moet er als volgt uitzien:

        logger.debug("MQTT: Using secure connection (ssl.CERT_REQUIRED)")
        ca_certificate_file = "/home/dsmr/dsmr-reader/certs/ca-crt.pem"
        client_certificate_file = "/home/dsmr/dsmr-reader/certs/client-crt.pem"
        client_key_file = "/home/dsmr/dsmr-reader/certs/client-key.pem"

        mqtt_client.tls_set(ca_certs = ca_certificate_file, certfile = client_certificate_file, keyfile = client_key_file, tls_version=ssl.PROTOCOL_TLSv1_2)
        mqtt_client.tls_insecure_set(False)

Als gebruiker moet je de certificaten op kunnen geven. Die heeft de client nodig om verbinding te kunnen maken met de server.
Wil je de GUI zodanig aanpassen dat als de tls-optie secure wordt geselecteerd dat er drie bestanden kunnen worden geüploaded?

Groeten,

Roel Simons

dennissiemensma, 2025-06-17:
Hey Roel, DSMR-reader gaat uiteindelijk naar container-hosting, dus wellicht is het dan handiger om zoiets via envvars te doen. Dat is vaak ook relatief weinig werk.

Je kunt het beste een nieuw issue hiervoor aanmaken. Al zal dit zeker niet op de korte of middellange termijn op de planning staan. Eerst moet DSMR-reader over naar containers (en daarmee nieuwere python en alles) voordat ik uberhaupt iets kan doen.