dsmrreader/dsmr-reader #936: Home Assistant Add-on DSMR Datalogger created

sanderdw, 2020-03-25:
Hi Dennis,

I created an Home Assistant add-on for the DSMR datalogger. Do you mind if i use the logo's etc?

Currently it looks like this in Home Assistant:
[image]

[image]

[image]

Any remarks or things i need to change, please let me know?

Love this project btw!

dennissiemensma, 2020-03-25:
It looks pretty neat. You can use the logo, as it's not mine anyway (https://www.flaticon.com/free-icon/eco-energy_25013) and it's credited in the DSMR-reader docs so you should be fine.

The only remark I have is that HTTPS is not required or default for the `DSMR_API_URL`, so you might take that into account.

dennissiemensma, 2020-03-27:
Let me know if you need anything else. For now I'll close this.

closed 2020-03-27 by dennissiemensma