dsmrreader/dsmr-reader #937: Dashboard €/uur berekening is minimaal 0

Phyxion, 2020-03-26:
**Wat gebruik je?**
* DSMR-reader versie: 3.7
* Hardware: RaspberryPi 3b
* Native of Docker: Docker
* DSMR-protocol slimme meter: v5

Screenshot ter verduidelijking:

[image]

Waar het hoog- en laagtarief wel negatieve waardes weergeven bij de kosten (er wordt immers stroom opgeleverd), staat de €/uur berekening vast op 0,00.

dennissiemensma, 2020-03-27:
Bedankt voor je melding. Ik ben wat terughoudend met het tonen en meenemen van opbrengsten van teruglevering, omdat die regels vrij complex zijn en ik dat een grijs gebied vind binnen dit project.

Echter, gezien het hier puur om een indicatie gaat op het dashboard, vind ik het (**voor nu**) prima om het ook daarin op te nemen.

dennissiemensma, 2020-03-27:
Aangepast voor de volgende release.

closed 2020-03-27 by dennissiemensma

Phyxion, 2020-03-27:
Fijn, bedankt. Qua regels is het niet veel anders dan nu voor alleen de kosten toch? Het lijkt mij dat je gewoon dat wat je al hebt en dan - watt * terugleveringtarief kan doen? Dat het niet helemaal 100% klopt maakt ook niet heel veel uit, een indicatie is wel leuk in ieder geval. De energiecontracten / kosten zijn juist zo leuk om te bekijken ☺