dsmrreader/dsmr-reader #938: Verschillende kleuren op dashboard voor negatieve kosten / opbrengst

Phyxion, 2020-03-26:
Op het dashboard zijn zowel kosten als opbrengsten zwart. Door verschillende kleuren te gebruiken is het makkelijker om onderscheid te maken tussen de daadwerkelijke kosten en opbrengsten.

Ik heb hieronder bg-olive gebruikt als voorbeeld.

[image]

dennissiemensma, 2020-03-27:
Bedankt voor je idee. Ik begrijp je punt en kan me persoonlijk ook wel vinden in je suggestie. 

Echter wil ik voorkomen dat de interface een kleurplaat/regenboog wordt. Het palet en de variatie die nu in de meest recente versie zit, vind ik ook al op het randje. Ik denk eerder dat ik het op termijn zelfs nog wat rustiger maak, door bijvoorbeeld alleen de namen een kleur te geven. Maar dat laat ik voor nu nog voor wat het is. Het is nog te overzien.

closed 2020-03-27 by dennissiemensma

Phyxion, 2020-03-27:
Het wordt inderdaad ook wel wat een kleurplaat dan. Misschien in plaats van groen is een grijstint dan beter. Is het anders een idee om dit configureerbaar te maken? Net als de grafiek kleuren nu, dan kan iedereen kiezen wat ie zelf wil. Voor mij, persoonlijk, vind ik bijvoorbeeld het oranje van gas weinig toevoegen (zeker omdat de totalen ook gewoon rood zijn).

dennissiemensma, 2020-03-28:
Het is aardig wat werk om ook al die dingen configureerbaar te maken. Ik zal ooit nog een andere matrix proberen, maar de huidige was degene die tot nu toe het duidelijkste was.