dsmrreader/dsmr-reader #939: Gas in kWh / Meten tussen 2 punten

DarkShadowNight, 2020-03-26:
Hallo,

Zou het mogelijk zijn het gasverbruik in kWh weer te geven?
Uiteraard zijn er verschillende gassamenstellingen, dus graag ergens een instelbare correctiefactor.

Ik vrees dat het niet mogelijk is om het verbruik tussen 2 "markeringen" te meten:
- Hoe veel gas heeft m'n douche / bad nu gebruikt
- Hoe veel gas heeft de vaatwas nu gebruikt (natuurlijk maakt het sluipverbruik hier een fout), zou leuk zijn...

dennissiemensma, 2020-03-27:
Bedankt voor je vraag. Zou je het nog wat anders willen proberen uit te leggen, want ik volg je maar deels (ik mis vast iets heel simpels in je verhaal)? Bedoel je toevallig gemixte grafieken?

Phyxion, 2020-03-31:
Ik denk dat hij bedoelt of er een optie in kan of de weergave van m3 omgezet kan worden naar kWh, 1 kuub is ongeveer 9,8 kWh. Maar dit wisselt per regio.

dennissiemensma, 2020-03-31:
Bedankt voor de opheldering. Ik had er zelf nog nooit van gehoord, maar als het inderdaad dat betreft (https://www.aanbieders.be/energie/veelgestelde-vragen/gas-energie/hoeveel-kwh-in-m3), dan begrijp ik de wens al beter. @DarkShadowNight is dat wat je bedoelt?

closed 2020-05-03 by dennissiemensma