dsmrreader/dsmr-reader #941: Error die wijst naar database-probleem.

JohannesIV, 2020-03-28:
**Wat gebruik je?**
* DSMR-reader versie: v3.6
* Hardware: UP2 (x86 Intel)
* Native of Docker: native
* DSMR-protocol slimme meter: ??

De interface van dsmr-reader werkt prima, en de 'total reading count' onder statistics loopt langzaam op. Echter, in 'Archive' missen de laatste weken. Als ik naar de pagina 'Status' ga krijg ik een server error:

```
Server Error

Sorry, something unexpected happened.
Exception:

InternalError: invalid page in block 307 of relation base/16385/16672
Traceback:
(gevolgd door diverse verwijzingen naar python scripts)
```
Ik heb "Invalid page in block 307 of relation base/16385/16672" gegoogled, en dit lijkt een bericht van postgresql - ofwel probleem met de database. Ik ben ingelogd op dsmrreader en de postgresql database. Dat levert volgende op:

```
dsmrreader=#  SELECT * FROM dsmr_consumption_electricityconsumption;
ERROR:  timestamp out of range
```

Ik heb geen idee wat ik nu kan doen. Ik heb backups die dateren van de laatste datums die in 'Archive' voorkomen, maar de readings count is intussen wel doorgelopen. 
* Is alle tussenliggende data als verloren te beschouwen of staat die nog in een buffer?
* Verhelpt een restore van de dsmr-backup dit probleem?
* Is deze database te redden of kan ik beter een backup terugzetten?

dennissiemensma, 2020-03-28:
Bedankt voor je melding. Ik heb hier zelf helaas geen oplossing voor. Wellicht wel goed om te kijken of je SD-kaart/disk nog goed is.

De belangrijkste data staat in `dsmr_stats_daystatistics` en `dsmr_stats_hourstatistics`. Je zou nog kunnen kijken of die die data er uit kan halen (kan ook via de CSV-export via de interface), maar als het Archief al data mist, ben ik bang dat het er niet meer in staat.

Je kunt nog dit proberen: https://lxadm.com/PostgreSQL:_ERROR:_invalid_page_header_in_block_13760_of_relation_base/16995/67484 . Maar ik weet niet wat het resultaat ervan is en of het betrouwbaar is. 

De betrouwbaarste oplossing is een database-backup terugzetten. Mocht je daar niet uit komen dan hoor ik het graag.

Tot slot kun je nog overwegen om de datalogger-sleep (admin interface) te verhogen als je een DSMR-v5-meter hebt. Dat slijt de storage wat minder snel.

JohannesIV, 2020-03-28:
Op basis van bovenstaande leek mij een backup-restore het meest voor de hand te liggen. Ik heb de database verwijderd een een lege/nieuwe aangemaakt:
```
$ sudo service postgresql restart
$ sudo sudo -u postgres psql -c "drop database dsmrreader"                                                                                                                                               
DROP DATABASE
$ sudo sudo -u postgres psql -c "create database dsmrreader"                                                                                                                                             
CREATE DATABASE
```

En geprobeerd de backup terug te zetten:
```
$ zcat /home/dsmr/dsmr-reader/backups/archive/2020/02/dsmrreader-postgresql-partial-backup-2020-02-08.sql.gz | sudo sudo -u postgres psql dsmrreader
SET
SET
SET
SET
SET
 set_config 
------------
 
(1 row)

SET
SET
SET
ERROR:  relation "public.dsmr_stats_daystatistics" does not exist
invalid command \.
ERROR:  syntax error at or near "1"
LINE 1: 1 2018-01-21 0.00 7.284 0.000 0.000 0.000 0.00 0.00 2.102 0....
        ^
```

Is mijn backup nu ook kapot?

dennissiemensma, 2020-03-28:
Heb je geen volledige backup meer? De partial backup is ook prima, maar bevat alleen de dag- en uurtotalen.

Als je de partial wil terugzetten zul je eerst de tabellen moeten (laten) aanmaken. De partial bevat alleen data en geen structuur.

Volgens mij is zoiets afdoende:
```
sudo supervisorctl stop all

sudo su - postgres
dropdb dsmrreader
createdb -O dsmrreader dsmrreader
logout

sudo su - dsmr
./manage.py migrate
logout

zcat /home/dsmr/dsmr-reader/backups/archive/2020/02/dsmrreader-postgresql-partial-backup-2020-02-08.sql.gz | sudo sudo -u postgres psql dsmrreader

sudo supervisorctl start all
```

Als het goed is zou je dan geen fouten moeten krijgen.

JohannesIV, 2020-03-28:
Ik heb in de map "backups/manually/" nog een volledige backup. Waarschijnlijk gemaakt voordat ik de upgrade naar v3 heb gewaagd. Ik heb de beschreven stappen uitgevoerd en alleen de laatste anders gedaan omdat het een .sql bestand was en geen .sql.gz. Hierbij kreeg ik voor alle regels de zelfde error over \N:

```
$ sudo sudo -u postgres psql dsmrreader -f /home/dsmr/dsmr-reader/backups/manually/dsmrreader-postgresql-backup-Monday.sql 
[...]
psql:/home/dsmr/dsmr-reader/backups/manually/dsmrreader-postgresql-backup-Monday.sql:3828283: invalid command \N
psql:/home/dsmr/dsmr-reader/backups/manually/dsmrreader-postgresql-backup-Monday.sql:3828284: invalid command \N
psql:/home/dsmr/dsmr-reader/backups/manually/dsmrreader-postgresql-backup-Monday.sql:3828285: invalid command \N
psql:/home/dsmr/dsmr-reader/backups/manually/dsmrreader-postgresql-backup-Monday.sql:3828286: invalid command \.
psql:/home/dsmr/dsmr-reader/backups/manually/dsmrreader-postgresql-backup-Monday.sql:3828288: ERROR:  syntax error at or near "1"
LINE 1: 1 2018-01-21 14:19:00+01 2648.865 0.000 3121.046 0.000 0.315...
        ^
```

Als ik in de .sql ga kijken naar de eerste 10 echte regels data lijkt die \N te slaan op de velden "phase_currently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3, phase_currently_returned_l1, phase_currently_returned_l2, phase_currently_returned_l3, phase_voltage_l1, phase_voltage_l2, phase_voltage_l3" - als ik zou moeten raden zijn deze \N te interpreteren als leeg/null.

```
$ head -n 2565 /home/dsmr/dsmr-reader/backups/manually/dsmrreader-postgresql-backup-Monday.sql  | tail 
COPY public.dsmr_consumption_electricityconsumption (id, read_at, delivered_1, returned_1, delivered_2, returned_2, currently_delivered, currently_returned, phase_currently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3, phase_currently_returned_l1, phase_currently_returned_l2, phase_currently_returned_l3, phase_voltage_l1, phase_voltage_l2, phase_voltage_l3) FROM stdin;
1       2018-01-21 14:19:00+01  2648.865        0.000   3121.046        0.000   0.315   0.000   \N      \N      \N      \N      \N      \N      \N      \N      \N
2       2018-01-21 14:20:00+01  2648.869        0.000   3121.046        0.000   0.317   0.000   \N      \N      \N      \N      \N      \N      \N      \N      \N
3       2018-01-21 14:21:00+01  2648.874        0.000   3121.046        0.000   0.315   0.000   \N      \N      \N      \N      \N      \N      \N      \N      \N
4       2018-01-21 14:22:00+01  2648.879        0.000   3121.046        0.000   0.315   0.000   \N      \N      \N      \N      \N      \N      \N      \N      \N
5       2018-01-21 14:23:00+01  2648.884        0.000   3121.046        0.000   0.310   0.000   \N      \N      \N      \N      \N      \N      \N      \N      \N
6       2018-01-21 14:23:07+01  2648.890        0.000   3121.046        0.000   0.311   0.000   \N      \N      \N      \N      \N      \N      \N      \N      \N
7       2018-01-21 14:23:17+01  2648.890        0.000   3121.046        0.000   0.309   0.000   \N      \N      \N      \N      \N      \N      \N      \N      \N
8       2018-01-21 14:23:27+01  2648.891        0.000   3121.046        0.000   0.303   0.000   \N      \N      \N      \N      \N      \N      \N      \N      \N
9       2018-01-21 14:23:37+01  2648.892        0.000   3121.046        0.000   0.313   0.000   \N      \N      \N      \N      \N      \N      \N      \N      \N
```

dennissiemensma, 2020-03-28:
Welke backup is recenter? De handmatige of de partial? Ik zou sowieso de meest recente pakken. Als dat de handmatige is, laat maar weten, dan moet je even de applicatie downgraden. Voor de partial is dat vermoedelijk niet nodig.

JohannesIV, 2020-03-29:
Ik heb de meest recente backup kunnen herstellen. Ik was eerst bang dat de nieuwere backup de corruptie in de database mee zou nemen, maar omdat de partial maar één tabel is bleek dat geen probleem.

Inmiddels snap ik ook waarom de manual backup niet wilde: de laatste entry is van april 2019, wellicht omdat daar de corrupte data zat.

```
$ tail -n 4 /home/dsmr/dsmr-reader/backups/manually/dsmrreader-postgresql-backup-Monday.sql 
3825943 2019-04-10 15:51:20+02  4476.936        0.000   5164.063        0.000   0.236   0.000   \N      \N      \N     \N       \N      \N      \N      \N      \N
\.

```

In ieder geval werkt het nu weer, dank voor alle hulp!

closed 2020-03-29 by JohannesIV