dsmrreader/dsmr-reader #942: Dag en nachttarief België omgewisseld

jeroomdm, 2020-03-29:
Bij België is het dag en nachttarief omgewisseld. Dit werd al in een topic besproken. De oplossing was om gewoon de tekst aan te passen. Dit werkt perfect bij de dsmr reader zelf.
Het probleem zit hem bij de MQTT berichten. Deze zegt bij Electricity tariff "high" en gebruikt nog steeds de termen Low en High tariff.

dennissiemensma, 2020-03-29:
Bedankt voor de input. Kun je aangeven waar je exact die termen ziet? Ik heb even snel in MQTT config gekeken, maar daar refereert het systeem naar de tarieven 1 en 2, zonder naam.

closed 2020-04-28 by dennissiemensma

Glodenox, 2022-04-26:
Sorry om in deze gesloten issue te antwoorden, maar ik kan bevestigen dat MQTT alleen een 1 of 2 doorgeeft. Alleen heb je integraties zoals DSMR voor Home Assistant [die er standaard van uit gaan dat 1 het lage tarief is en 2 het hoge tarief](https://github.com/home-assistant/core/blob/dev/homeassistant/components/dsmr_reader/definitions.py#L34). Dit lijkt me iets dat aanpasbaar moet worden in de integratie van HA, dus ik heb er al [een issue](https://github.com/home-assistant/core/issues/66084) voor aangemaakt, maar anderzijds lijkt het me voor alle andere integraties handiger om dit hier bij de bron ook te kunnen beheren met een flag.

fverpoest, 2022-04-26:
Het dubbel tarief verdwijnt in Belgie op 01/07. Dus ik denk dat dit probleem binnenkort definitief van de baan is? (voor de Belgen dan toch)

Glodenox, 2022-04-26:
Het dubbel tarief verdwijnt alleen maar voor de berekening van het distributietarief. De elektriciteit zelf zal voor de meeste leveranciers nog altijd een piek- en een daltarief kennen.

dennissiemensma, 2022-04-26:
Bedankt voor jullie verzoek/input. Je kunt hier vanuit diverse hoeken naar kijken, alleen zal ik dat vooral doen vanuit de data gezien (wat niet perse functioneel is).

De slimme meters werken met tarieven 1 en 2, ook in bijvoorbeeld Belgie. De namen van de tarieven staan daar los van. Elke verwerker van slimme meter data zal daarom zelf context moeten geven aan wat een tarief is, betekent en hoe het heet. Dit geldt voor DSMR-reader, maar ook voor simpelweg de netbeheerder of energiemaatschappijen. 
Echter, onderwater blijft tarief 1 is altijd tarief 1 en idem voor tarief 2. In Belgie hebben ze zelfs andere namen (`Dagtarief` / `Nachttarief`) en zijn ze dus niet eens perse omgewisseld, waardoor ik eigenlijk uberhaupt afraad om tariefnummers onderwater te gaan wisselen. 

Voor DMSR-reader zal ik het in ieder geval **niet** ondersteunen, omdat het ook andere praktische risico's en uitdagingen met zich meebrengt voor gebruikers op termijn. 
Zoals dat dit na een herinstallatie direct goed ingesteld moet worden en ook wanneer je een andere tool dan DSMR-reader gebruikt. Je kunt je dan vanaf dat moment eigenlijk niet meer veroorloven om een bron (zoals nu DSMR-reader) data te laten pushen waarbij tarief 1 weer opeens tarief 1 is, ipv de tarief 2 die je zelf omgewisseld hebt. Dus daar zul je vanaf dan altijd scherp op moeten zijn. 
Daarnaast worden de gegevens er ook mogelijk niet-overdraagbaar van. Als je ooit bijvoorbeeld weer wisselt van HA naar een andere tool (welke dan ook), dan moet je weer zorgen dat je al je oude data transformeert naar whatever die tool nodig heeft. Of je moet het voor die tool ook weer gaan aanvragen en instellen dat je wisselt. 
Eigenlijk zie ik, nu ik dit uitwerk, alleen maar nadelen en adviseer ik je vooral om niet te gaan hacken met de tariefnummers.

Begrijp me overigens niet verkeerd. Ik stond voor hetzelfde dilemma in #750. En ik was ook [geneigd om tarieven wisselbaar te maken](https://github.com/dsmrreader/dsmr-reader/issues/750#issuecomment-545539988). Daar ben ik [later van teruggekomen](https://github.com/dsmrreader/dsmr-reader/issues/750#issuecomment-549138360) en toen ik het nog weer later [werkend had](https://github.com/dsmrreader/dsmr-reader/issues/750#issuecomment-587841625) door gewoon gebruikers de namen van de tariefnummers in te laten stellen, waren alle issues opgelost. 
Voor iedereen daarna (zoals #1290 / #1396). Nooit meer wat over gehoord, want iedereen kan het voor zichzelf bepalen.

dennissiemensma, 2022-04-26:
@Glodenox aanvullend heb ik nog even het issue doorgelezen die je bij HA hebt aangemaakt. 

Ik ben zelf niet heel bekend met hoe die HA-componenten werken en weet niet hoeveel werk het is om daar een configureerbare user-setting te maken voor de tariefnamen. Echter is mijn advies dus om het wel in die richting te zoeken als het kan. 

Dan hebben jullie ook een oplossing die voor iedereen werkt en blijft werken. Niet alleen nu voor NL- en BE-tarieven, maar wellicht ook voor (toekomstige) andere landen, die weer andere benamingen hebben dan de onze.

Glodenox, 2022-04-26:
Ik begrijp de logica en kan me vinden in je uitleg. Bedankt voor het uitgebreide antwoord, dat wordt echt wel geapprecieerd!

Ik hoopte zo meteen een oplossing te vinden voor alle tools die dit probleem gaan hebben, maar met je uitleg zie ik wel dat dit ook redelijk wat andere impact gaat hebben. Het is uiteindelijk de fout van de tools om een hard-coded interpretatie doen op de data.

Mijn Python-kennis zit ver weg, maar ik denk wel dat ik me er zal kunnen door werken eenmaal ik een werkende ontwikkelomgeving opgezet krijg en wat tijd kan vinden. Om een of andere reden zijn er altijd problemen met dat versiebeheer als ik iets met Python wil doen...

Glodenox, 2022-05-11:
@dennissiemensma Ik zie wel dat er in de beschrijving van "Electricity tariff" op `/admin/dsmr_datalogger/meterstatistics/` nadrukkelijk vermeld staat dat tarief 1 het daltarief is. Het valt te overwegen om dit dan aan te passen om verwarring te voorkomen.

> Tariefindicatie. Dit kan gebruikt worden om te wisselen met tariefafhankelijke vraag, zoals bijvoorbeeld een boiler. Verantwoording is voor de gebruiker van de P1-poort. N.B.: Tariefcode 1 is gebruikt voor daltarief en tariefcode 2 voor piektarief.

dennissiemensma, 2022-05-11:
@Glodenox bedankt voor het doorgeven. Dit was volgens mij destijds de letterlijke tekst uit de DSMR-specs, maar je bent de eerste die het opvalt. 
Ik zal het een keer meenemen in een release via #1631.

dennissiemensma, 2022-05-24:
Laatste is meegenomen in [v5.2](https://dsmr-reader.readthedocs.io/en/v5/reference/changelog.html#v5-2-0-2022-05-24)