dsmrreader/dsmr-reader #943: HELP thousands of "NoneType: None" entries in dsmr_backend.log (size: 3.6M)

MarcoLooy, 2020-03-29:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.7
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 4
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: v5

**HELP** thousands of "**NoneType: None**" entries in **dsmr_backend.log** (current size: 3.6M)

Upgrade to the latest version (v3.7) did not resolve the problem, please advice?

----
_NoneType: None
[2020-03-29 15:03:38,291] ERROR    2020-03-29 02:00:00
NoneType: None
[2020-03-29 15:03:39,362] ERROR    2020-03-29 02:00:00
NoneType: None
[2020-03-29 15:03:40,395] ERROR    2020-03-29 02:00:00
NoneType: None
[2020-03-29 15:03:41,428] ERROR    2020-03-29 02:00:00
NoneType: None
[2020-03-29 15:03:42,462] ERROR    2020-03-29 02:00:00
NoneType: None
[2020-03-29 15:03:43,495] ERROR    2020-03-29 02:00:00
NoneType: None
[2020-03-29 15:03:44,528] ERROR    2020-03-29 02:00:00
NoneType: None
[2020-03-29 15:03:45,562] ERROR    2020-03-29 02:00:00
NoneType: None
[2020-03-29 15:03:46,595] ERROR    2020-03-29 02:00:00
NoneType: None
[2020-03-29 15:03:47,628] ERROR    2020-03-29 02:00:00
NoneType: None
[2020-03-29 15:03:48,662] ERROR    2020-03-29 02:00:00_
----

dennissiemensma, 2020-03-29:
Thanks for reporting. I've not seen this before and unfortunately I'm not having the same issue here.

You can do several things:
- First, increase the sleep of the backend, so it doesn't hammer as much. See the admin interface -> backend configuration -> Backend process sleep. 5 seconds or higher should be fine.

- Secondly, try to extend the logging format, so the file triggering the error is more easily identified:
```
sudo su - dsmr
vi dsmrreader/settings.py

# Add these three lines to bottom of settings.py
verbose_format = '[%(asctime)s] %(levelname)s | file: %(filename)s | path: %(pathname)s | %(message)s'
LOGGING['formatters']['simple']['format'] = verbose_format
LOGGING['formatters']['verbose']['format'] = verbose_format

# Save the file.

# Apply changes
./reload.sh
```
Check the logfile again to see oif the output has changed.

MarcoLooy, 2020-03-29:
Dennis,

Thanks a lot for helping out.

This is what I got so far:

NoneType: None
[2020-03-29 17:45:49,965] ERROR | file: schedule.py | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | 2020-03-29 02:00:00
NoneType: None
[2020-03-29 17:45:55,026] ERROR | file: schedule.py | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | 2020-03-29 02:00:00
NoneType: None
[2020-03-29 17:46:04,335] ERROR | file: schedule.py | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | 2020-03-29 02:00:00
NoneType: None
[2020-03-29 17:46:09,373] ERROR | file: schedule.py | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | 2020-03-29 02:00:00

Marco.

dennissiemensma, 2020-03-29:
Would you mind updating it to:

```
LOGGING['loggers']['commands']['level'] = 'DEBUG'
verbose_format = '[%(asctime)s] %(levelname)s | file: %(filename)s @ %(lineno)d | path: %(pathname)s | %(message)s'
LOGGING['formatters']['simple']['format'] = verbose_format
LOGGING['formatters']['verbose']['format'] = verbose_format
```

And then: `./reload.sh` again.

There should be more info about what the system is exactly trying to do

MarcoLooy, 2020-03-29:
This is the repetitive pattern:

NoneType: None
[2020-03-29 18:15:22,155] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 18:15:27,166] DEBUG | file: schedule.py @ 24 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 18:15:27,194] ERROR | file: schedule.py @ 18 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | 2020-03-29 02:00:00
NoneType: None
[2020-03-29 18:15:27,195] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 18:15:32,206] DEBUG | file: schedule.py @ 24 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 18:15:32,233] ERROR | file: schedule.py @ 18 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | 2020-03-29 02:00:00
NoneType: None
[2020-03-29 18:15:32,233] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 18:15:37,248] DEBUG | file: schedule.py @ 24 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 1 backend service(s) ready to run
[2020-03-29 18:15:37,248] DEBUG | file: schedule.py @ 27 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: Running "Generate consumption data" (dsmr_consumption.services.run)
[2020-03-29 18:15:37,839] DEBUG | file: services.py @ 58 | path: /home/dsmr/dsmr-reader/dsmr_consumption/services.py | Compact: Waiting for newer readings before grouping data...
[2020-03-29 18:15:39,468] DEBUG | file: schedule.py @ 76 | path: /home/dsmr/dsmr-reader/dsmr_backend/models/schedule.py | SP: Rescheduled "Generate consumption data" to 2020-03-29 18:15:52.840116+02:00 (ETA 0:00:13.372229)
[2020-03-29 18:15:39,517] ERROR | file: schedule.py @ 18 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | 2020-03-29 02:00:00

dennissiemensma, 2020-03-29:
What does this show:

```
sudo su - postgres
psql dsmrreader

select * from dsmr_backend_scheduledprocess;
```

dennissiemensma, 2020-03-29:
I think I'll create a branch for you with some additional logging.

MarcoLooy, 2020-03-29:
id |                   name                    |                  module                  |            planned            | active |       last_executed_at        
----+-------------------------------------------+------------------------------------------+-------------------------------+--------+-------------------------------
  4 | Upload gas meter position to MinderGas.nl | dsmr_mindergas.services.run              | 2019-11-19 19:37:56.127256+00 | f      | 
  1 | Backup per email                          | dsmr_backup.services.email.run           | 2020-04-04 14:34:18.468899+01 | t      | 2020-03-28 13:34:17.390159+00
  2 | Automatic update checker                  | dsmr_backend.services.update_checker.run | 2020-04-01 20:38:33.385598+01 | t      | 2020-03-25 19:38:33.188131+00
  6 | Generate consumption data                 | dsmr_consumption.services.run            | 2020-03-29 17:26:39.97317+01  | t      | 2020-03-29 17:26:24.915491+01
  5 | Weather update                            | dsmr_weather.services.run                | 2020-03-29 18:00:00+01        | t      | 2020-03-29 17:01:31.788271+01
  7 | Retention data rotation                   | dsmr_datalogger.services.retention.run   | 2020-03-29 14:02:03.258562+01 | f      | 2020-03-29 14:02:05.05439+01
  3 | Generate day and hour statistics          | dsmr_stats.services.run                  | 2020-03-29 23:00:00+01        | t      | 2020-03-28 23:05:04.807389+00
(7 rows)

MarcoLooy, 2020-03-29:
BTW, the goal of this weekend was to enable "Backup per email", everything else was working fine for months before this weekend.

dennissiemensma, 2020-03-29:
Can you try this and check the logs again?

```
sudo su - dsmr
git fetch
git checkout -b debug/943-logging origin/debug/943-logging
./reload.sh
```

MarcoLooy, 2020-03-29:
[2020-03-29 18:51:16,147] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 18:51:21,157] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 18:51:21,185] ERROR | file: schedule.py @ 22 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | (NonExistentTimeError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: 2020-03-29 02:00:00
[2020-03-29 18:51:21,185] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 18:51:26,195] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 18:51:26,222] ERROR | file: schedule.py @ 22 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | (NonExistentTimeError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: 2020-03-29 02:00:00
[2020-03-29 18:51:26,223] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 18:51:31,231] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 1 backend service(s) ready to run
[2020-03-29 18:51:31,232] DEBUG | file: schedule.py @ 32 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: Running "Generate consumption data" (dsmr_consumption.services.run)
[2020-03-29 18:51:31,257] DEBUG | file: services.py @ 58 | path: /home/dsmr/dsmr-reader/dsmr_consumption/services.py | Compact: Waiting for newer readings before grouping data...
[2020-03-29 18:51:31,264] DEBUG | file: schedule.py @ 76 | path: /home/dsmr/dsmr-reader/dsmr_backend/models/schedule.py | SP: Rescheduled "Generate consumption data" to 2020-03-29 18:51:46.258119+02:00 (ETA 0:00:14.994190)
[2020-03-29 18:51:31,290] ERROR | file: schedule.py @ 22 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | (NonExistentTimeError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: 2020-03-29 02:00:00
[2020-03-29 18:51:31,291] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 18:51:36,301] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 18:51:36,325] ERROR | file: schedule.py @ 22 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | (NonExistentTimeError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: 2020-03-29 02:00:00
[2020-03-29 18:51:36,325] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 18:51:41,335] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 18:51:41,362] ERROR | file: schedule.py @ 22 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | (NonExistentTimeError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: 2020-03-29 02:00:00
[2020-03-29 18:51:41,363] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 18:51:46,373] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 1 backend service(s) ready to run
[2020-03-29 18:51:46,374] DEBUG | file: schedule.py @ 32 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: Running "Generate consumption data" (dsmr_consumption.services.run)
[2020-03-29 18:51:46,407] DEBUG | file: services.py @ 58 | path: /home/dsmr/dsmr-reader/dsmr_consumption/services.py | Compact: Waiting for newer readings before grouping data...
[2020-03-29 18:51:46,414] DEBUG | file: schedule.py @ 76 | path: /home/dsmr/dsmr-reader/dsmr_backend/models/schedule.py | SP: Rescheduled "Generate consumption data" to 2020-03-29 18:52:01.407766+02:00 (ETA 0:00:14.993125)
[2020-03-29 18:51:46,442] ERROR | file: schedule.py @ 22 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | (NonExistentTimeError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: 2020-03-29 02:00:00
[2020-03-29 18:51:46,442] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 18:51:51,453] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 18:51:51,496] ERROR | file: schedule.py @ 22 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | (NonExistentTimeError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: 2020-03-29 02:00:00
[... cut ...]
[2020-03-29 18:51:56,508] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 18:51:56,534] ERROR | file: schedule.py @ 22 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | (NonExistentTimeError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: 2020-03-29 02:00:00
[2020-03-29 18:51:56,534] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.comm

dennissiemensma, 2020-03-29:
Great, so it does seem to be cause with the DST change: `NonExistentTimeError`

What's the output of:
```
sudo su - postgres
psql dsmrreader

select latest_backup from dsmr_backup_backupsettings;

select latest_sync, next_sync from dsmr_backup_dropboxsettings;
```

MarcoLooy, 2020-03-29:
dsmrreader=# select latest_backup from dsmr_backup_backupsettings;
         latest_backup         
-------------------------------
 2020-03-28 01:14:39.573453+00
(1 row)

dsmrreader=# select latest_sync, next_sync from dsmr_backup_dropboxsettings;
          latest_sync          |           next_sync           
-------------------------------+-------------------------------
 2020-03-28 23:28:37.517296+00 | 2020-03-29 00:28:37.517407+00
(1 row)

dennissiemensma, 2020-03-29:
Can you try this, to see if it stops the errors:

```
sudo su - postgres
psql dsmrreader

update dsmr_backup_backupsettings set latest_backup = now();

update dsmr_backup_dropboxsettings set latest_sync = now(), next_sync = now();
```

MarcoLooy, 2020-03-29:
Awesome, you fixed it!

```
[2020-03-29 19:27:19,402] DEBUG | file: schedule.py @ 76 | path: /home/dsmr/dsmr-reader/dsmr_backend/models/schedule.py | SP: Rescheduled "Generate consumption data" to 2020-03-29 19:27:34.368054+02:00 (ETA 0:00:14.965965)
[2020-03-29 19:27:19,459] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 19:27:24,470] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 19:27:24,501] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 19:27:29,511] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 19:27:29,539] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 19:27:34,550] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 1 backend service(s) ready to run
[2020-03-29 19:27:34,550] DEBUG | file: schedule.py @ 32 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: Running "Generate consumption data" (dsmr_consumption.services.run)
[2020-03-29 19:27:34,595] DEBUG | file: services.py @ 58 | path: /home/dsmr/dsmr-reader/dsmr_consumption/services.py | Compact: Waiting for newer readings before grouping data...
[2020-03-29 19:27:34,644] DEBUG | file: schedule.py @ 76 | path: /home/dsmr/dsmr-reader/dsmr_backend/models/schedule.py | SP: Rescheduled "Generate consumption data" to 2020-03-29 19:27:49.596317+02:00 (ETA 0:00:14.951831)
[2020-03-29 19:27:34,697] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 19:27:39,712] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 19:27:39,767] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 19:27:44,783] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 19:27:44,835] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 19:27:49,842] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 1 backend service(s) ready to run
[2020-03-29 19:27:49,842] DEBUG | file: schedule.py @ 32 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: Running "Generate consumption data" (dsmr_consumption.services.run)
[2020-03-29 19:27:50,375] DEBUG | file: services.py @ 58 | path: /home/dsmr/dsmr-reader/dsmr_consumption/services.py | Compact: Waiting for newer readings before grouping data...
[2020-03-29 19:27:50,407] DEBUG | file: schedule.py @ 76 | path: /home/dsmr/dsmr-reader/dsmr_backend/models/schedule.py | SP: Rescheduled "Generate consumption data" to 2020-03-29 19:28:05.377449+02:00 (ETA 0:00:14.970104)
[2020-03-29 19:27:50,443] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
[2020-03-29 19:27:55,455] DEBUG | file: schedule.py @ 29 | path: /home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py | SP: 0 backend service(s) ready to run
[2020-03-29 19:27:55,486] DEBUG | file: mixins.py @ 73 | path: /home/dsmr/dsmr-reader/dsmr_backend/mixins.py | dsmr_backend.management.commands.dsmr_backend: Sleeping 5.0s
```

dennissiemensma, 2020-03-29:
Great, don't forget to reset the settings and branch:

- Remove the 4 added lines from `dsmrreader/settings.py`
- Branch:
```
sudo su - dsmr
git checkout v3
./reload.sh
```

(make sure the application indicates v3.7 again in the top left header)

dennissiemensma, 2020-03-29:
I will further investigate and fix the issue at some other time. For now you should have no more errors then.

MarcoLooy, 2020-03-29:
You are a star! Thanks a lot for helping out. 

BTW, others might have the same issue, while they are not aware of it.

dennissiemensma, 2020-04-01:
I've refactored the code entirely so it uses the new 'scheduled process' mechanism I'm slowly moving all background processes to. I'm not a 100% sure if the bugged code is still there, but I think it should be gone. 
Nonetheless, I've also added the additional debug logging to the next release, so any similar problems will be identified and logged more easy.

closed 2020-04-01 by dennissiemensma