dsmrreader/dsmr-reader #944: Column missing (3.6 > 3.7 upgrade)

d0mmels, 2020-03-29:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v3.7
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 3B
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: v4

After updating 3.6 to 3.7 /status and /configuration are inaccesable 
Server error:
ProgrammingError: column dsmr_backup_backupsettings.compression_level does not exist

dennissiemensma, 2020-03-29:
Thanks for reporting. Would you mind trying this:

```
sudo su - dsmr
./manage.py migrate
```

What's the output of the last command?

d0mmels, 2020-03-29:
See attached file for the output:
[output.txt](https://github.com/dennissiemensma/dsmr-reader/files/4399100/output.txt)

dennissiemensma, 2020-03-29:
Can you try this:

```
sudo su - dsmr
./manage.py dsmr_sqlsequencereset

# And then
./manage.py migrate
```

Let me know if that fixes the error from your logs:
```
psycopg2.errors.UniqueViolation: duplicate key value violates unique constraint "django_migrations_pkey"
DETAIL:  Key (id)=(144) already exists.
```

d0mmels, 2020-03-29:
Yes, output was ok without any errors this time and all issues seem to be gone.
Thank you!

closed 2020-03-29 by dennissiemensma