dsmrreader/dsmr-reader #945: Updaten vanaf v2.2.3 zonder deploy.sh

voipmeister, 2020-03-29:
**Wat gebruik je?**
* DSMR-reader versie *(links- of rechtsbovenin applicatie zichtbaar)*: v2.2.3
* Hardware *(RaspberryPi 3, 4 of iets anders)*: RaspberryPi 3
* Native of Docker *(Native is de installatie direct op je RaspberryPi)*: Native
* DSMR-protocol slimme meter *(v4 of v5)*: v5

Ik draai momenteel nog v2.2.3 (ja, ik weet het 🙂 ) en zou deze graag upgraden naar 2.15.1.

Schijnbaar heb ik een zip gedownload en gebruikt bij de laatste SD-kaart crash, want ik heb geen git repo lokaal. Ik loop met `./deploy.sh` dan ook aan tegen:
```
 --- Checking Python version.
Running Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0]

 --- Pulling remote repository for new commits...
fatal: not a git repository (or any of the parent directories): .git

 --- Merging/updating checkout.
fatal: not a git repository (or any of the parent directories): .git

 >>> Running post-deployment script. <<<
```

Nu heb ik zelf nog geprobeerd om een nieuwe `git clone` te doen en dan de switchen naar de head in de v2 branch, om vervolgens deze install over mijn bestaande install heen te kopiëren en migrations te apply'en via `./manage.py migrate`, maar dan loop ik tegen het volgende aan:
```
Running migrations:
  Applying dsmr_backend.0004_backend_schedule_process_active_flag...Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.errors.UniqueViolation: duplicate key value violates unique constraint "django_migrations_pkey"
DETAIL:  Key (id)=(100) already exists.
```

Zou je me een zetje in de juiste richting willen geven voor het upgraden van 2.2.3 naar 2.15.1? 

Thanks!

dennissiemensma, 2020-03-29:
Wil je dit eens proberen:

```
sudo su - postgres
psql dsmrreader

BEGIN;
SELECT setval(pg_get_serial_sequence('"django_migrations"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "django_migrations";
COMMIT;
```

Daarna nogmaals proberen:
```
sudo su - dsmr
./manage.py migrate
```

voipmeister, 2020-03-30:
De outpout van het SQL statement is:
```
 setval 
--------
    107
(1 row)
```

Is het de bedoeling dat ik 2.15.1 over mijn bestaande 2.2.3 heen kopieer voordat ik:
```
sudo su - dsmr
./manage.py migrate
```  
doe?

dennissiemensma, 2020-03-30:
Ja klopt. Je kunt `migrate` zovaak uitvoeren als je wilt, zou altijd goed moeten gaan omdat die kijkt naar welke migraties gedaan zijn en welke niet.

voipmeister, 2020-03-30:
Dat gaat al wel beter, maar helaas niet 100%:
```
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ ./manage.py migrate
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  Applying dsmr_backend.0004_backend_schedule_process_active_flag... OK
  Applying dsmr_backend.0005_schedule_auto_update_check... OK
  Applying dsmr_backend.0006_backend_auto_update_check... OK
  Applying dsmr_backend.0007_schedule_stats_generator... OK
  Applying dsmr_backend.0008_scheduled_process_fields... OK
  Applying dsmr_backend.0009_update_email_settings_mail_from... OK
  Applying dsmr_backend.0010_process_sleep... OK
  Applying dsmr_backend.0011_disable_capabilities... OK
  Applying dsmr_backend.0012_fix_email_from... OK
  Applying dsmr_backup.0007_remove_backupsettings_compress... OK
  Applying dsmr_consumption.0011_track_phase_voltages... OK
  Applying dsmr_consumption.0012_add_voltage_indexes... OK
  Applying dsmr_datalogger.0012_track_phase_voltages... OK
  Applying dsmr_datalogger.0013_remove_datalogger_track_settings... OK
  Applying dsmr_datalogger.0014_remove_optional_crc_check... OK
  Applying dsmr_datalogger.0015_datalogger_foreign_countries... OK
  Applying dsmr_datalogger.0016_process_sleep... OK
  Applying dsmr_datalogger.0017_log_telegrams... OK
  Applying dsmr_frontend.0020_v230_release... OK
  Applying dsmr_frontend.0021_v260_release... OK
  Applying dsmr_frontend.0022_v270_release... OK
  Applying dsmr_frontend.0023_v290_release... OK
  Applying dsmr_frontend.0024_v2100_release... OK
  Applying dsmr_frontend.0025_increase_max_dashboard_graph_width... OK
  Applying dsmr_frontend.0026_v2140_release... OK
  Applying dsmr_frontend.0027_v215_release... OK
  Applying dsmr_mindergas.0005_schedule_mindergas_export... OK
  Applying dsmr_mqtt.0009_remove_optional_debug... OK
  Applying dsmr_mqtt.0010_mqtt_gas_consumption... OK
  Applying dsmr_mqtt.0011_mqtt_meta_description... OK
  Applying dsmr_mqtt.0012_mqtt_voltage_defaults... OK
  Applying dsmr_mqtt.0013_process_sleep... OK
  Applying dsmr_weather.0005_weather_refactor_scheduling... OK
  Applying dsmr_weather.0006_schedule_weather_update... OK
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.errors.UniqueViolation: duplicate key value violates unique constraint "django_content_type_pkey"
DETAIL:  Key (id)=(39) already exists.
```

voipmeister, 2020-03-30:
Hmm, meerdere malen de migrations draaien lijkt het opgelost te hebben. Een keer of vijf was nodig schijnbaar.

voipmeister, 2020-03-30:
Daarna nog `./deploy.sh` gedaan, de upgrade is gelukt. Bedankt voor je hulp!

closed 2020-03-30 by voipmeister

dennissiemensma, 2020-03-30:
Als je die melding vaker krijgt dan moet je eigenlijk de sequence resetten. Zeg maar het autoncrement-ID zoals je die voor MySQL kent. In recentere versies van DSMR-reader kun je dit even eenmalig doen:

```
sudo su - dsmr
./manage.py dsmr_sqlsequencereset
```

Hoe dan ook mooi dat het gelukt is nu!

dennissiemensma, 2020-03-30:
En aangezien je al Python 3.7 gebruikt, kun je later ook nog makkelijk verder upgraden naar 3.x als je daar behoefte aan hebt.